chipsalliance/Surelog #1399: Parameter imports are parsed wrong

RRozak, 2021-06-18:
Hi,
when the following file is parsed by Surelog:
```systemverilog
package my_pkg;
   parameter logic P = 1;
endpackage // my_pkg

module top;
   import my_pkg::P;
endmodule // top
```
the output is:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallPackages:
\_package: my_pkg (my_pkg::) simple_import.sv:1:1: , endln:3:11, parent:work@top
  |vpiDefName:my_pkg
  |vpiName:my_pkg
  |vpiFullName:my_pkg::
  |vpiParamAssign:
...
  |vpiParameter:
  \_parameter: (my_pkg::P), line:2:20, endln:2:25, parent:my_pkg::
|uhdmallModules:
\_module: work@top (work@top) simple_import.sv:5:1: , endln:7:10, parent:work@top
  |vpiDefName:work@top
  |vpiFullName:work@top
  |vpiTypedef:
  \_import: (my_pkg), line:6:11, endln:6:20
  |vpiParamAssign:
...
  |vpiParameter:
  \_parameter: (work@top.P), line:2:20, endln:2:25, parent:work@top
    |vpiName:P
    |vpiFullName:work@top.P
    |vpiImported:my_pkg
    |UINT:1
    |vpiTypespec:
    \_logic_typespec: (P), line:2:14, endln:2:19, parent:work@top.P
      |vpiName:P
      |vpiInstance:
      \_package: my_pkg (my_pkg::) simple_import.sv:1:1: , endln:3:11, parent:work@top
|uhdmtopModules:
\_module: work@top (work@top) simple_import.sv:5:1: , endln:7:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiTypedef:
  \_import: (my_pkg), line:6:11, endln:6:20
  |vpiParamAssign:
...
  |vpiParameter:
  \_parameter: (work@top.P), line:2:20, endln:2:25, parent:work@top
```
I removed the vpiParamAssign subtrees to make it shorter.

The import node is under vpiTypedef node. I think that it should by directly under vpiModule. 
There is also no information about parameter name ('P'). It looks like the whole package is imported.

alaindargelas, 2021-06-19:
The import node is under vpiTypedef node as a convention. It is not in the standard. Rafal added this convention which is good. Please discuss with him.

There is information about parameter P, you need to dump the IDs (Add -d vpi_ids) to the command line, parameter P has Id:18, it is fully described and the same parameter P with Id:18 is in the module (The text dump is not repeated that's why you thing there is no info, but there indeed is).

```
====== UHDM =======
design: (work@top), id:15
|vpiName:work@top
|vpiElaborated:1
|uhdmallPackages:
\_package: my_pkg (my_pkg::), id:0 tests/ScratchPad.sv:1:1: , endln:3:11, parent:work@top, parID:15
  |vpiDefName:my_pkg
  |vpiName:my_pkg
  |vpiFullName:my_pkg::
  |vpiParamAssign:
  \_param_assign: , id:3, line:2:20, endln:2:25, parent:my_pkg::, parID:0
    |vpiRhs:
    \_constant: , id:4, line:2:24, endln:2:25
      |vpiConstType:9
      |vpiDecompile:1
      |vpiSize:64
      |UINT:1
    |vpiLhs:
    \_parameter: (my_pkg::P), id:2, line:2:20, endln:2:25, parent:my_pkg::, parID:0
      |vpiName:P
      |vpiFullName:my_pkg::P
      |UINT:1
      |vpiTypespec:
      \_logic_typespec: (P), id:1, line:2:14, endln:2:19, parent:my_pkg::P, parID:2
        |vpiName:P
        |vpiInstance:
        \_package: my_pkg (my_pkg::), id:0 tests/ScratchPad.sv:1:1: , endln:3:11, parent:work@top, parID:15
  |vpiParameter:
  \_parameter: (my_pkg::P), id:2, line:2:20, endln:2:25, parent:my_pkg::, parID:0
|uhdmallModules:
\_module: work@top (work@top), id:16 tests/ScratchPad.sv:5:1: , endln:7:10, parent:work@top, parID:15
  |vpiDefName:work@top
  |vpiFullName:work@top
  |vpiTypedef:
  \_import: (my_pkg), id:11, line:6:11, endln:6:20
  |vpiParamAssign:
  \_param_assign: , id:7, line:2:20, endln:2:25, parent:work@top, parID:16
    |vpiRhs:
    \_constant: , id:8, line:2:24, endln:2:25, parID:7
      |vpiConstType:9
      |vpiDecompile:1
      |vpiSize:64
      |UINT:1
    |vpiLhs:
    \_parameter: (work@top.P), id:9, line:2:20, endln:2:25, parID:7
      |vpiName:P
      |vpiFullName:work@top.P
      |vpiImported:my_pkg
      |UINT:1
      |vpiTypespec:
      \_logic_typespec: (P), id:10, line:2:14, endln:2:19, parent:work@top.P, parID:9
        |vpiName:P
        |vpiInstance:
        \_package: my_pkg (my_pkg::), id:0 tests/ScratchPad.sv:1:1: , endln:3:11, parent:work@top, parID:15
          |vpiDefName:my_pkg
          |vpiName:my_pkg
          |vpiFullName:my_pkg::
          |vpiParamAssign:
          \_param_assign: , id:3, line:2:20, endln:2:25, parent:my_pkg::, parID:0
            |vpiRhs:
            \_constant: , id:4, line:2:24, endln:2:25
              |vpiConstType:9
              |vpiDecompile:1
              |vpiSize:64
              |UINT:1
            |vpiLhs:
            \_parameter: (my_pkg::P), id:2, line:2:20, endln:2:25, parent:my_pkg::, parID:0
              |vpiName:P
              |vpiFullName:my_pkg::P
              |UINT:1
              |vpiTypespec:
              \_logic_typespec: (P), id:1, line:2:14, endln:2:19, parent:my_pkg::P, parID:2
                |vpiName:P
                |vpiInstance:
                \_package: my_pkg (my_pkg::), id:0 tests/ScratchPad.sv:1:1: , endln:3:11, parent:work@top, parID:15
          |vpiParameter:
          \_parameter: (my_pkg::P), id:2, line:2:20, endln:2:25, parent:my_pkg::, parID:0
  |vpiParameter:
  \_parameter: (work@top.P), id:5, line:2:20, endln:2:25, parent:work@top, parID:16
    |vpiName:P
    |vpiFullName:work@top.P
    |vpiImported:my_pkg
    |UINT:1
    |vpiTypespec:
    \_logic_typespec: (P), id:6, line:2:14, endln:2:19, parent:work@top.P, parID:5
      |vpiName:P
      |vpiInstance:
      \_package: my_pkg (my_pkg::), id:0 tests/ScratchPad.sv:1:1: , endln:3:11, parent:work@top, parID:15
|uhdmtopModules:
\_module: work@top (work@top), id:17 tests/ScratchPad.sv:5:1: , endln:7:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiTypedef:
  \_import: (my_pkg), id:11, line:6:11, endln:6:20
  |vpiParamAssign:
  \_param_assign: , id:13, line:2:20, endln:2:25, parent:work@top, parID:17
    |vpiRhs:
    \_constant: , id:14, line:2:24, endln:2:25
      |vpiConstType:9
      |vpiDecompile:1
      |vpiSize:64
      |UINT:1
    |vpiLhs:
    \_parameter: (work@top.P), id:18, line:2:20, endln:2:25, parent:work@top, parID:17
      |vpiName:P
      |vpiFullName:work@top.P
      |vpiImported:my_pkg
      |UINT:1
      |vpiTypespec:
      \_logic_typespec: (P), id:19, line:2:14, endln:2:19, parent:work@top.P, parID:18
        |vpiName:P
        |vpiInstance:
        \_package: my_pkg (my_pkg::), id:0 tests/ScratchPad.sv:1:1: , endln:3:11, parent:work@top, parID:15
          |vpiDefName:my_pkg
          |vpiName:my_pkg
          |vpiFullName:my_pkg::
          |vpiParamAssign:
          \_param_assign: , id:3, line:2:20, endln:2:25, parent:my_pkg::, parID:0
            |vpiRhs:
            \_constant: , id:4, line:2:24, endln:2:25
              |vpiConstType:9
              |vpiDecompile:1
              |vpiSize:64
              |UINT:1
            |vpiLhs:
            \_parameter: (my_pkg::P), id:2, line:2:20, endln:2:25, parent:my_pkg::, parID:0
              |vpiName:P
              |vpiFullName:my_pkg::P
              |UINT:1
              |vpiTypespec:
              \_logic_typespec: (P), id:1, line:2:14, endln:2:19, parent:my_pkg::P, parID:2
                |vpiName:P
                |vpiInstance:
                \_package: my_pkg (my_pkg::), id:0 tests/ScratchPad.sv:1:1: , endln:3:11, parent:work@top, parID:15
          |vpiParameter:
          \_parameter: (my_pkg::P), id:2, line:2:20, endln:2:25, parent:my_pkg::, parID:0
  |vpiParameter:
  \_parameter: (work@top.P), id:18, line:2:20, endln:2:25, parent:work@top, parID:17
```

closed 2021-06-19 by alaindargelas

RRozak, 2021-06-21:
Thank you.
I didn't know that it is intentionally under vpiTypedef.