chipsalliance/Surelog #1449: Wrong variable type in ForInitStmt

kamilrakoczy, 2021-07-05:
TEST:
```
module dut ();
function automatic integer _clog2(integer value);
  integer result;
  value = value - 1;
  for (result = 0; value > 0; result = result + 1) begin
    value = value >> 1;
  end
  return result;
endfunction
endmodule
```
UHDM:
```
|vpiForInitStmt:
         \_assign_stmt: , parent:work@dut._clog2
           |vpiRhs:
           \_constant: , line:5
             |vpiConstType:9
             |vpiDecompile:0
             |vpiSize:64
           |vpiLhs:
           \_chandle_var: (work@dut._clog2.result), line:5
             |vpiName:result
             |vpiFullName:work@dut._clog2.result
```
Surelog reports wrong variable type for ForInitStmt, ``result`` was declared as ``integer``, but it is reported as ``chandle``

alaindargelas, 2021-07-06:
Fixed by #1459, check the vpiActual of the chandle_var

RRozak, 2021-07-07:
It still doesn't work when the `for loop` is inside the package.
```systemverilog
package prim_util_pkg;
   function automatic integer _clog2(integer value);
      integer result;

      for (result = 0; value > 1; result++) begin
	 value = value >> 1;
      end
      return result;
   endfunction
endpackage

module top(output integer o);
   assign o = prim_util_pkg::_clog2(15);
endmodule // top
```

alaindargelas, 2021-07-15:
Fixed by #1501

closed 2021-07-15 by alaindargelas