chipsalliance/Surelog #1454: Param assign RHS is of wrong size in top module

kbieganski, 2021-07-05:
Verilog source:
```systemverilog
module M;
  localparam logic [7:0] P = 8'h0;
endmodule
```
Generated UHDM:
```
|uhdmallModules:
\_module: work@M (work@M) dut.v:1:1: , endln:3:10, parent:work@M
  |vpiDefName:work@M
  |vpiFullName:work@M
  |vpiParamAssign:
  \_param_assign: , line:2:26, endln:2:34, parent:work@M
    |vpiRhs:
    \_constant: , line:2:30, endln:2:34
      |vpiConstType:5
      |vpiDecompile:8'h0
      |vpiSize:8
      |HEX:0
    |vpiLhs:
    \_parameter: (work@M.P), line:2:26, endln:2:34, parent:work@M
      |vpiName:P
      |vpiFullName:work@M.P
      |vpiLocalParam:1
      |HEX:0
      |vpiTypespec:
      ...
|uhdmtopModules:
\_module: work@M (work@M) dut.v:1:1: , endln:3:10
  |vpiDefName:work@M
  |vpiName:work@M
  |vpiParamAssign:
  \_param_assign: , line:2:26, endln:2:34, parent:work@M
    |vpiRhs:
    \_constant: , line:2:30, endln:2:34
      |vpiConstType:9
      |vpiDecompile:0
      |vpiSize:4
      |UINT:0
    |vpiLhs:
    \_parameter: (work@M.P), line:2:26, endln:2:34, parent:work@M
      |vpiName:P
      |vpiFullName:work@M.P
      |vpiLocalParam:1
      |HEX:0
      |vpiTypespec:
      ...
```
In `uhdmallModules` the RHS' size is correct (8 bits), but in `uhdmtopModules` it gets reduced to only 4 bits. (this also happens with nonlocal parameters)

kbieganski, 2021-07-05:
Not sure if it's related to this bug, but for this code:
```systemverilog
module M;
   parameter logic [7:0] P = '0;
endmodule
```
we get the following UHDM:
```
|uhdmtopModules:
\_module: work@M (work@M) /home/krzysztof/verilator-uhdm/uhdm-integration/tests/bug/dut.v:1:1: , endln:3:10
  |vpiDefName:work@M
  |vpiName:work@M
  |vpiParamAssign:
  \_param_assign: , line:2:26, endln:2:32, parent:work@M
    |vpiRhs:
    \_constant: , line:2:30, endln:2:32
      |vpiConstType:9
      |vpiDecompile:0
      |vpiSize:1
      |UINT:0
    |vpiLhs:
    \_parameter: (work@M.P), line:2:26, endln:2:32, parent:work@M
      |vpiName:P
      |vpiFullName:work@M.P
      |BIN:0
      |vpiTypespec:
      ...
```
But if I'm reading IEEE 1800-2017 5.7.1 correctly, this constant should be resized to `P`'s size, so it should be 8 bits, not 1.

alaindargelas, 2021-07-07:
Fixed #1471 (first part). 

For the 2nd problem:
```
module M;
   parameter logic [7:0] P = '0;
endmodule
```
Please create another Issue. It is a very different problem, which is not going to be addressed any time soon.
Surelog does not reconcile lhs and rhs expression dimensions at this time. 
This is left to the client application for now.

closed 2021-07-07 by alaindargelas