chipsalliance/Surelog #1465: Unnecessary vpiNet node

RRozak, 2021-07-06:
Hi,
when the following file is parsed by Surelog: 
```systemverilog
module dut (output logic o);
   assign o = 1'b1;
endmodule

module top (output logic o);
   typedef struct packed {
      logic 	  ping_p;
   } alert_rx_t;

   alert_rx_t  alert_rx_i;

   dut d(.o(alert_rx_i.ping_p));

   assign o = alert_rx_i.ping_p;
endmodule
```
there is wrong vpiNet node at the end:
```
  |vpiNet:
  \_logic_net: (work@top.alert_rx_i.ping_p), parent:work@top
    |vpiName:alert_rx_i.ping_p
    |vpiFullName:work@top.alert_rx_i.ping_p
```
I think it should be removed, but if it has to be in the tree (due to convention or something), there should be hier_path node inside.
I attach the uhdm tree:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module: work@dut (work@dut) uhdm-integration/tests/StructMemberAsModuleInput/top.sv:1:1: , endln:3:10, parent:work@top
  |vpiDefName:work@dut
  |vpiFullName:work@dut
...
  |vpiNet:
  \_logic_net: (work@dut.o), line:1:26, parent:work@dut
|uhdmallModules:
\_module: work@top (work@top) uhdm-integration/tests/StructMemberAsModuleInput/top.sv:5:1: , endln:15:10, parent:work@top
  |vpiDefName:work@top
  |vpiFullName:work@top
...
  |vpiNet:
  \_logic_net: (work@top.o), line:5:26, parent:work@top
  |vpiNet:
  \_logic_net: (work@top.alert_rx_i), line:10:16, parent:work@top
    |vpiName:alert_rx_i
    |vpiFullName:work@top.alert_rx_i
  |vpiTypedef:
  \_struct_typespec: (alert_rx_t), line:6:12, endln:6:18
    |vpiPacked:1
    |vpiName:alert_rx_t
    |vpiTypespecMember:
    \_typespec_member: (ping_p), line:7:16, endln:7:22
      |vpiName:ping_p
      |vpiTypespec:
      \_logic_typespec: , line:7:7, endln:7:12
|uhdmtopModules:
\_module: work@top (work@top) uhdm-integration/tests/StructMemberAsModuleInput/top.sv:5:1: , endln:15:10
  |vpiDefName:work@top
  |vpiName:work@top
...
  |vpiModule:
  \_module: work@dut (work@top.d) uhdm-integration/tests/StructMemberAsModuleInput/top.sv:12: , parent:work@top
    |vpiDefName:work@dut
    |vpiName:d
    |vpiFullName:work@top.d
...
  |vpiNet:
  \_logic_net: (work@top.o), line:5:26, endln:5:27, parent:work@top
  |vpiNet:
  \_struct_net: (work@top.alert_rx_i), line:10:16, endln:10:26, parent:work@top
    |vpiName:alert_rx_i
    |vpiFullName:work@top.alert_rx_i
    |vpiTypespec:
    \_struct_typespec: (alert_rx_t), line:6:12, endln:6:18
      |vpiPacked:1
      |vpiName:alert_rx_t
      |vpiTypespecMember:
      \_typespec_member: (ping_p), line:7:16, endln:7:22
        |vpiName:ping_p
        |vpiTypespec:
        \_logic_typespec: , line:7:7, endln:7:12
  |vpiNet:
  \_logic_net: (work@top.alert_rx_i.ping_p), parent:work@top
    |vpiName:alert_rx_i.ping_p
    |vpiFullName:work@top.alert_rx_i.ping_p
  |vpiTypedef:
  \_struct_typespec: (alert_rx_t), line:6:12, endln:6:18
```

RRozak, 2021-07-07:
Sometimes there are created `vpiNet` nodes for scalars, for example in ibex_simple_system.
I couldn't find any simpler example to reproduce it.
Part of uhdm tree:
```
  |vpiNet:
  \_logic_net: (work@ibex_simple_system.32'b0), parent:work@ibex_simple_system
    |vpiName:32'b0
    |vpiFullName:work@ibex_simple_system.32'b0
  |vpiNet:
  \_logic_net: (work@ibex_simple_system.32'h00100000), parent:work@ibex_simple_system
    |vpiName:32'h00100000
    |vpiFullName:work@ibex_simple_system.32'h00100000
  |vpiNet:
  \_logic_net: (work@ibex_simple_system.15'b0), parent:work@ibex_simple_system
    |vpiName:15'b0
    |vpiFullName:work@ibex_simple_system.15'b0
  |vpiNet:
  \_logic_net: (work@ibex_simple_system.4'b0), parent:work@ibex_simple_system
    |vpiName:4'b0
    |vpiFullName:work@ibex_simple_system.4'b0
```
It causes warnings like that: https://github.com/antmicro/verilator/runs/2997291561?check_suite_focus=true#step:7:7181
It can be reproduced by Makefile rule: https://github.com/antmicro/verilator/blob/uhdm-verilator/uhdm-tests/ibex/Makefile.in#L59

I also attach surelog.log
[surelog.log](https://github.com/chipsalliance/Surelog/files/6778075/surelog.log)

alaindargelas, 2021-07-08:
Fixed the first part with #1481

alaindargelas, 2021-07-08:
Fixed 2nd part with #1417

closed 2021-07-08 by alaindargelas