chipsalliance/Surelog #1466: Actual_Group bind problem in _hier_path

bsp13, 2021-07-06:
```
typedef struct{
reg state,nextstate;
}myState;
module fsm_structue (input clk, input reset, input x, output reg parity);
 myState a;
 parameter S0=0,S1=1;

always @(posedge clk) // always block to update a.state
if (reset)
 a.state <= S0;
else
 a.state <= a.nextstate;

always @(a.state or x) // always block to compute both output & nextstate
begin
 parity = 1'b0;
 	case(a.state)
 	S0: if(x)
	 	begin
		 parity = 1; a.nextstate = S1;
 	        end
 	    else
      		 begin parity=0; a.nextstate = S0; end
 	S1: if(x)
		 begin parity = 0; a.nextstate = S1; end
 	     else
 		 begin
                 parity = 1; a.nextstate = S0;
                 end
	 default:a.nextstate = S0;
	endcase
end
endmodule 
```

UHDM Output::
```
       |vpiStmt:
        \_case_stmt: , line:17:3, endln:31:9, parent:work@fsm_structue
          |vpiCaseType:1
          |vpiCondition:
          \_hier_path: (a.state), line:17:8, endln:17:15
            |vpiName:a.state
            |vpiActual:
            \_ref_obj: (a), parent:a.state
              |vpiName:a
              |vpiActual:
              \_struct_net: (work@fsm_structue.a), line:5:10, endln:5:11, parent:work@fsm_structue
            |vpiActual:
            \_ref_obj: (state), parent:a.state
              |vpiName:state
```

alaindargelas, 2021-09-05:
Revisiting this now

alaindargelas, 2021-09-05:
Fixed by #1983

closed 2021-09-05 by alaindargelas