chipsalliance/Surelog #1467: Actual_Group bind problem in Interface

bsp13, 2021-07-06:
```
interface logic_gate_if;

logic[3:0] a, b;
logic[5:0] c;
logic[5:0] c_2; // increment c by 2

endinterface

module or_ex(
    input logic[3:0] a, b,
    output logic[5:0] sum, sum_new
);

logic_gate_if lg ();  // import all signals from interface

assign lg.a = a;
assign lg.b = b;
assign sum = a+b;
assign sum_new = lg.a + lg.b + 2;  // increment sum by 2

endmodule 
```

UHDM Output::
```
 |vpiContAssign:
  \_cont_assign: , line:17:8, endln:17:16, parent:work@or_ex
    |vpiRhs:
    \_ref_obj: (work@or_ex.a), line:17:15, endln:17:16
      |vpiName:a
      |vpiFullName:work@or_ex.a
      |vpiActual:
      \_logic_net: (work@or_ex.a), line:11:22, endln:11:23, parent:work@or_ex
    |vpiLhs:
    \_hier_path: (lg.a), line:17:8, endln:17:10
      |vpiName:lg.a
      |vpiActual:
      \_ref_obj: (lg), parent:lg.a         <-------------------------
        |vpiName:lg                       
      |vpiActual:
      \_ref_obj: (a), parent:lg.a
        |vpiName:a
        |vpiActual:
        \_logic_net: (work@or_ex.a), line:11:22, endln:11:23, parent:work@or_ex
```

alaindargelas, 2021-07-25:
Both of these issues have low priority. Today Surelog only does binding for the base object of a hierarchical path. The selectors are not bound. This is a significant effort and it is an enhancement.

bsp13, 2021-07-27:
Both of these issues (#1467 or #1466). In One issue Surelog does binding base object and in another issue Surelog does binding selectors.

alaindargelas, 2021-09-15:
Fixed by #2102

closed 2021-09-15 by alaindargelas