chipsalliance/Surelog #1473: Assignment pattern copied to param without necessary info

kbieganski, 2021-07-07:
Let's say we have a `logic` param which we initialize with a struct which is initialized by an assignment pattern. Example:
```systemverilog
module Mod1 #(parameter logic [31:0] PARAM = '0) (output logic [31:0] x);
   assign x = PARAM;
endmodule

module Mod2;
   typedef struct packed { logic[31:0] x; } Struct;
   parameter Struct STRUCT = '{ x: '0 };
   logic[31:0] x;
   Mod1 #(.PARAM(STRUCT)) mod1(.x(x));
endmodule
```
UHDM generated by Surelog:
```
 |uhdmtopModules:
 \_module: work@Mod2 (work@Mod2) bug.sv:5:
   ...
   |vpiModule:
    \_module: work@Mod1 (work@Mod2.mod1) bug.sv:9: , parent:work@Mod2
     ...
     |vpiParamAssign:
     \_param_assign: , line:1, parent:work@Mod2.mod1
       |vpiRhs:
       \_operation: , line:9
         |vpiOpType:75
         |vpiOperand:
         \_tagged_pattern:
           |vpiTypespec:
           \_string_typespec: (x), line:7
             |vpiName:x
           |vpiPattern:
           \_constant: , line:7
             |vpiConstType:3
             |vpiDecompile:'b0
             |BIN:0
       |vpiLhs:
       \_parameter: (work@Mod2.mod1.PARAM), line:1, parent:work@Mod2.mod1
         |vpiName:PARAM
         |vpiFullName:work@Mod2.mod1.PARAM
         |BIN:0
         |vpiTypespec:
         \_logic_typespec: (PARAM), line:1, parent:work@Mod2.mod1.PARAM
           ...

```
The assignment pattern `{ x: '0 }` gets assigned directly to `PARAM`. The problem is, `PARAM` is of type `logic[31:0]`, so it doesn't contain member `x`. From what I see, there isn't a way to figure out what typespec is being 'used' by the pattern (there's no link between `PARAM` and `STRUCT` anywhere in the UHDM).

alaindargelas, 2021-07-12:
We can use the typespec member of expr

alaindargelas, 2021-07-14:
Fixed by #1492, check the typespec of the expression

closed 2021-07-14 by alaindargelas