chipsalliance/Surelog #1506: vpiChandleVar instead of vpiRefObj in ForInitStmt

RRozak, 2021-07-16:
I think that the proper uhdm tree should be different than we thought here: https://github.com/chipsalliance/Surelog/issues/1449
After the fix #1501 it has the correct type, but it looks like a declaration of different variable with the same name.
When the following file is compiled and run:
```systemverilog
package prim_util_pkg;
   function automatic integer _clog2(integer value);
      integer result;

      for (result = 0; value > 1; result++) begin
	 value = value >> 1;
      end
      return result;
   endfunction
endpackage

module top(output integer o);
   assign o = prim_util_pkg::_clog2(15);
endmodule // top
```
The `o` variable equals 0, but should be equal 3.

I think that the `result` variable in `vpiForInitStmt` should be represented as vpiRefObj.

RRozak, 2021-07-16:
I created the fix: https://github.com/chipsalliance/Surelog/pull/1507

alaindargelas, 2021-07-16:
The chandle has a vpiActual to the object so you know it is not a new definition. the vpiActual points you to the definition.
A chandle with no vpiActual would be a new definition.

You cannot return a ref_obj as it violates a lot of other cases where a chandle (which is a variable type) is needed.
Your fix is incorrect and creates massive failures because the above restriction.

I don't understand your statement:
"The o variable equals 0, but should be equal 3."  
Surelog does not evaluate the RHS because it is not required for Elaboration, so the UHDM model contains a symbolic expression with the function call and its argument, no value, certainly not 0. So I don't know who produces 0, it must be some other piece of code downstream.
If you replace the assignment to a net to a parameter assignment on the LHS, you'll see the right value (3) is computed.
Surelog does evaluate in this case as parameters are required to be evaluated during elaboration.

Is there a completely differnent problem (not in Surelog/UHDM) that prevents the downstream flow to perform the function evaluation?

RRozak, 2021-07-19:
Could you tell why `ref_obj` shouldn't be used in that case instead of `chandle`?
I think that chandle type is only for DPI-C and is not intended to be used in such places. Use of `ref_obj` will give a simpler tree and won't require `-elabuhdm` option. Since folded model does not contain vpiActual for this chandle, we cannot check that this is the same variable.

I found in standard that operator `++` should not be used on chandle variables:
```
Chandles are restricted in their usage, with the only legal uses being as
follows:
— Only the following operators are valid on chandle variables:
• Equality (==), inequality (!=) with another chandle or with null
• Case equality (===), case inequality (!==) with another chandle or with null (same semantics
as == and !=)
```
(from section 6.14 Chandle data type)

Is a `chandle` type with `vpiActual` a Surelog convention or is it described in the standard?

alaindargelas, 2021-07-19:
I'll create a new var_ref type that inherits from "variables" type.
ref_obj does not inherit from "variables" that is the reason it cannot be used.
You will have to invoke the Elaboration no matter what, either with the -elabuhdm or after loading UHDM using the Elaborator class. There is NO other way to do binding. Please just accept it.

alaindargelas, 2021-07-19:
Fixed by #1515.

closed 2021-07-19 by alaindargelas