chipsalliance/Surelog #1508: Void function has logic_var in vpiReturn

RRozak, 2021-07-16:
When the following file is parsed by Surelog:
```systemverilog
package my_pkg;
   function automatic void assign_1(output logic x);
      assign x = 1'b1;
   endfunction // assign_1
endpackage // my_pkg

module top(output o);
   initial begin
      my_pkg::assign_1(o);
   end
endmodule // top
```
there is logic_var in vpiReturn node.
Below I paste the part of uhdm tree:
```
  |vpiTaskFunc:
  \_function: (my_pkg::assign_1), line:2:4, endln:4:15, parent:my_pkg::
    |vpiVisibility:1
    |vpiAutomatic:1
    |vpiName:assign_1
    |vpiFullName:my_pkg::assign_1
    |vpiReturn:
    \_logic_var: 
    |vpiInstance:
    \_package: my_pkg (my_pkg::) ../uhdm-integration/tests/FunctionOutputArgument/top.sv:1:1: , endln:5:11, parent:work@top
    |vpiIODecl:
    \_io_decl: (x)
      |vpiName:x
      |vpiDirection:2
      |vpiTypedef:
      \_logic_typespec: , line:2:44, endln:2:49
        |vpiInstance:
        \_package: my_pkg (my_pkg::) ../uhdm-integration/tests/FunctionOutputArgument/top.sv:1:1: , endln:5:11, parent:work@top
    |vpiStmt:
    \_assign_stmt: , line:3:7, endln:3:22, parent:my_pkg::assign_1
      |vpiRhs:
      \_constant: , line:3:18, endln:3:22
        |vpiConstType:3
        |vpiDecompile:1'b1
        |vpiSize:1
        |BIN:1
      |vpiLhs:
      \_ref_obj: (my_pkg::assign_1::x), line:3:14, endln:3:15
        |vpiName:x
        |vpiFullName:my_pkg::assign_1::x
```

RRozak, 2021-07-16:
The problem is in these lines:
https://github.com/chipsalliance/Surelog/blob/f38f4ff9ba96eb8c78409292a0e68ab198515f73/src/DesignCompile/CompileStmt.cpp#L1688-L1697
I can't find in standard what should be be done when the function has return type `void`.
First I wanted to remove the vpiReturn node in such case, but I'm not sure.

alaindargelas, 2021-07-16:
You are right, this was a clear bug.
Fixed with #1510

closed 2021-07-16 by alaindargelas