chipsalliance/Surelog #1516: ref_var instead of chandle_var

RRozak, 2021-07-19:
When the following file is parsed by Surelog:
```systemverilog
module top;
   import "DPI-C"
   function chandle dmidpi_create();
endmodule // top
```
The return value of a function has type `ref_var`.
UHDM:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module: work@top (work@top) dmidpi_create.sv:1:1: , endln:4:10, parent:work@top
  |vpiDefName:work@top
  |vpiFullName:work@top
  |vpiTaskFunc:
  \_function: (work@top.dmidpi_create), line:2:4, endln:3:37, parent:work@top
    |vpiAccessType:4
    |vpiVisibility:1
    |vpiDPICStr:2
    |vpiName:dmidpi_create
    |vpiFullName:work@top.dmidpi_create
    |vpiReturn:
    \_ref_var: , line:3:13, endln:3:20
    |vpiInstance:
    \_module: work@top (work@top) dmidpi_create.sv:1:1: , endln:4:10, parent:work@top
|uhdmtopModules:
\_module: work@top (work@top) dmidpi_create.sv:1:1: , endln:4:10
  |vpiDefName:work@top
  |vpiName:work@top
```

RRozak, 2021-07-19:
https://github.com/chipsalliance/Surelog/pull/1517 should fix it

alaindargelas, 2021-07-20:
Fixed by #1517

closed 2021-07-20 by alaindargelas