chipsalliance/Surelog #1526: External CMake C++ project depending on Surelog

pieter3d, 2021-07-22:
Hello! It's not clear to me what the best way is to have an external C++ project pull in Surelog.
What I've done so far:
- set an environment variable SURELOG_ROOT where surelog is checked out and built.
- Add the following to my CMakeLists.txt: 
```cmake
find_library(SURELOG_LIBRARIES NAMES surelog  PATHS $ENV{SURELOG_ROOT}/build/lib)
```
- Depend on this with my target:
```cmake
target_link_libraries(my_executable PUBLIC ${SURELOG_LIBRARIES})
```

This results in a lot of linker errors (e.g. flatbuffers, UHDM) for all of Surelog's third_party libraries. I started to add those one by one, but it started getting messy, especially with Antlr since it's not built within Surelog's `build/` directory (somehow ends up in `dist/`). 

Checking out Surelog as a submodule to my project results in collisions with my project's googletest (age-old cmake problem), and fixing this also gets messy. 

What is the recommended way to do this?

hs-apotell, 2021-07-22:
Try CMake install.
```
cmake --install <folder path>
```

This will create a distribution which pulls in all the necessary headers and libraries under one root. You should be able to map your system environment variable to this new location.

Pitfalls - This is not where you build Surelog and so you can't have a continuous build and run cycle. You will have to build *and install* each time you happen to modify Surelog.

alaindargelas, 2021-07-22:
I'm guessing the preferred method would be to have Surelog as a subsystem and make it install it's lib and headers in a standard fashion? Who can help?

hs-apotell, 2021-07-22:
> ... Surelog as a subsystem ....

Do you mean [exporting](https://cmake.org/cmake/help/latest/guide/tutorial/Adding%20Export%20Configuration.html) and [packaging](https://cmake.org/cmake/help/latest/guide/tutorial/Packaging%20Debug%20and%20Release.html)?

alaindargelas, 2021-07-22:
I'm no expert, but it should be like Surelog uses cappnp or antlr

pieter3d, 2021-07-22:
I suppose also Surelog should be exporting a CMake module file in some way, that way other projects can use Surelog libraries with `find_package`

pieter3d, 2021-07-26:
Here's a workaround for anyone wanting to include Surelog's installed libraries and includes with their CMake project:

```cmake
find_package(Threads REQUIRED)
find_library(SURELOG_LIBRARIES NAMES surelog
  PATH_SUFFIXES surelog) 
find_library(FLATBUFFERS_LIBRARIES NAMES flatbuffers
  PATH_SUFFIXES surelog) 
find_library(ANTLR_LIBRARIES NAMES antlr4-runtime
  PATH_SUFFIXES surelog) 
find_library(UHDM_LIBRARIES NAMES uhdm
  PATH_SUFFIXES uhdm) 
find_library(CAPNP_LIBRARIES NAMES capnp
  PATH_SUFFIXES uhdm) 
find_library(KJ_LIBRARIES NAMES kj
  PATH_SUFFIXES uhdm) 
find_path(SURELOG_INCLUDE_DIR "surelog/surelog.h")

set(ALL_SURELOG_LIBS
  ${SURELOG_LIBRARIES}
  ${UHDM_LIBRARIES}
  ${CAPNP_LIBRARIES}
  ${KJ_LIBRARIES}
  ${FLATBUFFERS_LIBRARIES}
  ${ANTLR_LIBRARIES}
  Threads::Threads
)

set(ALL_SURELOG_INCLUDES
  ${SURELOG_INCLUDE_DIR}/surelog
  ${SURELOG_INCLUDE_DIR}/uhdm
  ${SURELOG_INCLUDE_DIR}/uhdm/headers
  ${SURELOG_INCLUDE_DIR}/uhdm/include
)

add_library(your_lib your_lib.cc)
target_include_directories(your_lib PRIVATE ${ALL_SURELOG_INCLUDES})
target_link_libraries(your_lib PUBLIC  ${ALL_SURELOG_LIBS})
```

alaindargelas, 2021-07-27:
Thanks Pieter, I'll leave the issue open until we have a proper pull request for the Surelog CMake

hs-apotell, 2021-08-15:
@pieter3d I just submitted a PR with the necessary changes to use Surelog as an external library. Could you please give a world. Thanks!

All you should need the following statement in your project's CMakeLists.txt file.
```
find_package(surelog PATHS <path to where you installed surelog>)
```

Note: You need to both build and **install** surelog. `find_package` expects external libraries in a specific format and `install` step does that setup.

pieter3d, 2021-08-15:
I was unable to build, seeing the same errors as in the PR's build tests. I would also say it's important to update the README explaining how to use the lib. At the very least, something should explain what variables are set by the package and how to use them with libraries

pieter3d, 2021-08-15:
the UHDM PR was merged, so the submodule update should be included in your PR (that allows it to build for me)

pieter3d, 2021-08-15:
Hmm - this doesn't work for me. UHDM has build errors:

```
/usr/local/include/uhdm/headers/parameters.h:29:10: fatal error: headers/uhdm_forward_decl.h: No such file or directory
   29 | #include "headers/uhdm_forward_decl.h"
```

Seems uhdm/ prefix is missing from an include path somewhere.

alaindargelas, 2021-08-15:
Yup, it's all broken, what did you do locally exactly to get it to work?

pieter3d, 2021-08-15:
I can get surelog to build by just updating the UHDM submodule to HEAD.

But my own surelog-using project doesn't with the aforementioned error. I guess these are the issues I see with the PR
* PR needs updating to include the newer version of UHDM (simple)
* Need some docs on how to use the find_package results. What vars does it set and how are those supposed to be used in projects?
* Still requires hard coded paths even when the surelog install dir is system-default (i.e. /usr/local)

alaindargelas, 2021-08-15:
@hs-apotell, I undid the change in UHDM temporarily. This needs to be streamlined with all the comments from @pieter3d

pieter3d, 2021-08-15:
The UHDM change seems harmless on its own though?

alaindargelas, 2021-08-16:
@hs-apotell, do you think the changes in UHDM were complete? If so, please resubmit the same change list.
In the CI build there was an error about uhdm_forward_decl.h not being found.

hs-apotell, 2021-08-16:
Few different things went wrong here -
1. UHDM change had to be merged ahead of Surelog merge. The submodule had to be updated as well in Surelog. I had pointed that in the comment on the PR. UHDM changes by themselves are harmless because they don't do much except present the paths as private or public.
2. Surelog build itself was passing on both linux and windows. What was failing was the UnitTests build. I failed to verify that beforehand locally. My apologies. I didn't think my changes had any impact on that.
3. While fixing the UnitTests build in Surelog, I just noticed that the windows builds had been failing all along, at least, for over the last 20+ green builds. The error, however, was not correctly being passed upstream and so CI continued to report it as "all good".
4. Will update the README with instructions
5. I don't have a linux box handy to test on it. But AFAIK `/usr/local` should be discoverable without providing any paths for the `find_package`. Someone with easier access to a linux will have to try out and provide further insight.

pieter3d, 2021-08-16:
`find_package` didn't work without a PATHS argument for me (Linux). I don't know enough about cmake to know how other packages accomplish it (maybe they just come with cmake internals?).

hs-apotell, 2021-08-16:
This might help - https://stackoverflow.com/questions/65735392/why-doesnt-cmake-find-package-search-usr-lib-by-default

alaindargelas, 2021-08-17:
@pieter3d, I merged both @hs-apotell pull requests, can you try and give feedback?

pieter3d, 2021-08-17:
I gave it a shot. Works well. In fact, I only have to do the following in my CMakeLists.txt:
```cmake
find_package(Curses REQUIRED)
```
Very nice.

I would recommend the README.md to be updated with this, I think it's quite a bit simpler than it suggests. There's no need to set any env vars or cmake variables (at least on linux).

hs-apotell, 2021-08-17:
@pieter3d Glad to head that the process worked out. Thanks for taking the time to confirm it.

On linux, since the library is installed in /usr/local it gets picked up. There is no such location for windows. For windows, one has to provide the location of the installation as part of the CMake command. If anything the README.md instructions can clarify on that but otherwise the steps as written are required for windows platforms.

closed 2021-08-19 by pieter3d