chipsalliance/Surelog #1527: Wrong order of elements in array parameter

RRozak, 2021-07-23:
When the following file is parsed by surelog:
```systemverilog
module top(output int a, b, c, d);
   parameter int x [3:0] = '{10, 11, 12, 13};

   assign a = x[0];
   assign b = x[1];
   assign c = x[2];
   assign d = x[3];
endmodule
```
the values elements of `x` are taken in reversed order.
`a`, `b`, `c`, `d` have wrong values assigned.
The problem occurs both in packed and unpacked arrays, but only when they are parameters.
Here is the uhdm tree:
```
 slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module: work@top (work@top) parameter_array.sv:1:1: , endln:8:10, parent:work@top
  |vpiDefName:work@top
  |vpiFullName:work@top
  |vpiPort:
  \_port: (a), line:1:23, parent:work@top
    |vpiName:a
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@top.a), line:1:23, parent:work@top
        |vpiName:a
        |vpiFullName:work@top.a
  |vpiPort:
  \_port: (b), line:1:26, parent:work@top
    |vpiName:b
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@top.b), line:1:26, parent:work@top
        |vpiName:b
        |vpiFullName:work@top.b
  |vpiPort:
  \_port: (c), line:1:29, parent:work@top
    |vpiName:c
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@top.c), line:1:29, parent:work@top
        |vpiName:c
        |vpiFullName:work@top.c
  |vpiPort:
  \_port: (d), line:1:32, parent:work@top
    |vpiName:d
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@top.d), line:1:32, parent:work@top
        |vpiName:d
        |vpiFullName:work@top.d
  |vpiContAssign:
  \_cont_assign: , line:4:11, endln:4:19, parent:work@top
    |vpiRhs:
    \_constant: , line:4:15, endln:4:19
      |vpiConstType:9
      |vpiDecompile:10
      |vpiSize:64
      |UINT:10
    |vpiLhs:
    \_ref_obj: (work@top.a), line:4:11, endln:4:12
      |vpiName:a
      |vpiFullName:work@top.a
  |vpiContAssign:
  \_cont_assign: , line:5:11, endln:5:19, parent:work@top
    |vpiRhs:
    \_constant: , line:5:15, endln:5:19
      |vpiConstType:9
      |vpiDecompile:11
      |vpiSize:64
      |UINT:11
    |vpiLhs:
    \_ref_obj: (work@top.b), line:5:11, endln:5:12
      |vpiName:b
      |vpiFullName:work@top.b
  |vpiContAssign:
  \_cont_assign: , line:6:11, endln:6:19, parent:work@top
    |vpiRhs:
    \_constant: , line:6:15, endln:6:19
      |vpiConstType:9
      |vpiDecompile:12
      |vpiSize:64
      |UINT:12
    |vpiLhs:
    \_ref_obj: (work@top.c), line:6:11, endln:6:12
      |vpiName:c
      |vpiFullName:work@top.c
  |vpiContAssign:
  \_cont_assign: , line:7:11, endln:7:19, parent:work@top
    |vpiRhs:
    \_constant: , line:7:15, endln:7:19
      |vpiConstType:9
      |vpiDecompile:13
      |vpiSize:64
      |UINT:13
    |vpiLhs:
    \_ref_obj: (work@top.d), line:7:11, endln:7:12
      |vpiName:d
      |vpiFullName:work@top.d
  |vpiNet:
  \_logic_net: (work@top.a), line:1:23, parent:work@top
  |vpiNet:
  \_logic_net: (work@top.b), line:1:26, parent:work@top
  |vpiNet:
  \_logic_net: (work@top.c), line:1:29, parent:work@top
  |vpiNet:
  \_logic_net: (work@top.d), line:1:32, parent:work@top
  |vpiParamAssign:
  \_param_assign: , line:2:18, endln:2:45, parent:work@top
    |vpiRhs:
    \_operation: , line:2:28, endln:2:45
      |vpiOpType:75
      |vpiOperand:
      \_constant: , line:2:30, endln:2:32
        |vpiConstType:9
        |vpiDecompile:10
        |vpiSize:64
        |UINT:10
      |vpiOperand:
      \_constant: , line:2:34, endln:2:36
        |vpiConstType:9
        |vpiDecompile:11
        |vpiSize:64
        |UINT:11
      |vpiOperand:
      \_constant: , line:2:38, endln:2:40
        |vpiConstType:9
        |vpiDecompile:12
        |vpiSize:64
        |UINT:12
      |vpiOperand:
      \_constant: , line:2:42, endln:2:44
        |vpiConstType:9
        |vpiDecompile:13
        |vpiSize:64
        |UINT:13
    |vpiLhs:
    \_parameter: (work@top.x), line:2:18, endln:2:45, parent:work@top
      |vpiName:x
      |vpiFullName:work@top.x
      |vpiSize:1
      |vpiRange:
      \_range: , line:2:21, endln:2:24, parent:work@top.x
        |vpiLeftRange:
        \_constant: , line:2:21, endln:2:22
          |vpiConstType:9
          |vpiDecompile:3
          |vpiSize:64
          |UINT:3
        |vpiRightRange:
        \_constant: , line:2:23, endln:2:24
          |vpiConstType:9
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
      |vpiTypespec:
      \_int_typespec: (x), line:2:14, endln:2:17, parent:work@top.x
        |vpiName:x
  |vpiParameter:
  \_parameter: (work@top.x), line:2:18, endln:2:45, parent:work@top
|uhdmtopModules:
\_module: work@top (work@top) parameter_array.sv:1:1: , endln:8:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiPort:
  \_port: (a), line:1:23, endln:1:24, parent:work@top
    |vpiName:a
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_int_var: (work@top.a), line:1:23, endln:1:24, parent:work@top
        |vpiName:a
        |vpiFullName:work@top.a
        |vpiAutomatic:1
        |vpiVisibility:1
  |vpiPort:
  \_port: (b), line:1:26, endln:1:27, parent:work@top
    |vpiName:b
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_int_var: (work@top.b), line:1:26, endln:1:27, parent:work@top
        |vpiName:b
        |vpiFullName:work@top.b
        |vpiAutomatic:1
        |vpiVisibility:1
  |vpiPort:
  \_port: (c), line:1:29, endln:1:30, parent:work@top
    |vpiName:c
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_int_var: (work@top.c), line:1:29, endln:1:30, parent:work@top
        |vpiName:c
        |vpiFullName:work@top.c
        |vpiAutomatic:1
        |vpiVisibility:1
  |vpiPort:
  \_port: (d), line:1:32, endln:1:33, parent:work@top
    |vpiName:d
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_int_var: (work@top.d), line:1:32, endln:1:33, parent:work@top
        |vpiName:d
        |vpiFullName:work@top.d
        |vpiAutomatic:1
        |vpiVisibility:1
  |vpiVariables:
  \_int_var: (work@top.a), line:1:23, endln:1:24, parent:work@top
  |vpiVariables:
  \_int_var: (work@top.b), line:1:26, endln:1:27, parent:work@top
  |vpiVariables:
  \_int_var: (work@top.c), line:1:29, endln:1:30, parent:work@top
  |vpiVariables:
  \_int_var: (work@top.d), line:1:32, endln:1:33, parent:work@top
  |vpiParamAssign:
  \_param_assign: , line:2:18, endln:2:45, parent:work@top
    |vpiRhs:
    \_operation: , line:2:28, endln:2:45
      |vpiOpType:75
      |vpiReordered:1
      |vpiOperand:
      \_constant: , line:2:42, endln:2:44
        |vpiConstType:9
        |vpiDecompile:13
        |vpiSize:64
        |UINT:13
      |vpiOperand:
      \_constant: , line:2:38, endln:2:40
        |vpiConstType:9
        |vpiDecompile:12
        |vpiSize:64
        |UINT:12
      |vpiOperand:
      \_constant: , line:2:34, endln:2:36
        |vpiConstType:9
        |vpiDecompile:11
        |vpiSize:64
        |UINT:11
      |vpiOperand:
      \_constant: , line:2:30, endln:2:32
        |vpiConstType:9
        |vpiDecompile:10
        |vpiSize:64
        |UINT:10
    |vpiLhs:
    \_parameter: (work@top.x), line:2:18, endln:2:45, parent:work@top
      |vpiName:x
      |vpiFullName:work@top.x
      |vpiSize:1
      |vpiRange:
      \_range: , line:2:21, endln:2:24, parent:work@top.x
        |vpiLeftRange:
        \_constant: , line:2:21, endln:2:22
          |vpiConstType:9
          |vpiDecompile:3
          |vpiSize:64
          |UINT:3
        |vpiRightRange:
        \_constant: , line:2:23, endln:2:24
          |vpiConstType:9
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
      |vpiTypespec:
      \_int_typespec: (x), line:2:14, endln:2:17, parent:work@top.x
        |vpiName:x
  |vpiParameter:
  \_parameter: (work@top.x), line:2:18, endln:2:45, parent:work@top
```

alaindargelas, 2021-07-26:
When I take the following example in one of the commercial simulators on EdaPlayground, Surelog gives the correct answer:

```
module top();
   parameter int x [1:0] = '{10, 11};

   parameter a = x[0];
   parameter b = x[1];
   initial begin
     $display("a=%d", a);
     $display("b=%d", b);
   end
endmodule
```

Can you check your statement and compare with the commercial simulators?

RRozak, 2021-07-26:
Yes, in case of your example it gives the correct answer. In case of my it gives wrong.
I tried to find the differences between these two examples and the reason is the kind of assignment.
If you change your example to
```systemverilog
module top();
   parameter int x [1:0] = '{10, 11};

   int a, b;
   assign a = x[0];
   assign b = x[1];
   initial begin
     $display("a=%d", a);
     $display("b=%d", b);
   end
endmodule
```
it will give wrong values.

alaindargelas, 2021-07-27:
Fixed by #1548

closed 2021-07-27 by alaindargelas