chipsalliance/Surelog #1528: Different pattern operations give the same uhdm tree

RRozak, 2021-07-23:
These two files gives the same uhdm tree:
```systemverilog
module top(output logic[7:0] o);
   assign o = '{8:1};
endmodule
```
```systemverilog
module top(output logic[7:0] o);
   assign o = '{8{1}};
endmodule
```
Here is the uhdm:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module: work@top (work@top) index.sv:1:1: , endln:3:10, parent:work@top
  |vpiDefName:work@top
  |vpiFullName:work@top
  |vpiPort:
  \_port: (o), line:1:30, parent:work@top
    |vpiName:o
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@top.o), line:1:30, parent:work@top
        |vpiName:o
        |vpiFullName:work@top.o
        |vpiNetType:36
  |vpiContAssign:
  \_cont_assign: , line:2:11, endln:2:21, parent:work@top
    |vpiRhs:
    \_operation: , line:2:15, endln:2:21
      |vpiOpType:75
      |vpiOperand:
      \_tagged_pattern: 
        |vpiTypespec:
        \_integer_typespec: , line:2:17, endln:2:18
          |INT:8
        |vpiPattern:
        \_constant: , line:2:19, endln:2:20
          |vpiConstType:9
          |vpiDecompile:1
          |vpiSize:64
          |UINT:1
    |vpiLhs:
    \_ref_obj: (work@top.o), line:2:11, endln:2:12
      |vpiName:o
      |vpiFullName:work@top.o
  |vpiNet:
  \_logic_net: (work@top.o), line:1:30, parent:work@top
|uhdmtopModules:
\_module: work@top (work@top) index.sv:1:1: , endln:3:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiPort:
  \_port: (o), line:1:30, endln:1:31, parent:work@top
    |vpiName:o
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@top.o), line:1:30, endln:1:31, parent:work@top
        |vpiName:o
        |vpiFullName:work@top.o
        |vpiNetType:36
        |vpiRange:
        \_range: , line:1:25, endln:1:28
          |vpiLeftRange:
          \_constant: , line:1:25, endln:1:26
            |vpiConstType:9
            |vpiDecompile:7
            |vpiSize:64
            |UINT:7
          |vpiRightRange:
          \_constant: , line:1:27, endln:1:28
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
  |vpiNet:
  \_logic_net: (work@top.o), line:1:30, endln:1:31, parent:work@top
```
The only differences between uhdm of file 1 and file 2 are column numbers.
Maybe the operation in 2nd file should be parsed as vpiMultiConcatOp, but I'm not sure.
If both of them are correct, could you add a field to distinguish them?

alaindargelas, 2021-07-29:
Fixed by #1564 to the first order, recursive patterns are not working

closed 2021-07-29 by alaindargelas

RRozak, 2021-07-29:
I didn't know it when I was creating that issue, but now I noticed that in case of second file the operation should have type `vpiMultiAssignmentPatternOp`. The similar example is described in 37.57.7.

alaindargelas, 2021-07-29:
Can you submit a change list based on the chanhe I just made? It should be easy to change the 2 operations back to 1 operation with the new vpioptype you mention

RRozak, 2021-07-30:
Ok, I created the PR: https://github.com/chipsalliance/Surelog/pull/1571.