chipsalliance/Surelog #1538: [ERR:UH0703] Internal Error: adding wrong object type (ref_obj) in a property_expr_group group!

pieter3d, 2021-07-26:
I receive the following error when parsing the [SweRV EH1](https://github.com/chipsalliance/Cores-SweRV) core:
`[ERR:UH0703] Internal Error: adding wrong object type (ref_obj) in a property_expr_group group!`

Steps to repro:

1. `git clone https://github.com/chipsalliance/Cores-SweRV.git`
2. `cd Cores-SweRV`
3. `export RV_ROOT=$PWD`
4. Generate a valid config: `./configs/swerv.config`
5. The SweRV core project uses generated defines that are not explicitly included in each file, and I don't see a way for Surelog to read a file full of  `` `define``s and apply that to all files (separate issue?). As a workaround, prepend an explicit include to each design file:
``find design/ -name '*.sv' -exec sed -i '1s/^/`include "common_defines.vh"\n/' {} \;``
``sed -i '1s/^/`include "common_defines.vh"\n/' design/include/global.h``
6. Parse with Surelog's helloworld using the included [surelog.f.txt](https://github.com/chipsalliance/Surelog/files/6875410/surelog.f.txt) file:
`hellosureworld -parse -f surelog.f.txt | grep ERR`

alaindargelas, 2021-07-26:
Hi there,
first of all, Surelog can parse both Verilog semantics: 
- The interpreted mode for which a define is valid from the time it is defined until the end of compilation or when it is undefined
- The fileunit compilation where each file only sees the define included in that file

By default, Surelog does compile like the interpreted mode, you need to specify -fileunit option to compile each source file as an independent file unit. 

Please check which Verilog semantic you want and command line option you are using,

pieter3d, 2021-07-26:
Ok - I can look into that separately. I did try to add the .vh file as a source file at the start of the .f without `-fileunit`, but it didn't seem to make any difference.

But anyway, for the purposes of this issue, the internal error is problem at hand.

alaindargelas, 2021-07-26:
I see.
Properties are only handled in a primitive fashion in Surelog at this point. 
If your intent is to Synthesize/Simulate the design, then ignore the error. 
If your intent is to analyze the property in the design in some fashion then we need to finish the property modeling in Surelog/UHDM.
Please specify your intent.

pieter3d, 2021-07-26:
Ok thanks for the context. I don't need to do anything with properties, just trying to browse the elaborated synthesizable design.
Is there a way for the API to expose the fact that an internal error is due to the limited property support?  I'd like to safely programmatically ignore this kind of error.

alaindargelas, 2021-07-26:
No way for the API to tell you that the modeling is unfinished business. What you get is the error you see.
Non-Synthesizable constructs like Constraints, Properties, Sequences, Classes are only very partially modeled in Surelog at this point, so they'll give out these errors or no errors (silently mishandled), until someone gets to finish that work. Note that all the constructs I just mentioned are actually parsed, but not modeled in UHDM.

alaindargelas, 2021-07-26:
The sure way to get rid of the error is to complete the modeling.

closed 2021-07-28 by alaindargelas