chipsalliance/Surelog #1540: Declaration in for loop treated as assignment

RRozak, 2021-07-26:
Declaration of variable `a` in 4th line of code is treated as assignment statement.
```systemverilog
module top(output logic o);
   function automatic logic theta();
      for (int x = 0 ; x < 5 ; x++) begin
	      int a;
      end
      return 1'b1;
   endfunction : theta

   assign o = theta();
endmodule : top
```
Here is part of uhdm tree:
```
      |vpiStmt:
      \_for_stmt: (work@top.theta), line:3:7, endln:3:10, parent:work@top.theta
        |vpiFullName:work@top.theta
        |vpiCondition:
        \_operation: , line:3:24, endln:3:25, parent:work@top.theta
          |vpiOpType:20
          |vpiOperand:
          \_ref_obj: (work@top.theta.x), line:3:24, endln:3:25
            |vpiName:x
            |vpiFullName:work@top.theta.x
          |vpiOperand:
          \_constant: , line:3:28, endln:3:29
            |vpiConstType:9
            |vpiDecompile:5
            |vpiSize:64
            |UINT:5
        |vpiForInitStmt:
        \_assign_stmt: , parent:work@top.theta
          |vpiRhs:
          \_constant: , line:3:20, endln:3:21
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
          |vpiLhs:
          \_int_var: (work@top.theta.x), line:3:12, endln:3:15
            |vpiName:x
            |vpiFullName:work@top.theta.x
        |vpiForIncStmt:
        \_operation: , line:3:32, endln:3:33, parent:work@top.theta
          |vpiOpType:62
          |vpiOperand:
          \_ref_obj: (x), line:3:32, endln:3:33
            |vpiName:x
        |vpiStmt:
        \_begin: (work@top.theta), line:3:37, endln:5:10, parent:work@top.theta
          |vpiFullName:work@top.theta
          |vpiStmt:
          \_assign_stmt: , parent:work@top.theta
            |vpiLhs:
            \_int_var: (work@top.theta.a), line:4:3, endln:4:6
              |vpiName:a
              |vpiFullName:work@top.theta.a
      |vpiStmt:
      \_return_stmt: , line:6:7, endln:6:13, parent:work@top.theta
        |vpiCondition:
        \_constant: , line:6:14, endln:6:18
          |vpiConstType:3
          |vpiDecompile:1'b1
          |vpiSize:1
          |BIN:1
```

alaindargelas, 2021-07-26:
That is the only way to model it. There is no declaration model in VPI , so the convention I took is the assign statement with no rhs

RRozak, 2021-07-27:
In 37.32 is said that `vpiFor` node can have `vpiLocalVarDecls` field. And if so, the for statement is a scope. (37.12.2), so declared variables can be handled in the same way as in function.

alaindargelas, 2021-07-28:
You are correct, fixed with #1556

closed 2021-07-28 by alaindargelas