chipsalliance/Surelog #1568: vpiParent for bit_select vs. part_select

hs-apotell, 2021-07-29:
`vpiParent` pointer for `part_select` is non-null and points to parent `net`. However, `bit_select`'s `vpiParent` is always null. Is this by design or an oversight/bug?

```
module select;
  reg [3:0] a;
 
  initial begin
    a[0] = 1;
    a[1:0] = a[2:0];
  end
endmodule
```

```
....
        \_bit_select: (work@select.a), line:5:5, endln:5:6
          |vpiName:a
          |vpiFullName:work@select.a
          |vpiIndex:
          \_constant: , line:5:7, endln:5:8, parent:work@select.a
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0

....
        \_part_select: , line:6:5, endln:6:6, parent:work@select.a
          |vpiConstantSelect:1
          |vpiParent:
          \_ref_obj: (work@select.a)
          |vpiLeftRange:
          \_constant: , line:6:7, endln:6:8
            |vpiConstType:9
            |vpiDecompile:1
            |vpiSize:64
            |UINT:1
          |vpiRightRange:
          \_constant: , line:6:9, endln:6:10
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
        |vpiRhs:
        \_part_select: , line:6:14, endln:6:20, parent:work@select.a
          |vpiConstantSelect:1
          |vpiParent:
          \_ref_obj: (work@select.a)
          |vpiLeftRange:
          \_constant: , line:6:16, endln:6:17
            |vpiConstType:9
            |vpiDecompile:2
            |vpiSize:64
            |UINT:2
          |vpiRightRange:
          \_constant: , line:6:18, endln:6:19
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
....
```

hs-apotell, 2021-07-29:
As it turns out, the `part_select`'s `vpiParent` pointer is valid (i.e. a ref_obj) but the corresponding `actual_group` is still null. This sure looks like a bug.

alaindargelas, 2021-07-31:
Fixed by #1577, introduced a ref_obj as the parent of bit_select to allow binding.

closed 2021-07-31 by alaindargelas