chipsalliance/Surelog #1595: Support Surelog compilation and verilator uhdm feature on MACOS

EngRaff92, 2021-08-02:
Hello all, I'm posting here the issue I'm facing on MACOS. It seems to be a problem of libraries used there (-lrt not available on MACOS) 

https://github.com/chipsalliance/UHDM-integration-tests/issues/387

I will dive deeper to understand more any support, suggestion and help would be really appreciated since I'm not an expert on doing this kind of stuff honestly speaking :) 

regards and thanks in advances

EngRaff92, 2021-08-02:
seems to work now:

117 #LIBS += -luhdm -lcapnp -lkj -ldl -lutil -lm -lrt -lpthread
118 LIBS += -luhdm -lcapnp -lkj -ldl -lutil -lm -lpthread

lines in the ~/verilator/src/Makefile_obj.in

It builds everything and creates the Verilator_bin

I'm still getting issues in having a fully working OOP SV testbench.

1. Programs I think are not fully supported in the TB_TOP module
2. we cannot assign the interface to a program or so
3. $dumpafile or $dumpavrs seems causing a VPI bad handle

Regards

hzeller, 2021-08-02:
It would probably be a good idea to add building and testing this in the CI on the MacOS platform, so that this serves not only as a way to make sure things will work there but as well serves as documentation e.g. special linker flags to be set.

EngRaff92, 2021-08-02:
Sorry for the dump question, what is a CI and how can i update it ?

hzeller, 2021-08-02:
ah sorry, CI is the 'continuous integration', a script that runs on every commit. They run on github and are configured in a so-called workflow file
https://github.com/chipsalliance/Surelog/blob/master/.github/workflows/main.yml

Here is an example how things look like https://github.com/chipsalliance/Surelog/runs/3217128373  <- there you see, that currently the windows-related builds are actually broken.
In the configuration you can choose the platform things are running on (study the above main.yml) and one of the choices is to use MacOS. Some examples I know of are e.g. [this](https://github.com/chipsalliance/verible/blob/master/.github/workflows/macos-compile.yml) or [this](https://github.com/hzeller/timg/blob/main/.github/workflows/macos.yml).

So, updating it for this case would be to add a section in the main.yml to run the bulid and tests on MacOS. You can update the file, create a pull request and then actually see the result of that in the pull request itself - it shows links to the action that have been run and if they are successful. Usually,  things won't work right away, but you can update the pull request with new commits until the test runs pass.

alaindargelas, 2021-08-20:
@EngRaff92, looks like #1679 by @rkapuscik is fixing this Issue, thanks @rkapuscik!
Can you confirm it works for you? And close the issue if it does.

rkapuscik, 2021-08-20:
Just to be clear - I added CI builds for Surelog as Alain mentioned and there is work in progress on Verilator here: https://github.com/antmicro/verilator/pull/436

Regarding Your points, @EngRaff92 - while we have good progress on Verilator simulation (like supporting Ibex core), we are still working towards full SV support.
It would be great if you could provide a small example based on your design that is not working in Verilator, preferably with UHDM tree obtained with `--dump-uhdm`, in a separate issue in [UHDM-Verilator repository](https://github.com/antmicro/verilator). It would then be easier to focus on the specific issue and find out if it needs to be fixed in Verilator (I can see that e.g. support for `$dumpfile` and `$dumpvars` needs to be added), in Surelog or in both tools.

alaindargelas, 2021-08-25:
Closing as the CI supports the MacOs build.

closed 2021-08-25 by alaindargelas

EngRaff92, 2021-08-25:
First off i wanna thank you all for the support and so sorry if i did not reply on time (extremely busy week at work). To recap basically after cloning it again i spotted the changes in the libraries and it worked straight after so thanks for the fix @rkapuscik.

Regarding the features not yet supported i will provide a simple example of errors I got. I will open a separate thread on UHDM-Verilator repository adding the --dump-uhdm option. Thanks again guys for the hard work