chipsalliance/Surelog #1597: Indexes are duplicated when --disable-feature=parametersubstitution flag is used

RRozak, 2021-08-02:
When the following file is parsed by Surelog:
```systemverilog
module top(output int a, b, c, d);
   parameter int P [1:0][2:0][3:0] = '{
				       '{'{0, 1, 2, 3},
					 '{10, 11, 12, 13},
					 '{20, 21, 22, 23}},
				       '{'{100, 101, 102, 103},
					 '{110, 111, 112, 113},
					 '{120, 121, 122, 123}}
				       };
   
   assign a = P[0][0][2];
endmodule
```
The last 2 indexes in expression `assign a = P[0][0][2]` are duplicated.
Below is uhdm tree:
```
build/top.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module: work@top (work@top) /home/rrozak/Documents/verilator/uhdm-integration/tests/ParameterUnpackedArray/top.sv:1:1: , endln:12:10, parent:work@top
  |vpiDefName:work@top
  |vpiFullName:work@top
  |vpiPort:
  \_port: (a), line:1:23, parent:work@top
    |vpiName:a
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@top.a), line:1:23, parent:work@top
        |vpiName:a
        |vpiFullName:work@top.a
  |vpiPort:
  \_port: (b), line:1:26, parent:work@top
    |vpiName:b
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@top.b), line:1:26, parent:work@top
        |vpiName:b
        |vpiFullName:work@top.b
  |vpiPort:
  \_port: (c), line:1:29, parent:work@top
    |vpiName:c
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@top.c), line:1:29, parent:work@top
        |vpiName:c
        |vpiFullName:work@top.c
  |vpiPort:
  \_port: (d), line:1:32, parent:work@top
    |vpiName:d
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@top.d), line:1:32, parent:work@top
        |vpiName:d
        |vpiFullName:work@top.d
  |vpiContAssign:
  \_cont_assign: , line:11:11, endln:11:25, parent:work@top
    |vpiRhs:
    \_var_select: (work@top.P), line:11:15, endln:11:25
      |vpiName:P
      |vpiFullName:work@top.P
      |vpiIndex:
      \_constant: , line:11:17, endln:11:18, parent:work@top.P
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiIndex:
      \_constant: , line:11:20, endln:11:21, parent:work@top.P
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiIndex:
      \_constant: , line:11:23, endln:11:24, parent:work@top.P
        |vpiConstType:9
        |vpiDecompile:2
        |vpiSize:64
        |UINT:2
      |vpiIndex:
      \_constant: , line:11:20, endln:11:21, parent:work@top.P
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiIndex:
      \_constant: , line:11:23, endln:11:24, parent:work@top.P
        |vpiConstType:9
        |vpiDecompile:2
        |vpiSize:64
        |UINT:2
    |vpiLhs:
    \_ref_obj: (work@top.a), line:11:11, endln:11:12
      |vpiName:a
      |vpiFullName:work@top.a
  |vpiNet:
  \_logic_net: (work@top.a), line:1:23, parent:work@top
  |vpiNet:
  \_logic_net: (work@top.b), line:1:26, parent:work@top
  |vpiNet:
  \_logic_net: (work@top.c), line:1:29, parent:work@top
  |vpiNet:
  \_logic_net: (work@top.d), line:1:32, parent:work@top
...
```

I used Surelog with flags `-parse -nobuiltin --disable-feature=parametersubstitution`

tgorochowik, 2021-08-02:
Possibly the same issue as #1596.

alaindargelas, 2021-08-03:
Fixed by #1599

closed 2021-08-03 by alaindargelas