chipsalliance/Surelog #1601: Reversed order of elements of packed array parameter in assignment

RRozak, 2021-08-03:
When the following file is parsed by Surelog:
```systemverilog
module top(output logic[3:0] a);
   parameter logic [0:0][3:0] P = '{'{1, 0, 0, 0}};

   assign a = P[0];
endmodule
```
the order of elements assigned to `a` is reversed.
Here is uhdm tree:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module: work@top (work@top) tests/ParameterPackedArray/top.sv:1:1: , endln:5:10, parent:work@top
  |vpiDefName:work@top
  |vpiFullName:work@top
  |vpiPort:
  \_port: (a), line:1:30, parent:work@top
    |vpiName:a
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@top.a), line:1:30, parent:work@top
        |vpiName:a
        |vpiFullName:work@top.a
        |vpiNetType:36
  |vpiContAssign:
  \_cont_assign: , line:4:11, endln:4:19, parent:work@top
    |vpiRhs:
    \_operation: , line:4:15, endln:4:19
      |vpiOpType:75
      |vpiReordered:1
      |vpiOperand:
      \_constant: , line:2:48, endln:2:49
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiOperand:
      \_constant: , line:2:45, endln:2:46
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiOperand:
      \_constant: , line:2:42, endln:2:43
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiOperand:
      \_constant: , line:2:39, endln:2:40
        |vpiConstType:9
        |vpiDecompile:1
        |vpiSize:64
        |UINT:1
    |vpiLhs:
    \_ref_obj: (work@top.a), line:4:11, endln:4:12
      |vpiName:a
      |vpiFullName:work@top.a
  |vpiNet:
  \_logic_net: (work@top.a), line:1:30, parent:work@top
  |vpiParamAssign:
  \_param_assign: , line:2:31, endln:2:51, parent:work@top
    |vpiRhs:
    \_operation: , line:2:35, endln:2:51
      |vpiOpType:75
      |vpiOperand:
      \_operation: , line:2:37, endln:2:50
        |vpiOpType:75
        |vpiReordered:1
        |vpiOperand:
        \_constant: , line:2:48, endln:2:49
          |vpiConstType:9
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
        |vpiOperand:
        \_constant: , line:2:45, endln:2:46
          |vpiConstType:9
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
        |vpiOperand:
        \_constant: , line:2:42, endln:2:43
          |vpiConstType:9
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
        |vpiOperand:
        \_constant: , line:2:39, endln:2:40
          |vpiConstType:9
          |vpiDecompile:1
          |vpiSize:64
          |UINT:1
    |vpiLhs:
    \_parameter: (work@top.P), line:2:31, endln:2:51, parent:work@top
      |vpiName:P
      |vpiFullName:work@top.P
      |vpiTypespec:
      \_logic_typespec: (P), line:2:14, endln:2:19, parent:work@top.P
        |vpiName:P
        |vpiRange:
        \_range: , line:2:21, endln:2:24
          |vpiLeftRange:
          \_constant: , line:2:21, endln:2:22
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
          |vpiRightRange:
          \_constant: , line:2:23, endln:2:24
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
        |vpiRange:
        \_range: , line:2:26, endln:2:29
          |vpiLeftRange:
          \_constant: , line:2:26, endln:2:27
            |vpiConstType:9
            |vpiDecompile:3
            |vpiSize:64
            |UINT:3
          |vpiRightRange:
          \_constant: , line:2:28, endln:2:29
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
  |vpiParameter:
  \_parameter: (work@top.P), line:2:31, endln:2:51, parent:work@top
|uhdmtopModules:
\_module: work@top (work@top) tests/ParameterPackedArray/top.sv:1:1: , endln:5:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiPort:
  \_port: (a), line:1:30, endln:1:31, parent:work@top
    |vpiName:a
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@top.a), line:1:30, endln:1:31, parent:work@top
        |vpiName:a
        |vpiFullName:work@top.a
        |vpiNetType:36
        |vpiRange:
        \_range: , line:1:25, endln:1:28
          |vpiLeftRange:
          \_constant: , line:1:25, endln:1:26
            |vpiConstType:9
            |vpiDecompile:3
            |vpiSize:64
            |UINT:3
          |vpiRightRange:
          \_constant: , line:1:27, endln:1:28
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
  |vpiNet:
  \_logic_net: (work@top.a), line:1:30, endln:1:31, parent:work@top
  |vpiParamAssign:
  \_param_assign: , line:2:31, endln:2:51, parent:work@top
    |vpiRhs:
    \_operation: , line:2:35, endln:2:51
      |vpiOpType:75
      |vpiOperand:
      \_operation: , line:2:37, endln:2:50
        |vpiOpType:75
        |vpiReordered:1
        |vpiOperand:
        \_constant: , line:2:48, endln:2:49
          |vpiConstType:9
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
        |vpiOperand:
        \_constant: , line:2:45, endln:2:46
          |vpiConstType:9
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
        |vpiOperand:
        \_constant: , line:2:42, endln:2:43
          |vpiConstType:9
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
        |vpiOperand:
        \_constant: , line:2:39, endln:2:40
          |vpiConstType:9
          |vpiDecompile:1
          |vpiSize:64
          |UINT:1
    |vpiLhs:
    \_parameter: (work@top.P), line:2:31, endln:2:51, parent:work@top
      |vpiName:P
      |vpiFullName:work@top.P
      |vpiTypespec:
      \_logic_typespec: (P), line:2:14, endln:2:19, parent:work@top.P
        |vpiName:P
        |vpiRange:
        \_range: , line:2:21, endln:2:24, parent:P
          |vpiLeftRange:
          \_constant: , line:2:21, endln:2:22
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
          |vpiRightRange:
          \_constant: , line:2:23, endln:2:24
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
        |vpiRange:
        \_range: , line:2:26, endln:2:29, parent:P
          |vpiLeftRange:
          \_constant: , line:2:26, endln:2:27
            |vpiConstType:9
            |vpiDecompile:3
            |vpiSize:64
            |UINT:3
          |vpiRightRange:
          \_constant: , line:2:28, endln:2:29
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
  |vpiParameter:
  \_parameter: (work@top.P), line:2:31, endln:2:51, parent:work@top
```

alaindargelas, 2021-08-07:
For
```
module top(output logic[3:0] a);
  parameter logic [0:0][3:0] P = '{'{1, 0, 0, 0}};

   assign a = P[0];
  initial $display("a: %d\n", a);
endmodule
```

Any simulator will output: a: 8.

Surelog makes an array of logic values, the 3rd element in the array is 1,   2^3 = 8, Surelog is correct.

RRozak, 2021-08-09:
Yes, the output is correct. But look at this assignment part of the uhdm above):
```
  |vpiContAssign:
  \_cont_assign: , line:4:11, endln:4:19, parent:work@top
    |vpiRhs:
    \_operation: , line:4:15, endln:4:19
      |vpiOpType:75
      |vpiReordered:1
      |vpiOperand:
      \_constant: , line:2:48, endln:2:49
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiOperand:
      \_constant: , line:2:45, endln:2:46
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiOperand:
      \_constant: , line:2:42, endln:2:43
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiOperand:
      \_constant: , line:2:39, endln:2:40
        |vpiConstType:9
        |vpiDecompile:1
        |vpiSize:64
        |UINT:1
    |vpiLhs:
    \_ref_obj: (work@top.a), line:4:11, endln:4:12
      |vpiName:a
      |vpiFullName:work@top.a
```
The assigned array is reversed.

Please consider the following file:
```systemverilog
module top(output logic [3:0] a);
   parameter logic [0:0][3:0] P = '{'{1, 0, 0, 0}};
   logic [3:0] b;
   assign a = P[0];
   assign b = '{1, 0, 0, 0};
endmodule
```
`a` and `b` should be equal, but in uhdm one of them has reversed array assigned.
```
  |vpiContAssign:
  \_cont_assign: , line:6:11, endln:6:19, parent:work@top
    |vpiRhs:
    \_operation: , line:6:15, endln:6:19
      |vpiOpType:75
      |vpiReordered:1
      |vpiOperand:
      \_constant: , line:4:48, endln:4:49
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiOperand:
      \_constant: , line:4:45, endln:4:46
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiOperand:
      \_constant: , line:4:42, endln:4:43
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiOperand:
      \_constant: , line:4:39, endln:4:40
        |vpiConstType:9
        |vpiDecompile:1
        |vpiSize:64
        |UINT:1
    |vpiLhs:
    \_ref_obj: (work@top.a), line:6:11, endln:6:12
      |vpiName:a
      |vpiFullName:work@top.a
  |vpiContAssign:
  \_cont_assign: , line:7:11, endln:7:28, parent:work@top
    |vpiRhs:
    \_operation: , line:7:15, endln:7:28
      |vpiOpType:75
      |vpiOperand:
      \_constant: , line:7:17, endln:7:18
        |vpiConstType:9
        |vpiDecompile:1
        |vpiSize:64
        |UINT:1
      |vpiOperand:
      \_constant: , line:7:20, endln:7:21
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiOperand:
      \_constant: , line:7:23, endln:7:24
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiOperand:
      \_constant: , line:7:26, endln:7:27
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
    |vpiLhs:
    \_ref_obj: (work@top.b), line:7:11, endln:7:12
      |vpiName:b
      |vpiFullName:work@top.b
```

I used flags `-parse -nobuiltin`.

alaindargelas, 2021-08-10:
Before we switch to another example that should be in a different Issue altogether, I'll close this one, then you can open a different issue with the new case.

So let's get back to the original case, it is not reversed!

```
module top(output logic[3:0] a);
   parameter logic [0:0][3:0] P = '{'{1, 0, 0, 0}};
                                                      ^_ this bit is the MSB since the vector has a range that is in the form:  [MSB:LSB]
   assign a = P[0];
endmodule
```
```
|vpiContAssign:
  \_cont_assign: , line:4:11, endln:4:19, parent:work@top
    |vpiRhs:
    \_operation: , line:4:15, endln:4:19
      |vpiOpType:75
      |vpiReordered:1           <<<< This property means that the vector was reordered to match the Lhs [MSB:LSB] situation.
                                                    If the Lhs was [LSB:MSB], then the vector would not have been reordered.
      |vpiOperand:
      \_constant: , line:2:48, endln:2:49
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0                            <----------------- 0th index bit of the vector  (LSB)
      |vpiOperand:
      \_constant: , line:2:45, endln:2:46
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiOperand:
      \_constant: , line:2:42, endln:2:43
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiOperand:
      \_constant: , line:2:39, endln:2:40
        |vpiConstType:9
        |vpiDecompile:1
        |vpiSize:64
        |UINT:1                             <----------------- 3rd index bit of the vector  (MSB)
```

closed 2021-08-10 by alaindargelas

RRozak, 2021-08-10:
Ok, you're right.
I didn't notice that field.