chipsalliance/Surelog #1614: Array-instances of modules naming doesn't match verilator's VCDs

pieter3d, 2021-08-07:
When a module is instanced multiple times:
```verilog
flop flop_instances[3:0] (
  .clk (clk),
  .d (d),
  .q (q)
);
```
Surelog names these instances `flop_instances0 flop_instances1, ...`, but verilator puts parentheses around the instance number when writing VCD files.

It would be nice if they matched, seeing as they should be compatible within the chipsalliance projects.

alaindargelas, 2021-08-07:
flop_instances(0) ?
What we really need is what the Standard says, can you check the commercial simulators in EdaPlayground?

pieter3d, 2021-08-07:
That's what I figured, but I couldn't find any language in the IEEE1800-2017 standard that says what it should be named. This is unlike the generate blocks with `genblk0` etc.
I checked both vcs and metrics, they use square brackets, `flop_instances[0]`. I have also seen tools use square brackets and  also prepend the escape character `\flop_instances[0]` since those brackets are technically not legal identifier characters if they were to appear in manually unrolled source code. So it's not totally consistent, but having surelog and verilator both use square brackets is probably closer to what we want.

alaindargelas, 2021-08-07:
Fixed by #1622

closed 2021-08-07 by alaindargelas

pieter3d, 2021-08-07:
I see you used the leading escape. Section 5.6.1 of the IEEE1800 spec says "Neither the leading backslash character nor the terminating white space is considered to be part of the identifier", but it would be there in source code. So I don't know if the UHDM should include the backslash or not. What do you think?

pieter3d, 2021-08-07:
Also - would it make sense to file a bug against verilator to match it? I imagine that could be pretty disruptive.

alaindargelas, 2021-08-07:
I have 0 preference, these are tool arbitrary choices, I'd rather we stay close to the consensus from commercial simulators and get verilator to follow.Can you submit a put request with the naming scheme you prefer and update the logs?Modify the same code I changed,
makemake regressioncd build../tests/regression.tcl update

pieter3d, 2021-08-07:
Will do.

pieter3d, 2021-08-08:
Hmm - When I run update, tons of the logs get polluted with my local
working directory, e.g.:
\_class_defn: ***@***.***) */home/pieter/work/pieter3*
${SURELOG_DIR}/build/bin/sv/builtin.sv:4:1: