chipsalliance/Surelog #1624: Escaped identifiers don't seem supported.

pieter3d, 2021-08-08:
Looks like there are syntax errors in this case. E.g.:
```
[SNT:PA0207] hw/primitives/sync2.sv:36:6: Syntax error: token recognition error at: '\',
logic \4e^sca'd|signal;
      ^-- ./slpp_all/work/primitives_8418434797781237852/sync2.sv:36:6:.
```
See section 5.6.1 in the IEEE1800 standard. Additionaly per that section, I would think the surelog vpiName should not include the leading backslash (not sure if would).

A cursory glance in the grammar files shows support for it, but I don't know enough about Surelog's parser design to root cause this quickly.

alaindargelas, 2021-08-08:
You are missing the space prior to the semicolumn

alaindargelas, 2021-08-08:
The escaped name with no space at the end is lexically illegal. The parser's lexer rightfully point that out. Include the full test to see if the preprocessor swallowed the space by mistake.

pieter3d, 2021-08-08:
Hmm - still see this:
```
[SNT:PA0207] hw/primitives/sync2.sv:36:6: Syntax error: token recognition error at: '\',
logic \223897d89h297d97237237&*%&*^Sa78d68h ;
      ^-- ./slpp_all/work/primitives_8418434797781237852/sync2.sv:36:6:.
```

Verilator doesn't complain.

alaindargelas, 2021-08-08:
#1627 gives more priority to escape name tokenization in the preprocessor lexer and gives also more priority in the parser.
That didn't make a difference for the complete sequence though: logic \223897d89h297d97237237&*%&*^Sa78d68h ;

Sub-sequences of the example work, can't find a true rationale, but it certainly has to do with the special characters.
I won't spend more time on this, you are welcome to analyze which token (maybe the "Special") in SV3_1aPpLexer.g4 is interfering with the "Escaped_identifier"

alaindargelas, 2021-08-23:
Closing if no more activity on this issue.

closed 2021-08-23 by alaindargelas