chipsalliance/Surelog #1633: Problems with parameter substitution

RRozak, 2021-08-09:
There are problems with parameter substitution in the following file:
```systemverilog
module prim_subreg #(
  parameter logic [31:0] RESVAL   = '0
) ();
endmodule

module prim_subreg_shadow #(
  parameter logic [31:0] RESVAL   = '0
) ();

  prim_subreg #(
    .RESVAL(~RESVAL)
  ) staged_reg();

endmodule
```
The output uhdm:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@prim_subreg_shadow)
|vpiName:work@prim_subreg_shadow
|uhdmallModules:
\_module: work@prim_subreg (work@prim_subreg) prim_subreg_simpler.sv:1:1: , endln:4:10, parent:work@prim_subreg_shadow
  |vpiDefName:work@prim_subreg
  |vpiFullName:work@prim_subreg
  |vpiParamAssign:
  \_param_assign: , line:2:26, endln:2:39, parent:work@prim_subreg
    |vpiRhs:
    \_constant: , line:2:37, endln:2:39
      |vpiConstType:3
      |vpiDecompile:'0
      |vpiSize:-1
      |BIN:0
    |vpiLhs:
    \_parameter: (work@prim_subreg.RESVAL), line:2:26, endln:2:39, parent:work@prim_subreg
      |vpiName:RESVAL
      |vpiFullName:work@prim_subreg.RESVAL
      |BIN:0
      |vpiTypespec:
      \_logic_typespec: (RESVAL), line:2:13, endln:2:18, parent:work@prim_subreg.RESVAL
        |vpiName:RESVAL
        |vpiRange:
        \_range: , line:2:20, endln:2:24
          |vpiLeftRange:
          \_constant: , line:2:20, endln:2:22
            |vpiConstType:9
            |vpiDecompile:31
            |vpiSize:64
            |UINT:31
          |vpiRightRange:
          \_constant: , line:2:23, endln:2:24
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
  |vpiParameter:
  \_parameter: (work@prim_subreg.RESVAL), line:2:26, endln:2:39, parent:work@prim_subreg
|uhdmallModules:
\_module: work@prim_subreg_shadow (work@prim_subreg_shadow) prim_subreg_simpler.sv:6:1: , endln:14:10, parent:work@prim_subreg_shadow
  |vpiDefName:work@prim_subreg_shadow
  |vpiFullName:work@prim_subreg_shadow
  |vpiParamAssign:
  \_param_assign: , line:7:26, endln:7:39, parent:work@prim_subreg_shadow
    |vpiRhs:
    \_constant: , line:7:37, endln:7:39
      |vpiConstType:3
      |vpiDecompile:'0
      |vpiSize:-1
      |BIN:0
    |vpiLhs:
    \_parameter: (work@prim_subreg_shadow.RESVAL), line:7:26, endln:7:39, parent:work@prim_subreg_shadow
      |vpiName:RESVAL
      |vpiFullName:work@prim_subreg_shadow.RESVAL
      |BIN:0
      |vpiTypespec:
      \_logic_typespec: (RESVAL), line:7:13, endln:7:18, parent:work@prim_subreg_shadow.RESVAL
        |vpiName:RESVAL
        |vpiRange:
        \_range: , line:7:20, endln:7:24
          |vpiLeftRange:
          \_constant: , line:7:20, endln:7:22
            |vpiConstType:9
            |vpiDecompile:31
            |vpiSize:64
            |UINT:31
          |vpiRightRange:
          \_constant: , line:7:23, endln:7:24
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
  |vpiParameter:
  \_parameter: (work@prim_subreg_shadow.RESVAL), line:7:26, endln:7:39, parent:work@prim_subreg_shadow
|uhdmtopModules:
\_module: work@prim_subreg_shadow (work@prim_subreg_shadow) prim_subreg_simpler.sv:6:1: , endln:14:10
  |vpiDefName:work@prim_subreg_shadow
  |vpiName:work@prim_subreg_shadow
  |vpiModule:
  \_module: work@prim_subreg (work@prim_subreg_shadow.staged_reg) prim_subreg_simpler.sv:10:3: , endln:12:18, parent:work@prim_subreg_shadow
    |vpiDefName:work@prim_subreg
    |vpiDefFile:prim_subreg_simpler.sv
    |vpiDefLineNo:1
    |vpiName:staged_reg
    |vpiFullName:work@prim_subreg_shadow.staged_reg
    |vpiInstance:
    \_module: work@prim_subreg_shadow (work@prim_subreg_shadow) prim_subreg_simpler.sv:6:1: , endln:14:10
    |vpiParamAssign:
    \_param_assign: , line:2:26, endln:2:39, parent:work@prim_subreg_shadow.staged_reg
      |vpiRhs:
      \_constant: , line:2:37, endln:2:39
        |vpiConstType:9
        |vpiDecompile:18446744073709551615
        |vpiSize:64
        |UINT:18446744073709551615
      |vpiLhs:
      \_parameter: (work@prim_subreg_shadow.staged_reg.RESVAL), line:2:26, endln:2:39, parent:work@prim_subreg_shadow.staged_reg
        |vpiName:RESVAL
        |vpiFullName:work@prim_subreg_shadow.staged_reg.RESVAL
        |BIN:0
        |vpiTypespec:
        \_logic_typespec: (RESVAL), line:2:13, endln:2:18, parent:work@prim_subreg_shadow.staged_reg.RESVAL
          |vpiName:RESVAL
          |vpiRange:
          \_range: , line:2:20, endln:2:24, parent:RESVAL
            |vpiLeftRange:
            \_constant: , line:2:20, endln:2:22
              |vpiConstType:9
              |vpiDecompile:31
              |vpiSize:64
              |UINT:31
            |vpiRightRange:
            \_constant: , line:2:23, endln:2:24
              |vpiConstType:9
              |vpiDecompile:0
              |vpiSize:64
              |UINT:0
    |vpiParameter:
    \_parameter: (work@prim_subreg_shadow.staged_reg.RESVAL), line:2:26, endln:2:39, parent:work@prim_subreg_shadow.staged_reg
  |vpiParamAssign:
  \_param_assign: , line:7:26, endln:7:39, parent:work@prim_subreg_shadow
    |vpiRhs:
    \_constant: , line:7:37, endln:7:39
      |vpiConstType:9
      |vpiDecompile:0
      |vpiSize:65535
      |UINT:0
    |vpiLhs:
    \_parameter: (work@prim_subreg_shadow.RESVAL), line:7:26, endln:7:39, parent:work@prim_subreg_shadow
      |vpiName:RESVAL
      |vpiFullName:work@prim_subreg_shadow.RESVAL
      |BIN:0
      |vpiTypespec:
      \_logic_typespec: (RESVAL), line:7:13, endln:7:18, parent:work@prim_subreg_shadow.RESVAL
        |vpiName:RESVAL
        |vpiRange:
        \_range: , line:7:20, endln:7:24, parent:RESVAL
          |vpiLeftRange:
          \_constant: , line:7:20, endln:7:22
            |vpiConstType:9
            |vpiDecompile:31
            |vpiSize:64
            |UINT:31
          |vpiRightRange:
          \_constant: , line:7:23, endln:7:24
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
  |vpiParameter:
  \_parameter: (work@prim_subreg_shadow.RESVAL), line:7:26, endln:7:39, parent:work@prim_subreg_shadow
```
1.  There are problems with sizes in `uhdmtopModules`. In first param assign node it have `vpiSize = 64`, the second time it has size `65535`. In uhdmallModules the size of that parameter is correct (`-1`).
2. `--disable-feature=parametersubstitution` option doesn't work. The output uhdm trees are the same regardless of that flag was used.
3. I think that line numbers in that subtree under `param_assign: , line:2:26, endln:2:39, parent:work@prim_subreg_shadow.staged_reg` are wrong.
I think it should point to line 11. I'm not sure about the convention, but that assignment is caused by line 11 and I think this line number would be more helpful in debugging.

alaindargelas, 2021-08-10:
Fixed by #1681

RRozak, 2021-08-10:
Now there is no problem with sizes, but there is problem with values. In first param_assign node in uhdmtopModules it should be `'1` or something like negation of `'0`. In the second param_assign node it should be `'0` instead of `0`.

What about points 2. and 3.? Are they bugs or not?

alaindargelas, 2021-08-11:
If you run this in a commercial simulator:
```
module prim_subreg #(
  parameter logic [63:0] RESVAL   = '0)();
  initial $display("RESVAL:%d", RESVAL);

endmodule

module prim_subreg_shadow #(
  parameter logic [63:0] RESVAL   = '0
) ();

  prim_subreg #(
    .RESVAL(~RESVAL)
  ) staged_reg();

endmodule
```
You'll see the output is: 
RESVAL:18446744073709551615 (Not 1 as you mention, but Surelog's output).
Note that Surelog does not do lhs-rhs word matching, so it always resolve to a 64 bit value when the size is not specified.

This is the bitwize inversion operator you are using, maybe you are mistaking it with the Boolean ! operator?
So, no there is no problem with the values.

"In the second param_assign node it should be '0 instead of 0": At that point, the value went through evaluation and is now in canonical form, so the ' was dropped as it would happen for other prefixes like 'h, 'b. This is cosmetic and in no way impact the analysis down the line. The size is maintained to -1.

#2, this parameter expansion option is limited to a very specific use case that does not apply here. I was against naming this option using a generic name when it was created, I knew it would create confusion for the users, please work with @kamilrakoczy on either changing the name or the help

#3 That is not a bug, nor a deviation of the standard (There is no standard for this) but a long project to implement.

closed 2021-08-11 by alaindargelas