chipsalliance/Surelog #1639: Add https://github.com/bombela/backward-cpp support to Surelog

alaindargelas, 2021-08-11:
@hzeller , @hs-apotell it would be a good to have an automatic stack trace produced when Surelog core dumps. Especially on Windows, during CI test, there are some tests failing with IEs.
This https://github.com/bombela/backward-cpp seems interresting.

hs-apotell, 2021-08-11:
Taking ownership of this. Will look into it sometime next week.

Few thoughts -
* Will bring it in as a third party dependency
* Surelog library cannot (and shouldn't) be dependent on this library. Only the binaries (i.e. executables) generated as part of the build should depend on it.

mithro, 2021-08-11:
I suggest looking at https://github.com/ianlancetaylor/libbacktrace

mithro, 2021-08-11:
> The libbacktrace library may be linked into a program or library and used to produce symbolic backtraces. Sample uses would be to print a detailed backtrace when an error occurs or to gather detailed profiling information. In general the functions provided by this library are async-signal-safe, meaning that they may be safely called from a signal handler.

> As of October 2020, libbacktrace supports ELF, PE/COFF, Mach-O, and XCOFF executables with DWARF debugging information. In other words, it supports GNU/Linux, *BSD, macOS, Windows, and AIX. The library is written to make it straightforward to add support for other object file and debugging formats.

alaindargelas, 2021-08-13:
@hs-apotell, agreed with your points.
Also, check https://github.com/ianlancetaylor/libbacktrace, it looks more portable than https://github.com/bombela/backward-cpp.

hs-apotell, 2021-08-14:
@alaindargelas @mithro Either libbacktrace doesn't support windows builds i.e. ones built with cl compiler and not the MSYS, or I just don't see how to make it work on windows. I don't see any instructions on how to build - neither for windows nor for any other platform. The only available build tool is autoconf and all scripts are unix/linux specific. I got it to build under unix and that works well. I tried making a simple cmake build script but even the code is using unix specific headers (pulls in `unistd.h` in a few places). This library is likely to need considerable effort to make it work across all supported platforms. Is there a strong preference to use libbacktrace over backward-cpp??

hs-apotell, 2021-08-14:
Here's an alternative that is supported across multiple platforms - https://chromium.googlesource.com/breakpad/breakpad

alaindargelas, 2021-08-14:
@hs-apotell, I don't have a preference, whichever library works on all OSs will works for me.

mithro, 2021-08-15:
Give that breakpad is used by Chrome and Firefox, it seems like a good choice?

mithro, 2021-08-15:
Looks like crashpad replaces breakpad?

mithro, 2021-08-15:
@umarcor - Any thoughts?

umarcor, 2021-08-16:
@mithro, not much. I know that libbacktrace is used by GHDL, and GHDL is built on Linux, macOS and Windows. However, I think that the feature is not being actually enabled/tested in the builds.

hs-apotell, 2021-08-16:
@umarcor I couldn't find any documentation on how to build libbacktrace for windows or for that matter for any platform whatsoever. Could you point to some reference where this library is actively builds and used across platforms.

umarcor, 2021-08-16:
@hs-apotell, see:

- https://packages.msys2.org/package/mingw-w64-x86_64-libbacktrace?repo=mingw64
- https://github.com/msys2/MINGW-packages/blob/master/mingw-w64-libbacktrace/PKGBUILD
- https://github.com/msys2/MINGW-packages/blob/master/mingw-w64-libbacktrace/libbacktrace.pc

However, as said, I did not use it and I don't think it's enabled in GHDL's builds. It's an optional feature.