chipsalliance/Surelog #1644: Unwanted pattern parameter substitution

RRozak, 2021-08-12:
`--disable-feature=parametersubstitution` option doesn't work in the following example:
```systemverilog
module dut(output logic [1:0] a);
   parameter logic [1:0] P = 0;
   assign a = P;
endmodule // dut

module top(output logic [1:0] o);
   parameter logic [1:0] X = '{0, 1};
   dut #(
      .P(~X)
   ) u_dut(
      .a(o)
   );
endmodule // top
```
Now `~X` is replaced  with
```
      |vpiRhs:
      \_operation: , line:9:10, endln:9:12
        |vpiOpType:4
        |vpiOperand:
        \_operation: , line:9:11, endln:9:12
          |vpiOpType:75
          |vpiReordered:1
          |vpiOperand:
          \_constant: , line:7:35, endln:7:36
            |vpiConstType:9
            |vpiDecompile:1
            |vpiSize:64
            |UINT:1
          |vpiOperand:
          \_constant: , line:7:32, endln:7:33
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
          |vpiTypespec:
          \_logic_typespec: (X), line:7:14, endln:7:19, parent:work@top.X
            |vpiName:X
            |vpiRange:
            \_range: , line:7:21, endln:7:24, parent:X
              |vpiLeftRange:
              \_constant: , line:7:21, endln:7:22
                |vpiConstType:9
                |vpiDecompile:1
                |vpiSize:64
                |UINT:1
              |vpiRightRange:
              \_constant: , line:7:23, endln:7:24
                |vpiConstType:9
                |vpiDecompile:0
                |vpiSize:64
                |UINT:0
```
I would expect vpiRefObj under negation node.
If it is intentional behavior, could you provide information where that option should work?

Here is uhdm:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module: work@dut (work@dut) negation_pattern_instance.sv:1:1: , endln:4:10, parent:work@top
  |vpiDefName:work@dut
  |vpiFullName:work@dut
  |vpiPort:
  \_port: (a), line:1:31, parent:work@dut
    |vpiName:a
[... 260 lines cut ...]
        |UINT:0
    |vpiLhs:
    \_parameter: (work@top.X), line:7:26, endln:7:37, parent:work@top
      |vpiName:X
      |vpiFullName:work@top.X
      |vpiTypespec:
      \_logic_typespec: (X), line:7:14, endln:7:19, parent:work@top.X
        |vpiName:X
        |vpiRange:
        \_range: , line:7:21, endln:7:24, parent:X
          |vpiLeftRange:
          \_constant: , line:7:21, endln:7:22
            |vpiConstType:9
            |vpiDecompile:1
            |vpiSize:64
            |UINT:1
          |vpiRightRange:
          \_constant: , line:7:23, endln:7:24
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
  |vpiParameter:
  \_parameter: (work@top.X), line:7:26, endln:7:37, parent:work@top
```

alaindargelas, 2021-08-23:
@RRozak, @kamilrakoczy as I mentioned before this option is wrongly named (the name was @kamilrakoczy's idea), it gives the bad impression that Surelog will prevent all type of parameter substitutions. 
The option only prevents one type of parameter substitution: if the parameter is of type pattern assigment.
Its only purpose was to circumvent a flaw in Verilator: https://github.com/chipsalliance/Surelog/issues/1013
This option is not intended to prevent all type of parameter substitution. You opened another case last week where I explaned that.
Is there a real problem with Verilator or Yosys you are trying to fix? If yes, reopen the issue, else put it to rest.

RRozak, 2021-08-23:
The parameter `X` value is a pattern assignment, so I think the option should work.
There is a problem with Verilator. Assignment pattern can't be underneath `not` operation. It is unsupported yet.

alaindargelas, 2021-08-24:
Fixed by #1702

closed 2021-08-24 by alaindargelas