chipsalliance/Surelog #1659: Inconsistent VpiListener callbacks

hs-apotell, 2021-08-15:
vpi_listener logic seems to be dependent on the 'type' of the `any` object when invoking callbacks. Shouldn't the callbacks always be for concrete implementation of the `any` object?

For instance, `cont_assign` holds a pointer to the rhs as an `expr` instance. This rhs can be a `logic_net`, an `operation` or something else. As it stand today, the callbacks received are `enterExpr` and `leaveExpr`. I expected a `enterLogic_net/leaveLogic_net` or `enterOperation/leaveOperation`  callbacks based on the concrete implementation of the object. Without this, I am forced to handle every concrete implementation type in each of its base class handler. The current implementation forces handling of say an `operation` in multiple different callbacks and the listener has to keep track of what the parent context is and have to deal with it accordingly, which isn't always trivial.

Here's a very simple program to demonstrate the issue. Two statements, one `cont_assign` and other `always`. Both have exactly the same expression on the rhs but the callbacks invoked are very different.

In short, all non-concrete class implementation should basically just call `listen_any` which in turn based on the UHDM type of the object, calls appropriate concrete listen_xxx function to invoke the corresponding callbacks.

P.S. Only the relevant callback events are reported here. The numbers in the brackets are the [line-start,column-start:line-end,column-end]

```
module Test (a, b, c, sum1, sum2);
input a, b, c;      // Input declaration
output sum1, sum2;  // Ouput declaration
wire a, b, c;       // Port Data types
wire sum1, sum2;    

assign sum1 = a + b + c;
always sum2 = a + b + c;

endmodule // End of Module Test
```
```
  enterDesign[0,0:0,0]
    enterModule[1,1:10,10]
      enterProcess_stmt[8,1:8,25]
        enterAssignment[8,8:8,24]
          enterExpr[8,8:8,12]
          leaveExpr[8,8:8,12]
          enterOperation[8,15:8,20]
            enterOperation[8,15:8,16]
              enterRef_obj[8,15:8,16]
                enterLogic_net[4,6:4,7]
                leaveLogic_net[4,6:4,7]
              leaveRef_obj[8,15:8,16]
              enterRef_obj[8,19:8,20]
                enterLogic_net[4,9:4,10]
                leaveLogic_net[4,9:4,10]
              leaveRef_obj[8,19:8,20]
            leaveOperation[8,15:8,16]
            enterRef_obj[8,23:8,24]
              enterLogic_net[4,12:4,13]
              leaveLogic_net[4,12:4,13]
            leaveRef_obj[8,23:8,24]
          leaveOperation[8,15:8,20]
        leaveAssignment[8,8:8,24]
      leaveProcess_stmt[8,1:8,25]
      enterCont_assign[7,8:7,24]
        enterExpr[7,15:7,20]
        leaveExpr[7,15:7,20]
        enterExpr[7,8:7,12]
        leaveExpr[7,8:7,12]
      leaveCont_assign[7,8:7,24]
    leaveModule[1,1:10,10]
  leaveDesign[0,0:0,0]
```

hs-apotell, 2021-08-17:
@alaindargelas Could you pen some thoughts/insight on this issue. Are we on to something or are we looking at it the wrong way. If this is an issue, do you think this can be resolved easily or this something that likely to put on back-burner. A change in this implementation is likely going to ripple through quite a few test cases.

alaindargelas, 2021-08-18:
@hs-apotell, I can change the call to be listen_any,  I think that will address your issue, but I need to add an argument like in vpi_visitor :  VisitedContainer* visited to prevent infinite loops.
Is that going to be a problem? You will probably have to change some code.

alaindargelas, 2021-08-18:
You can try to replace the attached version of model_gen.tcl under Surelog/thirdparty/UHDM/model_gen/

[model_gen.tcl.gz](https://github.com/chipsalliance/Surelog/files/7004540/model_gen.tcl.gz)

This version does not prevent infinite loops, but I first want to make sure that address your concern about concrete class callbacks

hs-apotell, 2021-08-18:
@alaindargelas Will revert back soon. Appreciate the prompt response.

hs-apotell, 2021-08-18:
Here's the output from the edited model_gen version you shared.
```
    enterModule[1,1:10,10]
      enterAlways[8,1:8,25]
        enterAssignment[8,8:8,24]
          enterRef_obj[8,8:8,12]
            enterLogic_net[5,12:5,16]
            leaveLogic_net[5,12:5,16]
          leaveRef_obj[8,8:8,12]
          enterOperation[8,15:8,20]
            enterOperation[8,15:8,16]
              enterRef_obj[8,15:8,16]
                enterLogic_net[4,6:4,7]
                leaveLogic_net[4,6:4,7]
              leaveRef_obj[8,15:8,16]
              enterRef_obj[8,19:8,20]
                enterLogic_net[4,9:4,10]
                leaveLogic_net[4,9:4,10]
              leaveRef_obj[8,19:8,20]
            leaveOperation[8,15:8,16]
            enterRef_obj[8,23:8,24]
              enterLogic_net[4,12:4,13]
              leaveLogic_net[4,12:4,13]
            leaveRef_obj[8,23:8,24]
          leaveOperation[8,15:8,20]
        leaveAssignment[8,8:8,24]
      leaveAlways[8,1:8,25]
      enterCont_assign[7,8:7,24]
        enterOperation[7,15:7,20]
          enterOperation[7,15:7,16]
            enterRef_obj[7,15:7,16]
              enterLogic_net[4,6:4,7]
              leaveLogic_net[4,6:4,7]
            leaveRef_obj[7,15:7,16]
            enterRef_obj[7,19:7,20]
              enterLogic_net[4,9:4,10]
              leaveLogic_net[4,9:4,10]
            leaveRef_obj[7,19:7,20]
          leaveOperation[7,15:7,16]
          enterRef_obj[7,23:7,24]
            enterLogic_net[4,12:4,13]
            leaveLogic_net[4,12:4,13]
          leaveRef_obj[7,23:7,24]
        leaveOperation[7,15:7,20]
        enterRef_obj[7,8:7,12]
          enterLogic_net[5,6:5,10]
          leaveLogic_net[5,6:5,10]
        leaveRef_obj[7,8:7,12]
      leaveCont_assign[7,8:7,24]
    leaveModule[1,1:10,10]
````

Few things to note with this output.
* The order of callbacks for LHS & RHS are swapped between `assignment` & `cont_assign`. The order matters and it has to be consistent.
* Order of callbacks should be more like C++ i.e. base class get iterated ahead of the current class. So, `atomic_stmt` class members should be iterated before iterating `assignment` class members. Within among the class members itself, the callback should be in the order of their declaration i.e. if `a` is declared ahead of `b` then `a` should get its enter/leave callbacks before `b`'s.
* vpi_listener.cpp has a lot of duplicate code. Here's an easy way to reduce the duplication.
```
void enumerate_atomic_stmt(....) {
  enumerate_baseclass_p(...);
  // Iterate relevant members without calling any enter/leave callbacks
}
void listen_atomic_stmt(..., listener, ....) {
  listener->enter();
  enumerate_atomic_stmt();
  listener->leave();
}
void enumerate_assignment(...) {
  enumerate_atomic_stmt();
  // Iterate relevant members without calling enter/leave callbacks
}
void listen_assignment(...., listener, ....) {
  listener->enter();
  enumerate_assignment(....);
  listener->leave();
}
```
Basically, split the callbacks from the enumeration of the members. The enumeration logic doesn't have to be duplicated and can be shared across all sub-classes (added advantage is that the call-order will automatically be consistent as per the previous bullet point). In this approach, non-concrete classes will have an enumeration_xxx function but no listen_xx function implementation.
* As part of solution to avoid infinite recursion, callbacks should still be invoked but the members shouldn't be enumerated.
```
class A {
  B *b;
  C *c;
}
class B {
  A *a;
}
class C {
  B *b
}
```
listen_A would generate the following callbacks -
```
enter A
  enter B
    enter A
    // Don't enumerate the members of A again here but enter/leave callbacks are still required.
    leave A
  leave B
  enter C
    enter B
    // Don't enumerate the member of B because B is already visited once but still invoke the callbacks.
    leave B
  leave C
leave A
```

Food for thought - Rather than tracking visited, would it be enough to track the calls/objects on the stack. With this strategy, objects could be called multiple times (which ideally they should be because the code says so) but will not cause an infinite loop.

hs-apotell, 2021-08-18:
@alaindargelas Confirming that the change does simplify the listener logic quite a lot. Some of the things that required unnecessary book-keeping on the listener side, aren't needed anymore and reduces a lot of branching based on UhdmType. We will have to modify/remove some of the workarounds that had been implemented but overall this would be a welcome change (of course, with the necessary work to avoid infinite loops).

alaindargelas, 2021-08-20:
#1683 fixes the lhs rhs ordering issue, the concrete class callback and the loop detection (Same as vpi_visitor).
Duplicated code is automatically generated, what you suggest requires too much intelligence in the code generator. 
Verilog declaration order cannot be respected, its not save anywhere in the model.

closed 2021-08-20 by alaindargelas