chipsalliance/Surelog #1661: bit_select objects in port connections have parent->actual == null

pieter3d, 2021-08-16:
consider the following source
```verilog
module sub ( input some_input);
endmodule

module top;
logic some_net;
sub sub_inst (.some_input ( some_net[0]));
endmodule
```
The bit_select object in the port connection has no way to refer it back to the `some_net` net. See the UHDM dump:
```
\_module: work@top (work@top) test.sv:4:1: , endln:7:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiModule:
  \_module: work@sub (work@top.sub_inst) test.sv:6:1: , endln:6:43, parent:work@top
    |vpiDefName:work@sub
    |vpiDefFile:test.sv
    |vpiDefLineNo:1
    |vpiName:sub_inst
    |vpiFullName:work@top.sub_inst
    |vpiPort:
    \_port: (some_input), line:1:20, endln:1:30, parent:work@top.sub_inst
      |vpiName:some_input
      |vpiDirection:1
      |vpiHighConn:
      \_bit_select: (some_net), line:6:29, endln:6:40, parent:some_net
        |vpiName:some_net
        |vpiIndex:
        \_constant: , line:6:38, endln:6:39, parent:some_net
          |vpiConstType:9
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
      |vpiLowConn:
      \_ref_obj: 
        |vpiActual:
        \_logic_net: (work@top.sub_inst.some_input), line:1:20, endln:1:30, parent:work@top.sub_inst
          |vpiName:some_input
          |vpiFullName:work@top.sub_inst.some_input
    |vpiNet:
    \_logic_net: (work@top.sub_inst.some_input), line:1:20, endln:1:30, parent:work@top.sub_inst
    |vpiInstance:
    \_module: work@top (work@top) test.sv:4:1: , endln:7:10
  |vpiNet:
  \_logic_net: (work@top.some_net), line:5:7, endln:5:15, parent:work@top
    |vpiName:some_net
    |vpiFullName:work@top.some_net
    |vpiNetType:36
```
The parent is a ref_obj, but actual is not set on it (nullptr).

pieter3d, 2021-08-16:
Update: This seems to be the case for several expression types in port connections.

pieter3d, 2021-08-16:
Similarly, it seems that any ref_obj has a null parent when found in a port connection.

alaindargelas, 2021-08-19:
#1675 Should take care of this issue, please confirm and close if it does

pieter3d, 2021-08-19:
Hmm, not quite. The FullName is now populated, but still no parent for any refobj in ports connections, still null.

pieter3d, 2021-08-19:
Ah, apologies I was viewing things wrong, parent is set properly.
I think this can be closed!

closed 2021-08-19 by pieter3d