chipsalliance/Surelog #1708: Recursive Parameter Substitution

RRozak, 2021-08-24:
There is a problem with the following modules:
```systemverilog
module dut1(output logic [1:0] a);
   parameter logic [1:0] P1 = 0;
   assign a = P1;
endmodule // dut1

module dut2(output logic [1:0] b);
   parameter logic [1:0] P2 = 2'b11;
   dut1 #(
      .P1(~P2)
   ) u_dut1(
      .a(b)
   );
endmodule // dut2

module top(output logic [1:0] o);
   parameter logic [1:0] X = '{0, 1};
   dut2 #(
      .P2(X)
   ) u_dut2(
      .b(o)
   );
endmodule // top
```
`P2` in `u_dut1` initialization is replaced with `X`, but `X` is not visible from `dut2` module.
Here is uhdm:
```
|uhdmtopModules:
\_module: work@top (work@top) top.sv:15:1: , endln:22:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiPort:
  \_port: (o), line:15:31, endln:15:32, parent:work@top
    |vpiName:o
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: (work@top.o)
      |vpiName:o
      |vpiFullName:work@top.o
      |vpiActual:
      \_logic_net: (work@top.o), line:15:31, endln:15:32, parent:work@top
        |vpiName:o
        |vpiFullName:work@top.o
        |vpiNetType:36
        |vpiRange:
        \_range: , line:15:26, endln:15:29
          |vpiLeftRange:
          \_constant: , line:15:26, endln:15:27
            |vpiConstType:9
            |vpiDecompile:1
            |vpiSize:64
            |UINT:1
          |vpiRightRange:
          \_constant: , line:15:28, endln:15:29
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
  |vpiModule:
  \_module: work@dut2 (work@top.u_dut2) top.sv:17:4: , endln:21:6, parent:work@top
    |vpiDefName:work@dut2
    |vpiDefFile:top.sv
    |vpiDefLineNo:6
    |vpiName:u_dut2
    |vpiFullName:work@top.u_dut2
    |vpiPort:
    \_port: (b), line:6:32, endln:6:33, parent:work@top.u_dut2
      |vpiName:b
      |vpiDirection:2
[... 197 lines cut ...]
          |vpiConstType:9
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
  |vpiParamAssign:
  \_param_assign: , line:16:26, endln:16:37, parent:work@top
    |vpiRhs:
    \_operation: , line:18:11, endln:18:12
      |vpiOpType:75
      |vpiReordered:1
      |vpiOperand:
      \_constant: , line:16:35, endln:16:36
        |vpiConstType:9
        |vpiDecompile:1
        |vpiSize:64
        |UINT:1
      |vpiOperand:
      \_constant: , line:16:32, endln:16:33
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
    |vpiLhs:
    \_parameter: (work@top.X), line:16:26, endln:16:37, parent:work@top
```

alaindargelas, 2021-08-25:
Surelog is doing exactly what you would expect:
- it does not substitute the value (as instructed by the option) and
- it gives you an upward reference to top.X (vpiFullName) to the parameter definition in the upward module (per standard section 23.8).

RRozak, 2021-08-25:
Parameters should be initialized with constant expressions, but dotted reference is not a constant expression (according to 11.2.1).
So `~P2` can't be substituted with `~top.X`.

alaindargelas, 2021-08-28:
I'm at loss. Do you want:
- substitution to happen with info on data type 
   or
- no substitution (Current solution with --disable-feature=parametersubstitution option)

@rkapuscik, @kbieganski please pitch in, this is to circumvent Verilator/Yosys limitations.

kbieganski, 2021-08-30:
We're actually considering getting rid of `--disable-feature=parametersubstitution` by working around Verilator's limitation. We need the assignment pattern to have a typespec for this, and it used to not have it. However, I just checked the latest Surelog and it seems you've already added it at some point. Now since #1728 has been fixed we can use it, so I guess this issue can be closed.

kbieganski, 2021-08-30:
I have a question regarding the operation typespec. For the example given by @RRozak, Surelog generates a pattern with a named typespec. The name of the typespec is `X`, like the original param. Is this the desired behavior, or a bug?

I'm asking because if the typespec has a name, we're letting Verilator resolve the type on its own. But in this case the type `X` does not exist (it's a parameter), so Verilator reports an error.
```
    |vpiParamAssign:
    \_param_assign: , line:7:26, endln:7:36, parent:work@top.u_dut2
      |vpiRhs:
      \_operation: , line:18:11, endln:18:12
        |vpiOpType:75
        |vpiReordered:1
        |vpiOperand:
        \_constant: , line:16:35, endln:16:36
          |vpiConstType:9
          |vpiDecompile:1
          |vpiSize:64
          |UINT:1
        |vpiOperand:
        \_constant: , line:16:32, endln:16:33
          |vpiConstType:9
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
        |vpiTypespec:
        \_logic_typespec: (X), line:16:14, endln:16:19, parent:work@top.X
          |vpiName:X
          |vpiRange:
          \_range: , line:16:21, endln:16:24, parent:X
            |vpiLeftRange:
            \_constant: , line:16:21, endln:16:22
              |vpiConstType:9
              |vpiDecompile:1
              |vpiSize:64
              |UINT:1
            |vpiRightRange:
            \_constant: , line:16:23, endln:16:24
              |vpiConstType:9
              |vpiDecompile:0
              |vpiSize:64
              |UINT:0
      |vpiLhs:
      \_parameter: (work@top.u_dut2.P2), line:7:26, endln:7:36, parent:work@top.u_dut2
```

closed 2021-08-30 by RRozak

alaindargelas, 2021-08-31:
I can remove the name from the typespec so there is no conflict.
Giving a name was an arbitrary decision on my part.