chipsalliance/Surelog #1762: Value of $bits of struct in concat op in param assign is not being calculated

kbieganski, 2021-08-30:
Similar to #1398, but with structs. Code:
(the concat op is redundant here, but it's just an example)
```systemverilog
package pkg;
  typedef struct packed {
    logic[7:0] x;
  } struct_t;
endpackage : pkg

module dut
  #(parameter int Width = 1)
  (input logic[Width-1:0] in, output logic[Width-1:0] out);
   assign out = in;
endmodule;

module top (input pkg::struct_t in, output logic[7:0] out);
   dut #(.Width($bits({in.x}))) dut(.in(in), .out(out));
endmodule
```

UHDM:
(the RHS is a reference to `in.x` from the parent module)

```
  |vpiModule:
  \_module: work@dut (work@top.dut) /home/krzysztof/verilator-uhdm/uhdm-integration/tests/bug/dut.v:14:4: , endln:14:57, parent:work@top
    ...
    |vpiParamAssign:
    \_param_assign: , line:8:19, endln:8:28, parent:work@top.dut
      |vpiRhs:
      \_sys_func_call: ($bits), line:14:17, endln:14:30
        |vpiName:$bits
        |vpiArgument:
        \_operation: , line:14:23, endln:14:29, parent:$bits
          |vpiOpType:33
          |vpiOperand:
          \_hier_path: (in.x), line:14:24, endln:14:28, parent:$bits
            |vpiName:in.x
            |vpiActual:
            \_ref_obj: (in)
              |vpiName:in
            |vpiActual:
            \_ref_obj: (x)
              |vpiName:x
      |vpiLhs:
      \_parameter: (work@top.dut.Width), line:8:19, endln:8:28, parent:work@top.dut
```

The same happens if I remove the `.x`:

```systemverilog
package pkg;
  typedef struct packed {
    logic[7:0] x;
  } struct_t;
endpackage : pkg

module dut
  #(parameter int Width = 1)
  (input logic[Width-1:0] in, output logic[Width-1:0] out);
   assign out = in;
endmodule;

module top (input pkg::struct_t in, output logic[7:0] out);
   dut #(.Width($bits({in}))) dut(.in(in), .out(out));
endmodule
```

UHDM:
(the RHS is a reference to `in` from the parent module)

```
  |vpiModule:
  \_module: work@dut (work@top.dut) dut.v:14:4: , endln:14:55, parent:work@top
    ...
    |vpiParamAssign:
    \_param_assign: , line:8:19, endln:8:28, parent:work@top.dut
      |vpiRhs:
      \_sys_func_call: ($bits), line:14:17, endln:14:28
        |vpiName:$bits
        |vpiArgument:
        \_operation: , line:14:23, endln:14:27, parent:$bits
          |vpiOpType:33
          |vpiOperand:
          \_ref_obj: (in), line:14:24, endln:14:26, parent:$bits
            |vpiName:in
            |vpiActual:
            \_struct_net: (work@top.in), line:13:33, endln:13:35, parent:work@top
      |vpiLhs:
      \_parameter: (work@top.dut.Width), line:8:19, endln:8:28, parent:work@top.dut
```

Note that if I change the type of `in` to `logic`:

```systemverilog
module dut
  #(parameter int Width = 1)
  (input logic[Width-1:0] in, output logic[Width-1:0] out);
   assign out = in;
endmodule;

module top (input logic[7:0] in, output logic[7:0] out);
   dut #(.Width($bits({in}))) dut(.in(in), .out(out));
endmodule
```

then the UHDM is correct:

```
|vpiModule:
  \_module: work@dut (work@top.dut) dut.v:8:4: , endln:8:55, parent:work@top
    ...
    |vpiParamAssign:
    \_param_assign: , line:2:19, endln:2:28, parent:work@top.dut
      |vpiRhs:
      \_constant: , line:2:27, endln:2:28
        |vpiConstType:9
        |vpiDecompile:8
        |vpiSize:64
        |UINT:8
        |vpiTypespec:
        \_int_typespec: (Width), line:2:15, endln:2:18, parent:work@dut.Width
          |vpiName:Width
      |vpiLhs:
      \_parameter: (work@top.dut.Width), line:2:19, endln:2:28, parent:work@top.dut
```

It also works correctly if I get rid of the concat op.

alaindargelas, 2021-08-31:
Fixed by #1923

kbieganski, 2021-08-31:
Unfortunately, the value isn't calculated correctly (or not in all cases).

For this code:

```systemverilog
package edn_pkg;
  typedef struct packed {
    logic                                 edn_ack;
    logic                                 edn_fips;
    logic [31:0]        edn_bus;
  } edn_rsp_t;
endpackage

module prim_sync_reqack_data #(
  parameter int unsigned Width       = 1
) (
  input  logic [Width-1:0] data_i
);
endmodule

module prim_edn_req (
  input  edn_pkg::edn_rsp_t   edn_i
);
  localparam int SyncWidth = $bits({edn_i.edn_fips, edn_i.edn_bus});
  prim_sync_reqack_data #(
    .Width(SyncWidth)
  ) u_prim_sync_reqack_data (
    .data_i     ( {edn_i.edn_fips, edn_i.edn_bus} )
  );
endmodule
```

we get this UHDM. Note that `Width` gets assigned 0, which leads to the left range of `data_i` being set to -1.

```
  |vpiModule:
  \_module: work@prim_sync_reqack_data (work@prim_edn_req.u_prim_sync_reqack_data) dut.v:20:3: , endln:24:5, parent:work@prim_edn_req
    ...
    \_port: (data_i), line:12:28, endln:12:34, parent:work@prim_edn_req.u_prim_sync_reqack_data
      ...
      |vpiLowConn:
      \_ref_obj: (work@prim_edn_req.u_prim_sync_reqack_data.data_i)
        |vpiName:data_i
        |vpiFullName:work@prim_edn_req.u_prim_sync_reqack_data.data_i
        |vpiActual:
        \_logic_net: (work@prim_edn_req.u_prim_sync_reqack_data.data_i), line:12:28, endln:12:34, parent:work@prim_edn_req.u_prim_sync_reqack_data
          |vpiName:data_i
          |vpiFullName:work@prim_edn_req.u_prim_sync_reqack_data.data_i
          |vpiNetType:36
          |vpiRange:
          \_range: , line:12:17, endln:12:26
            |vpiLeftRange:
            \_constant: , line:12:17, endln:12:22
              |vpiConstType:7
              |vpiDecompile:-1
              |vpiSize:64
              |INT:-1
            |vpiRightRange:
            \_constant: , line:12:25, endln:12:26
              |vpiConstType:9
              |vpiDecompile:0
              |vpiSize:64
              |UINT:0
    ...
    |vpiParamAssign:
    \_param_assign: , line:10:26, endln:10:41, parent:work@prim_edn_req.u_prim_sync_reqack_data
      |vpiRhs:
      \_constant: , line:10:40, endln:10:41
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
        |vpiTypespec:
        \_int_typespec: (Width), line:10:13, endln:10:16, parent:work@prim_sync_reqack_data.Width
          |vpiName:Width
      |vpiLhs:
      \_parameter: (work@prim_edn_req.u_prim_sync_reqack_data.Width), line:10:26, endln:10:41, parent:work@prim_edn_req.u_prim_sync_reqack_data
```

hzeller, 2021-08-31:
(there should be a unit test :) )

alaindargelas, 2021-09-02:
Fixed by #1929, including unittest.
BTW, this was a pretty different issue. Open a different case next time, do not reopen a case when you provide a different testcase. Issues might look the "same" but they are not.

closed 2021-09-02 by alaindargelas