chipsalliance/Surelog #1803: Why does Surelog use tmalloc?

pieter3d, 2021-09-05:
When including surelog in a separate project, statically linked, it results in a mix of tcmalloc libs and non-tcmalloc libs. For me at least this caused issues with ASAN/MSAN builds. Fixed by removing the tcmalloc dependency from the surelog `INTERFACE_LINK_LIBRARIES` in the installed `Surelog.cmake` file.

alaindargelas, 2021-09-06:
There is a cmake option in Surelog to not use TC_MALLOC. Maybe it is not obvious or even possible to use from the higher level project?
It gives a good 10% runtime improvement op top of the regular C++ allocator, so it should stay in the default build.

pieter3d, 2021-09-06:
Okay, yeah leaving it enabled by default seems reasonable.

alaindargelas, 2021-09-07:
Let us know if the NO_TCMALLOC define works. I see a comment in the Surelog/CMakeLists.txt that this option should be renamed WITH_TCMALLOC, maybe this is the right time to fix this.

pieter3d, 2021-09-07:
It's not causing any immediate problems. If it's possible to set something prior to the `find_package` that ensures no tcmalloc pull-in, it would be nice to document that in the CMake section of the readme. But certainly there's no urgency here.

alaindargelas, 2021-09-07:
Can you try:
(set NO_TCMALLOC True)
find_package(Surelog ....) 
If that works, I'll document it.

pieter3d, 2021-09-07:
Hmm, that doesn't seem to work. Still builds with tcmalloc.
if I grep for NO_TCMALLOC in the installed `/usr/local/cmake/` area, there are no results

pieter3d, 2021-09-08:
I just did a vanilla Surelog `make` + `sudo make install`. It installs stuff to `/usr/local/cmake`.

In my project I have nothing other than
```cmake
find_package(Surelog REQUIRED)
```
And later on:
```cmake
target_link_libraries(my_exe PRIVATE surelog)
```
No special cmake arguments. I use the Ninja generator normally, but this works with the default make generator too.

alaindargelas, 2021-09-08:
Thanks! that worked, I was doing a make install with a PREFIX, then pointing to that directory, that does not seem to work at all. It does install but not correctly, the correct option is DESTDIR= (I forgot).

With the default path, it does work.
Now I see that Surelog has to be built without TCMALLOC first, then installed, then used in the 3rd party project.

So the instructions if you don't want tcmalloc should be (I just update the Makefile):

cd Surelog; make release_no_tcmalloc
make install

Then the rest is the same.

pieter3d, 2021-09-08:
Okay, yeah that makes sense

alaindargelas, 2021-09-08:
Fixed by #1997

closed 2021-09-08 by alaindargelas