chipsalliance/Surelog #1807: Parameter imported from package is not substituted when it is on parameter port list

RRozak, 2021-09-06:
It is similar to https://github.com/chipsalliance/Surelog/issues/1774
The only difference is the place of parameter `P` declaration.
```systemverilog
package aes_pkg;
   parameter logic [7:0][3:0] X = 32'habcd;
endpackage // aes_pkg

module aes_cipher_core import aes_pkg::*;
   #(
      parameter logic [7:0][3:0] P = X
   );
endmodule // aes_cipher_core

module top;
   aes_cipher_core u_aes_cipher_core();
endmodule
```
There is the same problem as in #1774, but example from #1774 works well after the fix. 
Here is uhdm:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallPackages:
\_package: aes_pkg (aes_pkg::) param_pkg.sv:1:1: , endln:3:11, parent:work@top
  |vpiDefName:aes_pkg
  |vpiName:aes_pkg
  |vpiFullName:aes_pkg::
  |vpiParameter:
  \_parameter: (aes_pkg::X), line:2:31, endln:2:43, parent:aes_pkg::
    |vpiName:X
    |vpiFullName:aes_pkg::X
    |HEX:abcd
    |vpiTypespec:
    \_logic_typespec: (X), line:2:14, endln:2:19, parent:aes_pkg::X
      |vpiName:X
      |vpiRange:
      \_range: , line:2:21, endln:2:24
        |vpiLeftRange:
        \_constant: , line:2:21, endln:2:22
          |vpiConstType:9
          |vpiDecompile:7
          |vpiSize:64
          |UINT:7
        |vpiRightRange:
        \_constant: , line:2:23, endln:2:24
          |vpiConstType:9
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
      |vpiRange:
      \_range: , line:2:26, endln:2:29
        |vpiLeftRange:
        \_constant: , line:2:26, endln:2:27
          |vpiConstType:9
          |vpiDecompile:3
          |vpiSize:64
          |UINT:3
        |vpiRightRange:
        \_constant: , line:2:28, endln:2:29
          |vpiConstType:9
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
      |vpiInstance:
      \_package: aes_pkg (aes_pkg::) param_pkg.sv:1:1: , endln:3:11, parent:work@top
  |vpiParamAssign:
  \_param_assign: , line:2:31, endln:2:43, parent:aes_pkg::
    |vpiRhs:
    \_constant: , line:2:35, endln:2:43
[... 369 lines cut ...]
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
    |vpiParamAssign:
    \_param_assign: , line:7:33, endln:7:38, parent:work@top.u_aes_cipher_core
      |vpiRhs:
      \_ref_obj: (X), line:7:37, endln:7:38
        |vpiName:X
        |vpiActual:
        \_parameter: (work@top.u_aes_cipher_core.X), line:2:31, endln:2:43, parent:work@top.u_aes_cipher_core
      |vpiLhs:
      \_parameter: (work@top.u_aes_cipher_core.P), line:7:33, endln:7:38, parent:work@top.u_aes_cipher_core
    |vpiParamAssign:
    \_param_assign: , line:2:31, endln:2:43, parent:work@top.u_aes_cipher_core
      |vpiRhs:
      \_constant: , line:2:35, endln:2:43
        |vpiConstType:5
        |vpiDecompile:32'habcd
        |vpiSize:32
        |HEX:abcd
      |vpiLhs:
      \_parameter: (work@top.u_aes_cipher_core.X), line:2:31, endln:2:43, parent:work@top.u_aes_cipher_core
    |vpiTypedef:
    \_import: (aes_pkg), line:5:31, endln:5:41
```

alaindargelas, 2021-09-07:
I'm reading this case and the previous one and I have one question that I missed in the previous case:
The import statement imported all the parameters from the package, why aren't they visible?
the X parameter is actually visible in the submodule, just look above, 
```
|vpiParamAssign:
    \_param_assign: , line:2:31, endln:2:43, parent:work@top.u_aes_cipher_core
      |vpiRhs:
      \_constant: , line:2:35, endln:2:43
        |vpiConstType:5
        |vpiDecompile:32'habcd
        |vpiSize:32
        |HEX:abcd
      |vpiLhs:
      \_parameter: (work@top.u_aes_cipher_core.X), line:2:31, endln:2:43, parent:work@top.u_aes_cipher_core
    |vpiTypedef:
    \_import: (aes_pkg), line:5:31, endln:5:41
```
Why you are not passing that definition to Verilator?

RRozak, 2021-09-07:
We handle all vpiParamAssign nodes of the instance as they would be passed from the upper module. In this case it causes problem, because `X` is not visible from top module.
The output tree is correct, but our verilator doesn't work with it.

That vpiTypedef with import is under vpiModule, not under vpiParamAssign, so we don't know if `X` is defined in that package. There can be more packages imported.

alaindargelas, 2021-09-08:
Fixed by #2001

closed 2021-09-08 by alaindargelas