chipsalliance/Surelog #1808: Problems with pattern parameters in nested module

RRozak, 2021-09-06:
There are 2 problems with the following file:
```systemverilog
module prim_subreg;
   parameter logic [4:0] RESVAL = '0;
   int a = int'(RESVAL);
endmodule // prim_subreg

module prim_subreg_shadow;
   typedef struct packed {
      logic [2:0] a;
      logic [1:0] b;
   } struct_ab;

   parameter struct_ab RESVAL = '{
      a: '0,
      b: '1
   };

   prim_subreg #(
      .RESVAL(RESVAL)
   ) staged_reg();
endmodule // prim_subreg_shadow

module top;
   typedef struct packed {
      logic [1:0] a;
      logic [2:0] b;
   } struct_ab;

   parameter struct_ab CTRL_RESET = '{
      a: '1,
      b: '0
   };

   prim_subreg_shadow #(
      .RESVAL(CTRL_RESET)
   ) u_ctrl_reg_shadowed();
endmodule // top
```
1. The value of `RESVAL` parameter in instance `top.u_ctrl_reg_shadowed.staged_reg` is wrong. `a` should be equal to `'1`, `b` to `'0`. It looks like the default value from `prim_subreg` was taken.

2. The typename is ambiguous.  It is unknown if that `struct_ab` type is the one defined in `top` or in `prim_subreg`. Could you add `vpiFullName` field?

Here is uhdm:
```
|uhdmtopModules:
\_module: work@top (work@top) pattern.sv:22:1: , endln:36:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiModule:
  \_module: work@prim_subreg_shadow (work@top.u_ctrl_reg_shadowed) pattern.sv:33:4: , endln:35:28, parent:work@top
    |vpiDefName:work@prim_subreg_shadow
    |vpiDefFile:pattern.sv
    |vpiDefLineNo:6
    |vpiName:u_ctrl_reg_shadowed
    |vpiFullName:work@top.u_ctrl_reg_shadowed
    |vpiModule:
    \_module: work@prim_subreg (work@top.u_ctrl_reg_shadowed.staged_reg) pattern.sv:17:4: , endln:19:19, parent:work@top.u_ctrl_reg_shadowed
      |vpiDefName:work@prim_subreg
      |vpiDefFile:pattern.sv
      |vpiDefLineNo:1
      |vpiName:staged_reg
      |vpiFullName:work@top.u_ctrl_reg_shadowed.staged_reg
      |vpiInstance:
      \_module: work@prim_subreg_shadow (work@top.u_ctrl_reg_shadowed) pattern.sv:33:4: , endln:35:28, parent:work@top
      |vpiVariables:
      \_int_var: (work@top.u_ctrl_reg_shadowed.staged_reg.a), line:3:8, endln:3:9, parent:work@top.u_ctrl_reg_shadowed.staged_reg
        |vpiName:a
        |vpiFullName:work@top.u_ctrl_reg_shadowed.staged_reg.a
        |vpiAutomatic:1
[... 316 lines cut ...]
          \_constant: , line:24:16, endln:24:17
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
    |vpiTypespecMember:
    \_typespec_member: (b), line:25:19, endln:25:20
      |vpiName:b
      |vpiTypespec:
      \_logic_typespec: , line:25:7, endln:25:12
        |vpiRange:
        \_range: , line:25:14, endln:25:17, parent:struct_ab
          |vpiLeftRange:
          \_constant: , line:25:14, endln:25:15
            |vpiConstType:9
            |vpiDecompile:2
            |vpiSize:64
            |UINT:2
          |vpiRightRange:
          \_constant: , line:25:16, endln:25:17
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
```

alaindargelas, 2021-09-12:
1) issue is fixed with #1829

alaindargelas, 2021-09-12:
2) issue fixed with #2047

alaindargelas, 2021-09-12:
Added content to vpiInstance relation which is in the Standard instead of vpiFullName that is not in the Standard.

closed 2021-09-12 by alaindargelas