chipsalliance/Surelog #1832: antlr submodule github action issue

alaindargelas, 2021-09-12:
I wanted to temporary remore antlr from the list of submodules, until @hs-apotell  provides a dull solution.... but 
there is an error that happens only on the github actions (Not when checking out).
It stills mention antlr in the .gitmodules but there is no definition for it, there is some sort of caching happening.

hs-apotell, 2021-09-12:
No clue what's happening here. Looks like the .git folder got cached somehow. You might want to try removing the caching in the workflow file. Caching was a recent change and might have something to do with it (don't see how but nevertheless worth a shot).

alaindargelas, 2021-09-12:
Fixed, somehow the directory was still there with no definition in the .gitmodules, that was creating this issue

closed 2021-09-12 by alaindargelas