chipsalliance/Surelog #1835: Proposal: Relax line-length limit with clang format

hs-apotell, 2021-09-12:
Given how the language has changed in past few years (template specializations, lambdas and many more features), 80 character limit makes the code quite unreadable. I propose to relax the line character limit to 140 characters or higher.

I know request like these are very close to developer's heart. Don't intend to start a war of words on this. If this feels unreasonable or not up for debate, feel free to close the request.

alaindargelas, 2021-09-21:
@hzeller, @mithro for comments

hzeller, 2021-09-21:
We're using the [existing Google coding style](https://github.com/chipsalliance/Surelog/blob/master/.clang-format), which has 80 characters. But we also only do 2 character indentation, so typically things fit well.

I frequently have multiple editors open side-by-side and I very much enjoy the advantage of 80 characters (5 editors with 80 character width fit nicely on 4k :) ).

So I'd say it is a pretty fixed limit right now.

(In Java, things a quite a bit more verbose, so for antlr changes, 100 characters might be a good limit for these files)

alaindargelas, 2021-10-03:
We are keeping the style as it is.

closed 2021-10-03 by alaindargelas