chipsalliance/Surelog #1854: Clean valgrind warnings

alaindargelas, 2021-09-14:
There are way too many warnings in Valgrind which makes it difficult to find real issues.
Looks like a lot of them are "string" related. Too many "const std::string&" are used instead of proper "std::string" being constructed. Then later get deleted. 
[valgrind.log](https://github.com/chipsalliance/Surelog/files/7163488/valgrind.log)

hzeller, 2021-09-14:
Note, the 'mismatch free / delete / delete []' messages might be due to https://github.com/gperftools/gperftools/issues/792
(looks like strings are allocated with new[] and deleted with free(), so there is the mismatch).

So for your tests, you might not compile with heapcheck.

closed 2021-09-14 by hzeller

hzeller, 2021-09-15:
Other than memory leaks, there don't seem to be any issues with a valgrind run of that one

```
==2046459== Memcheck, a memory error detector
==2046459== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==2046459== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==2046459== Command: ~/src/my/Surelog/dbuild/bin/surelog -parse -elabuhdm -verbose -DVERILATOR=1 --assert -sv --top-module Main_Zynq_Wrapper -Wno-WIDTH -Wno-INITIALDLY -Wno-UNOPTFLAT +define+RSD_FUNCTIONAL_SIMULATION +define+RSD_FUNCTIONAL_SIMULATION_VERILATOR +define+RSD_MARCH_INT_ISSUE_WIDTH=2 --Mdir ../Project/Verilator/obj_dir +incdir+. --trace -CFLAGS -Os -output-split 15000 ./MicroArchConf.sv ./BasicTypes.sv ./Memory/MemoryMapTypes.sv ./Cache/CacheSystemTypes.sv ./Memory/MemoryTypes.sv ./Decoder/OpFormat.sv ./Decoder/MicroOp.sv ./RegisterFile/BypassTypes.sv ./FetchUnit/FetchUnitTypes.sv ./LoadStoreUnit/LoadStoreUnitTypes.sv ./RenameLogic/RenameLogicTypes.sv ./Scheduler/SchedulerTypes.sv ./Pipeline/PipelineTypes.sv ./IO/IO_UnitTypes.sv ./Privileged/CSR_UnitTypes.sv ./Debug/DebugTypes.sv ./Main_Zynq_Wrapper.sv ./Main_Zynq.sv ./Core.sv ./Pipeline/FetchStage/NextPCStage.sv ./Pipeline/FetchStage/NextPCStageIF.sv ./Pipeline/FetchStage/FetchStage.sv ./Pipeline/FetchStage/FetchStageIF.sv ./Pipeline/FetchStage/PC.sv ./Pipeline/PreDecodeStage.sv ./Pipeline/PreDecodeStageIF.sv ./Pipeline/DecodeStage.sv ./Pipeline/DecodeStageIF.sv ./Pipeline/RenameStage.sv ./Pipeline/RenameStageIF.sv ./Pipeline/DispatchStage.sv ./Pipeline/DispatchStageIF.sv ./Pipeline/ScheduleStage.sv ./Pipeline/ScheduleStageIF.sv ./Pipeline/IntegerBackEnd/IntegerIssueStage.sv ./Pipeline/IntegerBackEnd/IntegerIssueStageIF.sv ./Pipeline/IntegerBackEnd/IntegerRegisterReadStage.sv ./Pipeline/IntegerBackEnd/IntegerRegisterReadStageIF.sv ./Pipeline/IntegerBackEnd/IntegerExecutionStageIF.sv ./Pipeline/IntegerBackEnd/IntegerExecutionStage.sv ./Pipeline/IntegerBackEnd/IntegerRegisterWriteStageIF.sv ./Pipeline/IntegerBackEnd/IntegerRegisterWriteStage.sv ./Pipeline/ComplexIntegerBackEnd/ComplexIntegerIssueStage.sv ./Pipeline/ComplexIntegerBackEnd/ComplexIntegerIssueStageIF.sv ./Pipeline/ComplexIntegerBackEnd/ComplexIntegerRegisterReadStage.sv ./Pipeline/ComplexIntegerBackEnd/ComplexIntegerRegisterReadStageIF.sv ./Pipeline/ComplexIntegerBackEnd/ComplexIntegerExecutionStageIF.sv ./Pipeline/ComplexIntegerBackEnd/ComplexIntegerExecutionStage.sv ./Pipeline/ComplexIntegerBackEnd/ComplexIntegerRegisterWriteStage.sv ./Pipeline/MemoryBackEnd/MemoryIssueStage.sv ./Pipeline/MemoryBackEnd/MemoryIssueStageIF.sv ./Pipeline/MemoryBackEnd/MemoryRegisterReadStage.sv ./Pipeline/MemoryBackEnd/MemoryRegisterReadStageIF.sv ./Pipeline/MemoryBackEnd/MemoryExecutionStageIF.sv ./Pipeline/MemoryBackEnd/MemoryExecutionStage.sv ./Pipeline/MemoryBackEnd/MemoryAccessStageIF.sv ./Pipeline/MemoryBackEnd/MemoryAccessStage.sv ./Pipeline/MemoryBackEnd/MemoryTagAccessStageIF.sv ./Pipeline/MemoryBackEnd/MemoryTagAccessStage.sv ./Pipeline/MemoryBackEnd/MemoryRegisterWriteStageIF.sv ./Pipeline/MemoryBackEnd/MemoryRegisterWriteStage.sv ./Pipeline/CommitStageIF.sv ./Pipeline/CommitStage.sv ./RegisterFile/RegisterFile.sv ./RegisterFile/RegisterFileIF.sv ./RegisterFile/BypassController.sv ./RegisterFile/BypassNetwork.sv ./RegisterFile/BypassNetworkIF.sv ./RegisterFile/VectorBypassNetwork.sv ./ExecUnit/BitCounter.sv ./ExecUnit/IntALU.sv ./ExecUnit/Shifter.sv ./ExecUnit/MultiplierUnit.sv ./ExecUnit/VectorUnit.sv ./ExecUnit/PipelinedRefDivider.sv ./ExecUnit/DividerUnit.sv ./MulDivUnit/MulDivUnitIF.sv ./MulDivUnit/MulDivUnit.sv ./LoadStoreUnit/LoadStoreUnit.sv ./LoadStoreUnit/LoadStoreUnitIF.sv ./LoadStoreUnit/LoadQueue.sv ./LoadStoreUnit/StoreQueue.sv ./LoadStoreUnit/StoreCommitter.sv ./RenameLogic/RenameLogic.sv ./RenameLogic/RenameLogicIF.sv ./RenameLogic/ActiveListIF.sv ./RenameLogic/ActiveList.sv ./RenameLogic/RMT.sv ./RenameLogic/RetirementRMT.sv ./RenameLogic/RenameLogicCommitter.sv ./Decoder/Decoder.sv ./Decoder/DecodedBranchResolver.sv ./FetchUnit/BTB.sv ./FetchUnit/BranchPredictor.sv ./FetchUnit/Gshare.sv ./FetchUnit/Bimodal.sv ./Scheduler/SchedulerIF.sv ./Scheduler/IssueQueue.sv ./Scheduler/ReplayQueue.sv ./Scheduler/WakeupSelectIF.sv ./Scheduler/DestinationRAM.sv ./Scheduler/ReadyBitTable.sv ./Scheduler/Scheduler.sv ./Scheduler/SelectLogic.sv ./Scheduler/SourceCAM.sv ./Scheduler/WakeupLogic.sv ./Scheduler/WakeupPipelineRegister.sv ./Scheduler/ProducerMatrix.sv ./Scheduler/MemoryDependencyPredictor.sv ./Cache/CacheSystemIF.sv ./Cache/DCache.sv ./Cache/DCacheIF.sv ./Cache/ICache.sv ./Cache/MemoryAccessController.sv ./Cache/CacheFlushManager.sv ./Cache/CacheFlushManagerIF.sv ./Memory/Memory.sv ./Recovery/RecoveryManager.sv ./Recovery/RecoveryManagerIF.sv ./ControllerIF.sv ./Controller.sv ./ResetController.sv ./Privileged/InterruptController.sv ./Privileged/CSR_Unit.sv ./Privileged/CSR_UnitIF.sv ./IO/IO_Unit.sv ./IO/IO_UnitIF.sv ./Primitives/FlipFlop.sv ./Primitives/FreeList.sv ./Primitives/Queue.sv ./Primitives/RAM.sv ./Primitives/LRU_Counter.sv ./Primitives/Picker.sv ./Primitives/Multiplier.sv ./Primitives/Divider.sv ./Debug/Debug.sv ./Debug/DebugIF.sv ./Debug/PerformanceCounter.sv ./Debug/PerformanceCounterIF.sv ./Memory/Axi4LiteControlRegisterIF.sv ./Memory/Axi4LiteControlRegister.sv ./Memory/ControlQueue.sv ./Memory/Axi4Memory.sv ./Memory/Axi4MemoryIF.sv ./Memory/MemoryReadReqQueue.sv ./Memory/MemoryWriteDataQueue.sv ./Memory/MemoryLatencySimulator.sv ./SysDeps/Verilator/VerilatorHelper.s [... cut ...]
==2046459==    still reachable: 1,116,469,922 bytes in 17,425,929 blocks
==2046459==                       of which reachable via heuristic:
==2046459==                         newarray           : 336 bytes in 7 blocks
==2046459==         suppressed: 0 bytes in 0 blocks
==2046459== Rerun with --leak-check=full to see details of leaked memory
==2046459== 
==2046459== For lists of detected and suppressed errors, rerun with: -s
==2046459== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

alaindargelas, 2021-09-15:
Great thanks, now I should be able to debug what is wrong with my branch

PR #1856 (2021-09-14): Don't use tcmalloc in debug binary.
It interferes badly with valgrind.

Fixes: #1854
commit 61bfa5bf (2021-09-14): Don't use tcmalloc in debug binary.

It interferes badly with valgrind.

Fixes: #1854