chipsalliance/Surelog #1865: Wrong position of expressions in assignment pattern

kamilrakoczy, 2021-09-15:
According to standard (37.57 Expressions 5)): 
```
For an operation of type vpiAssignmentPatternOp, the operand iteration shall return the expressions as if the
assignment pattern were written with the positional notation. Nesting of assignment patterns shall be preserved.
Example 1:
struct {
  int A;
  struct {
    logic B;
    real C;
  } BC1, BC2;
} ABC = '{BC1: '{1'b1, 1.0}, int: 0, BC2: '{default: 0}};
The assignment pattern that initializes the struct variable ABC uses member, type, and default keys. The
vpiOperand traversal would represent this assignment pattern expression as:
'{0, '{1'b1, 1.0}, '{0, 0}}
or some other equivalent positional assignment pattern.
```
Surelog doesn't return expressions as if the assignment pattern were written with the positional notation:
```
   |vpiVariables:                                                                                                                                                                                                  
   \_struct_var: (work@dut.ABC), line:8:3, endln:8:6, parent:work@dut                                                                                                                                              
     |vpiName:ABC                                                                                                                                                                                                  
     |vpiFullName:work@dut.ABC                                                                                                                                                                                     
     |vpiAutomatic:1                                                                                                                                                                                               
     |vpiVisibility:1                                                                                                                                                                                              
     |vpiExpr:                                                                                                                                                                                                     
     \_operation: , line:8:9, endln:8:55                                                                                                                                                                           
       |vpiOpType:75                                                                                                                                                                                               
       |vpiOperand:                                                                                                                                                                                                
       \_tagged_pattern:                                                                                                                                                                                           
         |vpiPattern:                                                                                                                                                                                              
         \_operation: , line:8:16, endln:8:26                                                                                                                                                                      
           |vpiOpType:75                                                                                                                                                                                           
           |vpiOperand:                                                                                                                                                                                            
           \_constant: , line:8:18, endln:8:22                                                                                                                                                                     
             |vpiConstType:3                                                                                                                                                                                       
             |vpiDecompile:1'b1                                                                                                                                                                                    
             |vpiSize:1                                                                                                                                                                                            
             |BIN:1                                                                                                                                                                                                
           |vpiOperand:                                                                                                                                                                                            
           \_constant: , line:8:24, endln:8:25                                                                                                                                                                     
             |vpiConstType:9                                                                                                                                                                                       
             |vpiDecompile:1                                                                                                                                                                                       
             |vpiSize:64                                                                                                                                                                                           
             |UINT:1                                                                                                                                                                                               
         |vpiTypespec:                                                                                                                                                                                             
         \_string_typespec: (BC1), line:8:11, endln:8:14                                                                                                                                                           
           |vpiName:BC1   
       |vpiOperand:                                                                                      
       \_tagged_pattern: 
         |vpiPattern: 
         \_constant: , line:8:33, endln:8:34     
           |vpiConstType:9    
           |vpiDecompile:0                                                                               
           |vpiSize:64 
           |UINT:0        
         |vpiTypespec:                                                                                   
         \_int_typespec: , line:8:28, endln:8:31
       |vpiOperand:                                                                                      
       \_tagged_pattern: 
         |vpiPattern:     
         \_operation: , line:8:41, endln:8:54
           |vpiOpType:75
           |vpiOperand:
           \_tagged_pattern: 
             |vpiPattern:
             \_constant: , line:8:52, endln:8:53
               |vpiConstType:9
               |vpiDecompile:0
               |vpiSize:64
[... 20 lines cut ...]
           \_typespec_member: (B), line:5:11, endln:5:12
             |vpiName:B
             |vpiTypespec:
             \_logic_typespec: , line:5:5, endln:5:10
           |vpiTypespecMember:
           \_typespec_member: (C), line:6:9, endln:6:10
             |vpiName:C
             |vpiTypespec:
             \_int_typespec: , line:6:5, endln:6:8
       |vpiTypespecMember:
       \_typespec_member: (BC2), line:7:10, endln:7:13
         |vpiName:BC2
         |vpiTypespec:
         \_struct_typespec: , line:4:3, endln:4:9
           |vpiTypespecMember:
           \_typespec_member: (B), line:5:11, endln:5:12
             |vpiName:B
             |vpiTypespec:
             \_logic_typespec: , line:5:5, endln:5:10
           |vpiTypespecMember:
           \_typespec_member: (C), line:6:9, endln:6:10
             |vpiName:C
             |vpiTypespec:
             \_int_typespec: , line:6:5, endln:6:8
```

kamilrakoczy, 2021-09-15:
Also according to ``37.70 Case, pattern`` there are 3 types of patterns:
 - any pattern
 - tagged pattern
 - struct pattern
In above example we are assigning to the struct, so I think Surelog should return ``vpiStructPattern`` instead of ``vpiTaggedPattern``

alaindargelas, 2021-09-18:
Which testcase is blocked by this?

kamilrakoczy, 2021-09-20:
Real testcase:
```
package aes_pkg; 
typedef enum logic {
  AES_ENC = 1'b0,
  AES_DEC = 1'b1
} aes_op_e; 

typedef enum logic [3:0] {
  AES_ECB  = 4'b0001,               
  AES_CBC  = 4'b0010,
  AES_CTR  = 4'b0100,
  AES_NONE = 4'b1000
} aes_mode_e;

typedef enum logic [2:0] {
  AES_128 = 3'b001,
  AES_192 = 3'b010,
  AES_256 = 3'b100
} key_len_e;

typedef struct packed {
  aes_op_e   operation;
  aes_mode_e mode;
  key_len_e  key_len;
  logic      manual_operation;
} ctrl_reg_t;

parameter ctrl_reg_t CTRL_RESET = '{
  operation:        AES_ENC,
  mode:             AES_NONE,
  key_len:          AES_128,
  manual_operation: '0
};
endpackage

module test #(
parameter int DW = 32,
parameter logic [DW-1:0] TEST = '0
) ();

endmodule

module not_top ();

import aes_pkg::*;

test #(
  .DW($bits(ctrl_reg_t)),
  .TEST(CTRL_RESET)
) t ();

endmodule

module dut();

not_top m();

endmodule
```
UHDM:
```
...
|vpiParamAssign:                                                                                         
\_param_assign: , line:37:26, endln:37:35, parent:work@dut.m.t
  |vpiRhs:                        
  \_operation: , line:48:9, endln:48:19
    |vpiOpType:75                                                                                        
    |vpiOperand:          
    \_tagged_pattern:                                                                                    
[... 152 lines cut ...]
            |UINT:2
            |vpiSize:3
          |vpiEnumConst:
          \_enum_const: (AES_256), line:17:3, endln:17:10, parent:aes_pkg::key_len_e
            |vpiName:AES_256
            |vpiDecompile:3'b100
            |UINT:4
            |vpiSize:3
          |vpiInstance:
          \_package: aes_pkg (aes_pkg::) dut.sv:1:1: , endln:33:11, parent:work@dut
      |vpiTypespecMember:
      \_typespec_member: (manual_operation), line:24:14, endln:24:30, parent:aes_pkg::ctrl_reg_t
        |vpiName:manual_operation
        |vpiTypespec:
        \_logic_typespec: , line:24:3, endln:24:8, parent:manual_operation
          |vpiInstance:
          \_package: aes_pkg (aes_pkg::) dut.sv:1:1: , endln:33:11, parent:work@dut
      |vpiInstance:
      \_package: aes_pkg (aes_pkg::) dut.sv:1:1: , endln:33:11, parent:work@dut
  |vpiLhs:
  \_parameter: (work@dut.m.t.TEST), line:37:26, endln:37:35, parent:work@dut.m.t
  ...
```

We are getting assignment pattern on Rhs with Lhs that is just array.

kbieganski, 2021-09-21:
We consider it a blocker because it is a case found in OpenTitan.

Verilator is able to handle this using the typespec provided with the assignment pattern, but Yosys has no concept of assignment patterns.

Thus we'd have to implement in Yosys pretty much the same thing you're working on here. As this is needed anyway to support the standard, it would be smarter to do it once in Surelog and avoid duplicate work.

alaindargelas, 2021-09-21:
OK, I keep working on it.

alaindargelas, 2021-09-27:
Fixed by #2229

kamilrakoczy, 2021-10-01:
Thank you for working this.
Example from the standard is now working as expected, but when we instead of assigning struct when it is created, we create typedef and then assign to variable of this typedef, it is still returning ``vpiTaggedPattern`` without positional notation. 

Example:
```
module dut();                                                                                                                                                                                                      
typedef struct {                                                                                         
  int A;
  struct {
    logic B;
    int C;
  } BC1, BC2;
} ABC;

ABC test;

assign test = '{BC1: '{1'b1, 1}, int: 0, BC2: '{default: 0}};
endmodule
```
UHDM:
```   
|vpiContAssign:                                                                                       
   \_cont_assign: , line:12:8, endln:12:61, parent:work@dut                                                                                                                                                        
     |vpiRhs:                                                                                                                                                                                                      
     \_operation: , line:12:15, endln:12:61                                                                                                                                                                        
       |vpiOpType:75                                                                                     
       |vpiOperand:                                                                                                                                                                                                
       \_tagged_pattern:                                                                                                                                                                                           
         |vpiPattern:                                                                                                                                                                                              
         \_operation: , line:12:22, endln:12:32                                                                                                                                                                    
           |vpiOpType:75                                                                                                                                                                                           
           |vpiOperand:                                                                                                                                                                                            
           \_constant: , line:12:24, endln:12:28                                                                                                                                                                   
             |vpiConstType:3                                                                                                                                                                                       
             |vpiDecompile:1'b1                                                                                                                                                                                    
             |vpiSize:1                                                                                                                                                                                            
             |BIN:1                                                                                                                                                                                                
           |vpiOperand:                                                                                                                                                                                            
           \_constant: , line:12:30, endln:12:31                                                                                                                                                                   
             |vpiConstType:9                                                                             
             |vpiDecompile:1                                                                             
             |vpiSize:64                                                                                                                                                                                           
             |UINT:1                                                                                                                                                                                               
         |vpiTypespec:                                                                                   
         \_string_typespec: (BC1), line:12:17, endln:12:20                                                                                                                                                         
[... cut ...]
               |vpiConstType:9                                                                                                                                                                                     
               |vpiDecompile:0                      
               |vpiSize:64
               |UINT:0
             |vpiTypespec:
             \_string_typespec: (default), line:12:49, endln:12:56
               |vpiName:default
         |vpiTypespec:
         \_string_typespec: (BC2), line:12:42, endln:12:45
           |vpiName:BC2
     |vpiLhs:
     \_ref_obj: (work@dut.test), line:12:8, endln:12:12
       |vpiName:test
       |vpiFullName:work@dut.test
```

alaindargelas, 2021-10-01:
Open a different issue, code change will be different, test case is different. Do not reopen issues please

closed 2021-10-01 by alaindargelas