chipsalliance/Surelog #1866: Wrong typespec of pattern

RRozak, 2021-09-15:
When the following file is parsed, there is no ranges in the typespec of pattern:
```
module top;
   localparam int P [2] = '{11, 12};
endmodule
```
Here is uhdm:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module: work@top (work@top) pattern_type.sv:1:1: , endln:3:10, parent:work@top
  |vpiDefName:work@top
  |vpiFullName:work@top
  |vpiParameter:
  \_parameter: (work@top.P), line:2:19, endln:2:36, parent:work@top
    |vpiName:P
    |vpiFullName:work@top.P
    |vpiLocalParam:1
    |vpiSize:1
    |vpiRange:
    \_range: , line:2:22, endln:2:23, parent:work@top.P
      |vpiLeftRange:
      \_constant: , line:2:22, endln:2:23
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiRightRange:
      \_operation: 
        |vpiOpType:11
        |vpiOperand:
        \_constant: , line:2:22, endln:2:23
          |vpiConstType:9
          |vpiDecompile:2
          |vpiSize:64
          |UINT:2
        |vpiOperand:
        \_constant: 
          |vpiConstType:7
          |vpiSize:64
          |INT:1
    |vpiTypespec:
    \_int_typespec: (P), line:2:15, endln:2:18, parent:work@top.P
      |vpiName:P
  |vpiParamAssign:
  \_param_assign: , line:2:19, endln:2:36, parent:work@top
    |vpiRhs:
    \_operation: , line:2:27, endln:2:36
      |vpiOpType:75
      |vpiOperand:
      \_constant: , line:2:29, endln:2:31
        |vpiConstType:9
        |vpiDecompile:11
        |vpiSize:64
        |UINT:11
      |vpiOperand:
      \_constant: , line:2:33, endln:2:35
        |vpiConstType:9
        |vpiDecompile:12
        |vpiSize:64
        |UINT:12
    |vpiLhs:
    \_parameter: (work@top.P), line:2:19, endln:2:36, parent:work@top
|uhdmtopModules:
\_module: work@top (work@top) pattern_type.sv:1:1: , endln:3:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiParameter:
  \_parameter: (work@top.P), line:2:19, endln:2:36, parent:work@top
    |vpiName:P
    |vpiFullName:work@top.P
    |vpiLocalParam:1
    |vpiSize:1
    |vpiRange:
    \_range: , line:2:22, endln:2:23, parent:work@top.P
      |vpiLeftRange:
      \_constant: , line:2:22, endln:2:23
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiRightRange:
      \_operation: 
        |vpiOpType:11
        |vpiOperand:
        \_constant: , line:2:22, endln:2:23
          |vpiConstType:9
          |vpiDecompile:2
          |vpiSize:64
          |UINT:2
        |vpiOperand:
        \_constant: 
          |vpiConstType:7
          |vpiSize:64
          |INT:1
    |vpiTypespec:
    \_int_typespec: (P), line:2:15, endln:2:18, parent:work@top.P
      |vpiName:P
  |vpiParamAssign:
  \_param_assign: , line:2:19, endln:2:36, parent:work@top
    |vpiRhs:
    \_operation: , line:2:27, endln:2:36
      |vpiOpType:75
      |vpiOperand:
      \_constant: , line:2:29, endln:2:31
        |vpiConstType:9
        |vpiDecompile:11
        |vpiSize:64
        |UINT:11
      |vpiOperand:
      \_constant: , line:2:33, endln:2:35
        |vpiConstType:9
        |vpiDecompile:12
        |vpiSize:64
        |UINT:12
      |vpiTypespec:
      \_int_typespec: (P), line:2:15, endln:2:18                               <-------------
        |vpiName:P
    |vpiLhs:
    \_parameter: (work@top.P), line:2:19, endln:2:36, parent:work@top
```
I think that the typespec should contain the information about the type of the whole pattern, not the type of one of the elements. Am I right?

RRozak, 2021-09-15:
Another problem is the name of the typespec. It is not type name, but parameter name and it gives problems in verilator. In other places we created workarounds, but in the case of patterns it would be more complicated. You said that you can remove it: https://github.com/chipsalliance/Surelog/issues/1708#issuecomment-908856994
Could you do it?

alaindargelas, 2021-09-16:
Fixed by #2106

RRozak, 2021-09-16:
There is still `int typespec` without ranges in that line.
And the ranges of parameter are now duplicated. There are both in the typespec and directly under `parameter` node.

RRozak, 2021-09-16:
Here is uhdm:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module: work@top (work@top) pattern_type.sv:1:1: , endln:3:10, parent:work@top
  |vpiDefName:work@top
  |vpiFullName:work@top
  |vpiParameter:
  \_parameter: (work@top.P), line:2:19, endln:2:36, parent:work@top
    |vpiName:P
    |vpiFullName:work@top.P
    |vpiLocalParam:1
    |vpiSize:1
    |vpiRange:
    \_range: , line:2:22, endln:2:23, parent:work@top.P
      |vpiLeftRange:
      \_constant: , line:2:22, endln:2:23
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiRightRange:
      \_operation: 
        |vpiOpType:11
        |vpiOperand:
        \_constant: , line:2:22, endln:2:23
          |vpiConstType:9
          |vpiDecompile:2
          |vpiSize:64
          |UINT:2
        |vpiOperand:
        \_constant: 
          |vpiConstType:7
          |vpiSize:64
          |INT:1
    |vpiTypespec:
    \_array_typespec: 
      |vpiRange:
      \_range: , line:2:22, endln:2:23, parent:work@top.P
      |vpiElemTypespec:
      \_int_typespec: , line:2:15, endln:2:18, parent:work@top.P
  |vpiParamAssign:
  \_param_assign: , line:2:19, endln:2:36, parent:work@top
    |vpiRhs:
    \_operation: , line:2:27, endln:2:36
      |vpiOpType:75
      |vpiOperand:
      \_constant: , line:2:29, endln:2:31
        |vpiConstType:9
        |vpiDecompile:11
        |vpiSize:64
        |UINT:11
      |vpiOperand:
      \_constant: , line:2:33, endln:2:35
        |vpiConstType:9
        |vpiDecompile:12
        |vpiSize:64
        |UINT:12
    |vpiLhs:
    \_parameter: (work@top.P), line:2:19, endln:2:36, parent:work@top
|uhdmtopModules:
\_module: work@top (work@top) pattern_type.sv:1:1: , endln:3:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiParameter:
  \_parameter: (work@top.P), line:2:19, endln:2:36, parent:work@top
    |vpiName:P
    |vpiFullName:work@top.P
    |vpiLocalParam:1
    |vpiSize:1
    |vpiRange:
    \_range: , line:2:22, endln:2:23, parent:work@top.P
      |vpiLeftRange:
      \_constant: , line:2:22, endln:2:23
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiRightRange:
      \_operation: 
        |vpiOpType:11
        |vpiOperand:
        \_constant: , line:2:22, endln:2:23
          |vpiConstType:9
          |vpiDecompile:2
          |vpiSize:64
          |UINT:2
        |vpiOperand:
        \_constant: 
          |vpiConstType:7
          |vpiSize:64
          |INT:1
    |vpiTypespec:
    \_array_typespec: , parent:work@top.P
      |vpiRange:
      \_range: , line:2:22, endln:2:23
        |vpiLeftRange:
        \_constant: , line:2:22, endln:2:23
          |vpiConstType:9
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
        |vpiRightRange:
        \_operation: 
          |vpiOpType:11
          |vpiOperand:
          \_constant: , line:2:22, endln:2:23
            |vpiConstType:9
            |vpiDecompile:2
            |vpiSize:64
            |UINT:2
          |vpiOperand:
          \_constant: 
            |vpiConstType:7
            |vpiSize:64
            |INT:1
      |vpiElemTypespec:
      \_int_typespec: , line:2:15, endln:2:18
  |vpiParamAssign:
  \_param_assign: , line:2:19, endln:2:36, parent:work@top
    |vpiRhs:
    \_operation: , line:2:27, endln:2:36
      |vpiOpType:75
      |vpiOperand:
      \_constant: , line:2:29, endln:2:31
        |vpiConstType:9
        |vpiDecompile:11
        |vpiSize:64
        |UINT:11
      |vpiOperand:
      \_constant: , line:2:33, endln:2:35
        |vpiConstType:9
        |vpiDecompile:12
        |vpiSize:64
        |UINT:12
      |vpiTypespec:
      \_int_typespec: , line:2:15, endln:2:18
    |vpiLhs:
    \_parameter: (work@top.P), line:2:19, endln:2:36, parent:work@top
```

alaindargelas, 2021-09-16:
It is correct, you need to turn on the -d vpi_ids option so you can see the ids:

```
|uhdmtopModules:
\_module: work@top (work@top), id:40 tests/ScratchPad.sv:1:1: , endln:3:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiParameter:
  \_parameter: (work@top.P), id:41, line:2:19, endln:2:36, parent:work@top, parID:40
    |vpiName:P
    |vpiFullName:work@top.P
    |vpiLocalParam:1
    |vpiSize:1
    |vpiRange:
    \_range: , id:42, line:2:22, endln:2:23, parent:work@top.P, parID:41
      |vpiLeftRange:
      \_constant: , id:43, line:2:22, endln:2:23, parID:42
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiRightRange:
      \_operation: , id:44, parID:42
        |vpiOpType:11
        |vpiOperand:
        \_constant: , id:45, line:2:22, endln:2:23, parID:44
          |vpiConstType:9
          |vpiDecompile:2
          |vpiSize:64
          |UINT:2
        |vpiOperand:
        \_constant: , id:46, parID:44
          |vpiConstType:7
          |vpiSize:64
          |INT:1
    |vpiTypespec:
    \_array_typespec: , id:47, parent:work@top.P, parID:41           <<<<< That is the typespec of the LHS (Parameter), of type array of ints, this is what the bug really was about, the typespec of the parameter missing the ranges and is now fixed.
      |vpiRange:
      \_range: , id:49, line:2:22, endln:2:23, parID:47
        |vpiLeftRange:
        \_constant: , id:50, line:2:22, endln:2:23, parID:49
          |vpiConstType:9
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
        |vpiRightRange:
        \_operation: , id:51, parID:49
          |vpiOpType:11
          |vpiOperand:
          \_constant: , id:52, line:2:22, endln:2:23, parID:51
            |vpiConstType:9
            |vpiDecompile:2
            |vpiSize:64
            |UINT:2
          |vpiOperand:
          \_constant: , id:53, parID:51
            |vpiConstType:7
            |vpiSize:64
            |INT:1
      |vpiElemTypespec:
      \_int_typespec: , id:48, line:2:15, endln:2:18, parID:47
  |vpiParamAssign:
  \_param_assign: , id:16, line:2:19, endln:2:36, parent:work@top, parID:40
    |vpiRhs:
    \_operation: , id:33, line:2:27, endln:2:36, parID:16
      |vpiOpType:75
      |vpiOperand:
      \_constant: , id:34, line:2:29, endln:2:31, parID:33
        |vpiConstType:9
        |vpiDecompile:11
        |vpiSize:64
        |UINT:11
      |vpiOperand:
      \_constant: , id:35, line:2:33, endln:2:35, parID:33
        |vpiConstType:9
        |vpiDecompile:12
        |vpiSize:64
        |UINT:12
      |vpiTypespec:
      \_int_typespec: , id:36, line:2:15, endln:2:18, parID:33   <<< That is the typespec of the RHS (The pattern assignment), it does only reflect the type of the data member (int), it is not adjusted for what it assigns to, it that important?
    |vpiLhs:
    \_parameter: (work@top.P), id:41, line:2:19, endln:2:36, parent:work@top, parID:40

```

alaindargelas, 2021-09-16:
"And the ranges of parameter are now duplicated. There are both in the typespec and directly under parameter node." and this what we exactly want.

alaindargelas, 2021-09-16:
Mmm let me reconsider...

alaindargelas, 2021-09-16:
Is the Yosys plugin confused by the fact the parameter typespec contains the range of the unpacked array and the parameter itself has the range of the unpacked array? But reading the Standard yesterday again, I think the typespec has to be complete and contain both unpacked an packed dimensions so the typespec alone should specify the data type of the parameter. I also keep the range of the unpacked array on the parameter, which is harmless and I think is still compliant with the standard (But to be verified by a 2nd opinion).

alaindargelas, 2021-09-17:
#2110 should fix this issue

closed 2021-09-17 by alaindargelas

RRozak, 2021-09-17:
"Is the Yosys plugin confused by the fact the parameter typespec contains the range of the unpacked array and the parameter itself has the range of the unpacked array?" - Verilator is confused. But if the ranges are now in the typespec, maybe we can omit handling the ranges in parameter. I will check.

RRozak, 2021-09-17:
It works.
I see that `vpiTypespec` of pattern now reflects the type of whole pattern, not only the data member. It was important, because if pattern was the value of type struct, the typespec was struct, not the field of the struct. But in case of array, the typespec reflected the type of the array element. Now in both cases it reflects the type of whole pattern.
Thank you.