chipsalliance/Surelog #1894: No always statement in interface

RRozak, 2021-09-20:
Similar to https://github.com/chipsalliance/Surelog/issues/1872
There is no `always` node in uhdm.

Code:
```systemverilog
interface sw_test_status_if(output int x);
   always begin
      x = 10;
   end
endinterface

module top(output int o);
   sw_test_status_if u_sw(.x(o));
endmodule // top
```
Uhdm:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallInterfaces:
\_interface: work@sw_test_status_if (work@sw_test_status_if) interface_always.sv:1:1: , endln:5:13, parent:work@top
  |vpiDefName:work@sw_test_status_if
  |vpiFullName:work@sw_test_status_if
  |vpiPort:
  \_port: (x), line:1:40, parent:work@sw_test_status_if
    |vpiName:x
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@sw_test_status_if.x), line:1:40, parent:work@sw_test_status_if
        |vpiName:x
        |vpiFullName:work@sw_test_status_if.x
  |vpiNet:
  \_logic_net: (work@sw_test_status_if.x), line:1:40, parent:work@sw_test_status_if
|uhdmallModules:
\_module: work@top (work@top) interface_always.sv:7:1: , endln:9:10, parent:work@top
  |vpiDefName:work@top
  |vpiFullName:work@top
  |vpiPort:
  \_port: (o), line:7:23, parent:work@top
    |vpiName:o
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@top.o), line:7:23, parent:work@top
        |vpiName:o
        |vpiFullName:work@top.o
  |vpiNet:
  \_logic_net: (work@top.o), line:7:23, parent:work@top
|uhdmtopModules:
\_module: work@top (work@top) interface_always.sv:7:1: , endln:9:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiPort:
  \_port: (o), line:7:23, endln:7:24, parent:work@top
    |vpiName:o
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: (work@top.o)
      |vpiName:o
      |vpiFullName:work@top.o
      |vpiActual:
      \_int_var: (work@top.o), line:7:23, endln:7:24, parent:work@top
        |vpiName:o
        |vpiFullName:work@top.o
        |vpiAutomatic:1
        |vpiVisibility:1
  |vpiInterface:
  \_interface: work@sw_test_status_if (work@top.u_sw) interface_always.sv:8:4: , endln:8:34, parent:work@top
    |vpiDefName:work@sw_test_status_if
    |vpiDefFile:interface_always.sv
    |vpiDefLineNo:1
    |vpiName:u_sw
    |vpiFullName:work@top.u_sw
    |vpiPort:
    \_port: (x), line:1:40, endln:1:41, parent:work@top.u_sw
      |vpiName:x
      |vpiDirection:2
      |vpiHighConn:
      \_ref_obj: (work@top.o), line:8:30, endln:8:31
        |vpiName:o
        |vpiFullName:work@top.o
        |vpiActual:
        \_int_var: (work@top.o), line:7:23, endln:7:24, parent:work@top
      |vpiLowConn:
      \_ref_obj: (work@top.u_sw.x)
        |vpiName:x
        |vpiFullName:work@top.u_sw.x
        |vpiActual:
        \_int_var: (work@top.u_sw.x), line:1:40, endln:1:41, parent:work@top.u_sw
          |vpiName:x
          |vpiFullName:work@top.u_sw.x
          |vpiAutomatic:1
          |vpiVisibility:1
    |vpiInstance:
    \_module: work@top (work@top) interface_always.sv:7:1: , endln:9:10
    |vpiVariables:
    \_int_var: (work@top.u_sw.x), line:1:40, endln:1:41, parent:work@top.u_sw
  |vpiVariables:
  \_int_var: (work@top.o), line:7:23, endln:7:24, parent:work@top
```

alaindargelas, 2021-09-24:
Fixed by #2182

RRozak, 2021-09-24:
Now there is the same vpiAlways node in uhdmAllInterfaces and in uhdmtopModules:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallInterfaces:
\_interface: work@sw_test_status_if (work@sw_test_status_if) interface_always.sv:1:1: , endln:5:13, parent:work@top
  |vpiDefName:work@sw_test_status_if
  |vpiFullName:work@sw_test_status_if
  |vpiProcess:
  \_always: , line:2:4, endln:4:7, parent:work@top.u_sw
    |vpiAlwaysType:1
    |vpiStmt:
    \_begin: (work@top.u_sw), line:2:11, endln:4:7
      |vpiFullName:work@top.u_sw
      |vpiStmt:
      \_assignment: , line:3:7, endln:3:13, parent:work@top.u_sw
        |vpiOpType:82
        |vpiBlocking:1
        |vpiRhs:
        \_constant: , line:3:11, endln:3:13
          |vpiConstType:9
          |vpiDecompile:10
          |vpiSize:64
          |UINT:10
        |vpiLhs:
        \_ref_obj: (work@top.u_sw.x), line:3:7, endln:3:8, parent:work@top.u_sw
          |vpiName:x
          |vpiFullName:work@top.u_sw.x
          |vpiActual:
          \_int_var: (work@top.u_sw.x), line:1:40, endln:1:41, parent:work@top.u_sw
            |vpiName:x
            |vpiFullName:work@top.u_sw.x
            |vpiAutomatic:1
            |vpiVisibility:1
  |vpiPort:
  \_port: (x), line:1:40, parent:work@sw_test_status_if
    |vpiName:x
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@sw_test_status_if.x), line:1:40, parent:work@sw_test_status_if
        |vpiName:x
        |vpiFullName:work@sw_test_status_if.x
  |vpiNet:
  \_logic_net: (work@sw_test_status_if.x), line:1:40, parent:work@sw_test_status_if
|uhdmallModules:
\_module: work@top (work@top) interface_always.sv:7:1: , endln:9:10, parent:work@top
  |vpiDefName:work@top
  |vpiFullName:work@top
  |vpiPort:
  \_port: (o), line:7:23, parent:work@top
    |vpiName:o
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@top.o), line:7:23, parent:work@top
        |vpiName:o
        |vpiFullName:work@top.o
  |vpiNet:
  \_logic_net: (work@top.o), line:7:23, parent:work@top
|uhdmtopModules:
\_module: work@top (work@top) interface_always.sv:7:1: , endln:9:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiPort:
  \_port: (o), line:7:23, endln:7:24, parent:work@top
    |vpiName:o
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: (work@top.o)
      |vpiName:o
      |vpiFullName:work@top.o
      |vpiActual:
      \_int_var: (work@top.o), line:7:23, endln:7:24, parent:work@top
        |vpiName:o
        |vpiFullName:work@top.o
        |vpiAutomatic:1
        |vpiVisibility:1
  |vpiInterface:
  \_interface: work@sw_test_status_if (work@top.u_sw) interface_always.sv:8:4: , endln:8:34, parent:work@top
    |vpiDefName:work@sw_test_status_if
    |vpiDefFile:interface_always.sv
    |vpiDefLineNo:1
    |vpiName:u_sw
    |vpiFullName:work@top.u_sw
    |vpiProcess:
    \_always: , line:2:4, endln:4:7, parent:work@top.u_sw
      |vpiAlwaysType:1
      |vpiStmt:
      \_begin: (work@top.u_sw), line:2:11, endln:4:7
        |vpiFullName:work@top.u_sw
        |vpiStmt:
        \_assignment: , line:3:7, endln:3:13, parent:work@top.u_sw
          |vpiOpType:82
          |vpiBlocking:1
          |vpiRhs:
          \_constant: , line:3:11, endln:3:13
            |vpiConstType:9
            |vpiDecompile:10
            |vpiSize:64
            |UINT:10
          |vpiLhs:
          \_ref_obj: (work@top.u_sw.x), line:3:7, endln:3:8, parent:work@top.u_sw
            |vpiName:x
            |vpiFullName:work@top.u_sw.x
            |vpiActual:
            \_int_var: (work@top.u_sw.x), line:1:40, endln:1:41, parent:work@top.u_sw
              |vpiName:x
              |vpiFullName:work@top.u_sw.x
              |vpiAutomatic:1
              |vpiVisibility:1
    |vpiPort:
    \_port: (x), line:1:40, endln:1:41, parent:work@top.u_sw
      |vpiName:x
      |vpiDirection:2
      |vpiHighConn:
      \_ref_obj: (work@top.o), line:8:30, endln:8:31
        |vpiName:o
        |vpiFullName:work@top.o
        |vpiActual:
        \_int_var: (work@top.o), line:7:23, endln:7:24, parent:work@top
      |vpiLowConn:
      \_ref_obj: (work@top.u_sw.x)
        |vpiName:x
        |vpiFullName:work@top.u_sw.x
        |vpiActual:
        \_int_var: (work@top.u_sw.x), line:1:40, endln:1:41, parent:work@top.u_sw
    |vpiInstance:
    \_module: work@top (work@top) interface_always.sv:7:1: , endln:9:10
    |vpiVariables:
    \_int_var: (work@top.u_sw.x), line:1:40, endln:1:41, parent:work@top.u_sw
  |vpiVariables:
  \_int_var: (work@top.o), line:7:23, endln:7:24, parent:work@top
```

Could you remove one of them?

alaindargelas, 2021-09-24:
This is correct and follows the same semantic as allModules and topModules.

closed 2021-09-24 by alaindargelas

RRozak, 2021-09-28:
Could you tell what the semantic of allModules and topModules is? I have never seen a duplicate in case of modules.

alaindargelas, 2021-09-28:
"I have never seen a duplicate in case of modules"
Can you run 
```
interface sw_test_status_if(output int x);
   always begin
      x = 10;
   end
endinterface

module top(output int o);
   sw_test_status_if u_sw(.x(o));
  always begin
      o = 10;
   end
endmodule // top
```

"Could you tell what the semantic of allModules and topModules is?"
Please read:
https://github.com/chipsalliance/UHDM#model-concepts