chipsalliance/Surelog #1895: Surelog gives segmentation fault on bind statement

RRozak, 2021-09-20:
It is continuation of https://github.com/chipsalliance/Surelog/issues/1876.
There is seg fault on the following file:
```systemverilog
module rv_dm(input logic clk_i);
endmodule // rv_dm

module dmidpi(input logic clk_i);
endmodule // dmidpi

module top(output int o);
   bind rv_dm dmidpi u_dmidpi(.clk_i);
   assign o = 10;
endmodule // top
```

It is only the simple example to reproduce the problem. We need it in Opentitan. Segmentation fault is thrown there when `rv_dm` and `top_earlgrey_verilator` modules are parsed together by Surelog.
If you want to reproduce it you can remove that line: https://github.com/antmicro/verilator/blob/41cef450b52cc343d268e71019f4c6cb6c0f2bc7/uhdm-tests/opentitan/Makefile.in#L15 and run `make uhdm/verilator/opentitan TEST=tests/opentitan` in `uhdm-integration`.

The problem occurs also in that test: https://github.com/chipsalliance/UHDM-integration-tests/pull/472. My intention was to create case that doesn't require whole Opentitan. 
I don't know if in all these cases the problem is exactly the same, but it looks so.

alaindargelas, 2021-09-20:
Fixed by #2148

RRozak, 2021-09-21:
There is still segmentation fault on all these 3 cases.

alaindargelas, 2021-09-21:
@RRozak, I'm closing this issue which was for this testcase.
Please open one issue per testcase failing and give clear instructions on how Surelog needs to be run:
- What repository needs to be checkout out 
- which directory to run in
- a Surelog command line (No Makefile) to run and reproduce the problem.
Thanks

RRozak, 2021-09-21:
This testcase still gives segmentation fault.
I run Surelog with command:
`surelog -parse -sverilog -d coveruhdm <filename>`

RRozak, 2021-09-22:
I reopen this issue because the file from the beginning is still parsed with Seg fault.
And created the bigger test: https://github.com/chipsalliance/Surelog/issues/1910

alaindargelas, 2021-09-24:
Fixed by #2178

closed 2021-09-24 by alaindargelas