chipsalliance/Surelog #1904: No parameter assignment in instance initialization

RRozak, 2021-09-21:
When the following file is parsed by Surelog:
```systemverilog
package keymgr_pkg;
   typedef logic [63:0][5:0] lfsr_perm_t;
   parameter lfsr_perm_t RndCnstLfsrPermDefault = {
      128'h16108c9f9008aa37e5118d1ec1df64a7,
      256'h24f3f1b73537f42d38383ee8f897286df81d49ab54b6bbbb666cbd1a16c41252
   };
endpackage // keymgr_pkg

module keymgr;
   import keymgr_pkg::*;
   parameter lfsr_perm_t RndCnstLfsrPerm = RndCnstLfsrPermDefault;
endmodule // keymgr

module top;
   keymgr #(
      .RndCnstLfsrPerm(384'b1)
   ) u_keymgr ();
endmodule // top
```
the parameter of `u_keymgr` is initialized with the default value.
There are no corresponding nodes to `.RndCnstLfsrPerm(384'b1)` in uhdm.

Here is topModules section of uhdm:
```
|uhdmtopModules:
\_module: work@top (work@top) keymgr_simple.sv:14:1: , endln:18:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiModule:
  \_module: work@keymgr (work@top.u_keymgr) keymgr_simple.sv:15:4: , endln:17:18, parent:work@top
    |vpiDefName:work@keymgr
    |vpiDefFile:keymgr_simple.sv
    |vpiDefLineNo:9
    |vpiName:u_keymgr
    |vpiFullName:work@top.u_keymgr
    |vpiInstance:
    \_module: work@top (work@top) keymgr_simple.sv:14:1: , endln:18:10
    |vpiParameter:
    \_parameter: (work@top.u_keymgr.RndCnstLfsrPermDefault), line:3:26, endln:6:5, parent:work@top.u_keymgr
      |vpiName:RndCnstLfsrPermDefault
      |vpiFullName:work@top.u_keymgr.RndCnstLfsrPermDefault
      |vpiImported:keymgr_pkg
      |vpiTypespec:
      \_logic_typespec: (lfsr_perm_t), line:2:12, endln:2:17, parent:work@top.u_keymgr.RndCnstLfsrPermDefault
        |vpiName:lfsr_perm_t
        |vpiRange:
        \_range: , line:2:19, endln:2:23, parent:lfsr_perm_t
          |vpiLeftRange:
          \_constant: , line:2:19, endln:2:21
            |vpiConstType:9
            |vpiDecompile:63
            |vpiSize:64
            |UINT:63
          |vpiRightRange:
          \_constant: , line:2:22, endln:2:23
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
        |vpiRange:
        \_range: , line:2:25, endln:2:28, parent:lfsr_perm_t
          |vpiLeftRange:
          \_constant: , line:2:25, endln:2:26
            |vpiConstType:9
            |vpiDecompile:5
            |vpiSize:64
            |UINT:5
          |vpiRightRange:
[... 212 lines cut ...]
          |vpiDecompile:128'h16108c9f9008aa37e5118d1ec1df64a7
          |vpiSize:128
          |HEX:16108c9f9008aa37e5118d1ec1df64a7
        |vpiOperand:
        \_constant: , line:5:7, endln:5:76
          |vpiConstType:5
          |vpiDecompile:256'h24f3f1b73537f42d38383ee8f897286df81d49ab54b6bbbb666cbd1a16c41252
          |vpiSize:256
          |HEX:24f3f1b73537f42d38383ee8f897286df81d49ab54b6bbbb666cbd1a16c41252
      |vpiLhs:
      \_parameter: (work@top.u_keymgr.RndCnstLfsrPermDefault), line:3:26, endln:6:5, parent:work@top.u_keymgr
    |vpiParamAssign:
    \_param_assign: , line:11:26, endln:11:66, parent:work@top.u_keymgr
      |vpiRhs:
      \_ref_obj: (RndCnstLfsrPermDefault), line:11:44, endln:11:66
        |vpiName:RndCnstLfsrPermDefault
        |vpiActual:
        \_parameter: (work@top.u_keymgr.RndCnstLfsrPermDefault), line:3:26, endln:6:5, parent:work@top.u_keymgr
      |vpiLhs:
      \_parameter: (work@top.u_keymgr.RndCnstLfsrPerm), line:11:26, endln:11:66, parent:work@top.u_keymgr
    |vpiTypedef:
    \_logic_typespec: (keymgr_pkg::lfsr_perm_t), line:2:12, endln:2:17
    |vpiTypedef:
    \_import: (keymgr_pkg), line:10:11, endln:10:24
```

RRozak, 2021-09-21:
I have a question regarding `vpiInstance` node. I ask here, because we have an example uhdm.
Why the instance subtree contains the whole copy of the vpiPackage? Is it desired?

alaindargelas, 2021-09-24:
To the question:
"I have a question regarding vpiInstance node. I ask here, because we have an example uhdm.
Why the instance subtree contains the whole copy of the vpiPackage? Is it desired?".

This is simply the UHDM printer that is printing relations in a depth first search way.
The vpiInstance contains a link to the package, then the package is printed in a depth first search manner. So you think there is a copy, but not really.

alaindargelas, 2021-09-27:
Fixed with #2204

closed 2021-09-27 by alaindargelas