chipsalliance/Surelog #1908: Surelog for packaging

aolofsson, 2021-09-21:
We are currently using verilator for pre-processing and elaboration to import some design files:

verilator --debug
cat obj_dir/*.vpp > pickle.v

Seems that Surelog already has the information needed to do this, but it is not be currently accessible by the user without a UHDM analyzer. Would it be possible to output the a list that includes all the files starting with the design topmodule (after pre-processing). This would allow us to switch to surelog to do the pickeling (collecting) of all the pre-processed files in place of verilator.

alaindargelas, 2021-09-23:
Will be working on this next

alaindargelas, 2021-09-23:
Fixed by #2174

Look for slpp_all/file_elab.lst that gives you the list of files needed to elaborate the design (subset of slpp_all/file.lst which is the list of all the files parsed and preprocessed)

closed 2021-09-23 by alaindargelas