chipsalliance/Surelog #1918: Changing meaning of -y to match Icarus/Verilator

aolofsson, 2021-09-23:
It seems that tools like verilator and Icarus have a different interpretation of the the -y directive.
Not sure if the -y directive is actually standardized, I couldn't find a reference in the Verilog 2005 standard.

They don't pre-process all the files in that directory, they only use it to resolve undefined modules. 
Both verilator, icaruus have a switch for manually specifying the top level module in case of multiple modules specified. With the top of the tree guaranteed, it seems possible to only pre-process files from the -y directory when an undefined module is encountered.
 
What do you think about having surelog behave the same as verilator and icarus (and other tools) wrt to the -y and -I switches?

There are instances when I just want to grab a small module from a large directory,  and it would seem a shame to pre-process all files, when I am only using 1/100 files in that direcotory.

Verilator example:

verilator  $OHOME/stdlib/hdl/oh_fifo_async.v -f libs.cmd --lint-only

Man page from icarus (iverilog) for reference:

```

iverilog(1)                                                                                 Version 10.3 (stable)                                                                                 iverilog(1)

NAME
       iverilog - Icarus Verilog compiler

SYNOPSIS
       iverilog  [-ESVv]  [-Bpath]  [-ccmdfile|-fcmdfile]  [-Dmacro[=defn]]  [-Pparameter=value] [-pflag=value] [-dname] [-g1995|-g2001|-g2005|-g2005-sv|-g2009|-g2012|-g<feature>] [-Iincludedir] [-mmodule]
       [-M[mode=]file] [-Nfile] [-ooutputfilename] [-stopmodule] [-ttype] [-Tmin/typ/max] [-Wclass] [-ypath] [-lfile] sourcefile

DESCRIPTION
       iverilog is a compiler that translates Verilog source code into executable programs for simulation, or other netlist formats for further processing. The currently supported targets are vvp for simu‐
       lation, and fpga for synthesis. Other target types are added as code generators are implemented.

OPTIONS
       iverilog accepts the following options:

       -Bbase  The  iverilog  program  uses  external programs and configuration files to preprocess and compile the Verilog source. Normally, the path used to locate these tools is built into the iverilog
               program. However, the -B switch allows the user to select a different set of programs. The path given is used to locate ivlpp, ivl, code generators and the VPI modules.

       -cfile -ffile
               These flags specify an input file that contains a list of Verilog source files. This is similar to the command file of other Verilog simulators, in that it is a file that contains  the  file
               names instead of taking them on the command line. See Command Files below.

       -Dmacro Defines macro macro with the string `1' as its definition. This form is normally only used to trigger ifdef conditionals in the Verilog source.

       -Dmacro=defn
               Defines macro macro as defn.

       -Pparameter=value
               Override  (i.e. defparam) a parameter in a root module. This allows the user to override at compile time (defparam) a parameter in a root module instance. For example, -Pmain.foo=2 overrides
               the parameter foo in the root instance main with the value 2.

       -dname  Activate a class of compiler debugging messages. The -d switch may be used as often as necessary to activate all the desired messages.  Supported names are scopes, eval_tree, elaborate,  and
               synth2; any other names are ignored.

       -E      Preprocess  the  Verilog source, but do not compile it. The output file is the Verilog input, but with file inclusions and macro references expanded and removed. This is useful, for example,
               to preprocess Verilog source for use by other compilers.

       -g1995|-g2001|-g2001-noconfig|-g2005|-g2005-sv|-g2009|-g2012
               Select the Verilog language generation to support in the compiler.  This selects between IEEE1364-1995, IEEE1364-2001, IEEE1364-2005, IEEE1800-2005, IEEE1800-2009, or IEEE1800-2012.   Icarus
               Verilog  currently defaults to the IEEE1364-2005 generation of the language. This flag is used to restrict the language to a set of keywords/features, this allows simulation of older Verilog
               code that may use newer keywords and for compatibility with other tools. Much of the IEEE1800 generations functionality is not currently supported.  The IEEE1800 generations do parse all the
               keywords, so they can be used to verify that IEEE1364 compliant Verilog code does not use any of the new IEEE1800 keywords.

       -gverilog-ams|-gno-verilog-ams
               Enable or disable (default) support for Verilog-AMS.  Very little Verilog-AMS specific functionality is currently supported.

       -gspecify|-gno-specify
               Enable  or  disable  (default)  specify  block  support. When enabled, specify block code is elaborated. When disabled, specify blocks are parsed but ignored. Specify blocks are commonly not
               needed for RTL simulation, and in fact can hurt performance of the simulation. However, disabling specify blocks reduces accuracy of full-timing simulations.

       -gstd-include|-gno-std-include
[... 247 lines cut ...]
            iverilog hello.v

       To compile hello.v to an executable file called hello:

            iverilog -o hello hello.v

       To compile and run explicitly using the vvp runtime:

            iverilog -ohello.vvp -tvvp hello.v

AUTHOR
       Steve Williams (steve@icarus.com)

SEE ALSO
       vvp(1), <http://iverilog.icarus.com/>

       Tips on using, debugging, and developing the compiler can be found at <http://iverilog.wikia.com/>

COPYRIGHT
       Copyright ©  2002-2016 Stephen Williams

       This document can be freely redistributed according to the terms of the
       GNU General Public License version 2.0

                                                                                                Oct 2nd, 2016                                                                                     iverilog(1)

alaindargelas, 2021-09-23:
This would be unfortunately a huge change in the way Surelog compilation flow works.
Surelog has a pretty simple work flow: 

```
Preprocess all,   (Can be parallel in case of -fileunit)
Parse all,            (Can be parallel with -mt or -mp )
Compile all, 
Elaborate, 
Compile Generate,    <-|   loop
Elaborate Generate,   --|
Produce UHDM.
```

To achieve the preprocessing time saving you are describing, especially with the new features of Verilog 2001 and SystemVerilog, is a lot of work for no real gain in the majority of the cases (Besides the "wasted" time spent preprocessing unsused library module).

Let me explain:
1) The cells in those library directories do need to get preprocessed to be parsed, they do contain preprocessing directives:

```
oh/stdlib/hdl$ grep "\`" oh_*
oh_mux4.v:`ifdef TARGET_SIM
oh_mux4.v:`endif
oh_pll.v:`ifdef TARGET_SIM
oh_pll.v:`endif
...
```

2) In order to know which cell is "only required" to load for elaboration, you need to actually do the elaboration, especially with SystemVerilog, you can't write a quick and dirty parser (grep style) that finds module instantiations and compute what is needed to elaborate the design, generate statements make it almost impossible to do a static analysis of the source code to compute a hierarchy tree quickly. You need to actually compile, elaborate, unroll the generates, then compile some more and elaborate some more to unroll the hierarchy. If the library cells are are the leaf of the elaboration, you'd have to start preprocessing and parsing those cells in the middle of the elaboration pass, which is feasible, but not really clean and easy flow.

3) In a large design, the ratio of unused library cells should not be 1 to 100. So maybe the "wasted" time is more apparent for a unit test, but no so much on a real design. 

I'd have to decline that request if it's just on the account of the wasted runtime. On the other hand if you see a difference in functionality of the resulting design, that is another conversation.

You mention -I, is there another problem there?

alaindargelas, 2021-09-23:
Also, -v <libfile> might be a better alternative than -y <libdir> if you need to handpick only a few files from a directory.

aolofsson, 2021-09-24:
Thanks for the explanation, that's fair! I think there might still be an issue here wrt to the top module though? Maybe there is a better way, but this is the way I have been working with verilator for since 2008 and the command files are built around some assumptions of how -y and -I works. I can always start from scratch and refactor as needed (not a big deal for me), but others might not be as friendly.:-)

Seems like there is a restriction on not having a top level module in a directory that is searched with -y? With the -top switch, the top level name should be known right? 

Working verilator example:
verilator $OHOME/stdlib/hdl/oh_fifo_async.v -f -y $OHOME/stdlib/hdl --lint-only 

```
  >> surelog  $OHOME/stdlib/hdl/oh_fifo_async.v -I${OHOME}/stdlib/hdl/ -y ${OHOME}/stdlib/hdl/ -parse  -nobuiltin -top oh_fifo_async

[FTL:EL0533] Cmd line top level module does not exist "oh_fifo_async".
[NTE:EL0508] Nb Top level modules: 0.
[NTE:EL0509] Max instance depth: 0.
[NTE:EL0510] Nb instances: 0.
[NTE:EL0511] Nb leaf instances: 0.
```

```
>>> surelog  $OHOME/stdlib/hdl/oh_fifo_async.v -I${OHOME}/stdlib/hdl/  -parse  -nobuiltin -top oh_fifo_async

[NTE:EL0508] Nb Top level modules: 1.
[NTE:EL0509] Max instance depth: 2.
[NTE:EL0510] Nb instances: 18.
[NTE:EL0511] Nb leaf instances: 17.
[WRN:EL0512] Nb undefined modules: 4.
[WRN:EL0513] Nb undefined instances: 17.

alaindargelas, 2021-09-24:
Surelog behavior is correct, I think if your try commercial simulators, they'll agree: 
the cell oh_fifo_async is a cell if you have it in the -y directory, a cell cannot be a top module, period.
When you invoke the 2nd command line without the -y, oh_fifo_async is treated as a module and it is the top, but you can't find the other components.

With the 1rst command line, the cell oh_fifo_async is part of the "cells" because of the -y, moreover, it is used inside another cell: 
~/oh/stdlib/hdl$ grep oh_fifo_async *
oh_fifo_cdc.v:   oh_fifo_async  #(.SYN(SYN),
oh_fifo_cdc.v:   oh_fifo_async (

You need to provide a real module as the top module (Not sitting in a -y directory, neither specified with -v). Or a list of modules, and pick the one you want with -top

alaindargelas, 2021-09-24:
Mmm I'll need to eat my socks.... 
I just tried on edaplayground.com

Cmdline:    -timescale=1ns/1ns +vcs+flush+all +warn=all -sverilog -y . +libext+.v test.v

Empty design.sv and testbench.sv
and a module in test.v and VCS compiles the module test.v even it is in the same dir as the -y .

OK, I can relax that rule, that a specified module on the command line supersedes the -y cell.

alaindargelas, 2021-09-24:
Fixed by #2186

closed 2021-09-24 by alaindargelas