chipsalliance/Surelog #1952: Range of queue is parsed wrong

RRozak, 2021-09-28:
According to 37.22.1, a range corresponding to queue should be empty. 
Now it is parsed like array range.
Code:
```systemverilog
module top;
   int q[$];
   initial begin
      q.push_back(1);
   end
endmodule // top
```
uhdm:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module: work@top (work@top) queue.sv:1:1: , endln:6:10, parent:work@top
  |vpiDefName:work@top
  |vpiFullName:work@top
  |vpiProcess:
  \_initial: , parent:work@top
    |vpiStmt:
    \_begin: (work@top), line:3:12, endln:5:7
      |vpiFullName:work@top
      |vpiStmt:
      \_method_func_call: (push_back), line:4:7, endln:4:22, parent:work@top
        |vpiName:push_back
        |vpiPrefix:
        \_ref_obj: (q)
          |vpiName:q
        |vpiArgument:
        \_constant: , line:4:19, endln:4:20
          |vpiConstType:9
          |vpiDecompile:1
          |vpiSize:64
          |UINT:1
  |vpiNet:
  \_logic_net: (work@top.q), line:2:8, parent:work@top
    |vpiName:q
    |vpiFullName:work@top.q
|uhdmtopModules:
\_module: work@top (work@top) queue.sv:1:1: , endln:6:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiVariables:
  \_array_var: (work@top.q), line:2:8, endln:2:9, parent:work@top
    |vpiArrayType:1
    |vpiName:q
    |vpiFullName:work@top.q
    |vpiAutomatic:1
    |vpiRandType:1
    |vpiVisibility:1
    |vpiSize:1
    |vpiReg:
    \_int_var: (work@top.q), parent:work@top.q
      |vpiFullName:work@top.q
    |vpiRange:
    \_range: , line:2:10, endln:2:11
      |vpiLeftRange:
      \_constant: , line:2:10, endln:2:11
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiRightRange:
      \_constant: , line:2:10, endln:2:11
        |vpiConstType:10
        |vpiDecompile:$
        |vpiSize:1
        |STRING:$
```

alaindargelas, 2021-09-29:
This is more explicit and should not be a problem to understand by the client application. I think the Standard in this case is not explicit enough.

closed 2021-09-29 by alaindargelas

RRozak, 2021-09-29:
What do you think about marking that vpiRange with special field? Now it looks like any other range and the difference is not visible until the constant of range is parsed.

alaindargelas, 2021-09-29:
I don't think that is necessary.
I'm planning to also treat int q[] = { ... }  with a range marked 0 : "unsized"