chipsalliance/Surelog #1956: Associative array parsed as int

RRozak, 2021-09-28:
In the following file, the associative array `a` is parsed as `int_var` and element selection is parsed as `bit_select`.
```systemverilog
module top;
   int a [int];
   assign a[5] = 5;
endmodule
```
Here is uhdm:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module: work@top (work@top) associative_array.sv:1:1: , endln:4:10, parent:work@top
  |vpiDefName:work@top
  |vpiFullName:work@top
  |vpiContAssign:
  \_cont_assign: , line:3:11, endln:3:19, parent:work@top
    |vpiRhs:
    \_constant: , line:3:18, endln:3:19
      |vpiConstType:9
      |vpiDecompile:5
      |vpiSize:64
      |UINT:5
    |vpiLhs:
    \_bit_select: (work@top.a), line:3:11, endln:3:12, parent:work@top.a
      |vpiName:a
      |vpiFullName:work@top.a
      |vpiIndex:
      \_constant: , line:3:13, endln:3:14, parent:work@top.a
        |vpiConstType:9
        |vpiDecompile:5
        |vpiSize:64
        |UINT:5
  |vpiNet:
  \_logic_net: (work@top.a), line:2:8, parent:work@top
    |vpiName:a
    |vpiFullName:work@top.a
|uhdmtopModules:
\_module: work@top (work@top) associative_array.sv:1:1: , endln:4:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiVariables:
  \_array_var: (work@top.a), line:2:8, endln:2:9, parent:work@top
    |vpiArrayType:1
    |vpiName:a
    |vpiFullName:work@top.a
    |vpiAutomatic:1
    |vpiRandType:1
    |vpiVisibility:1
    |vpiSize:1
    |vpiReg:
    \_int_var: (work@top.a), parent:work@top.a
      |vpiFullName:work@top.a
```

alaindargelas, 2021-10-02:
Fixed by #2309

RRozak, 2021-10-05:
Maybe it would be better to change the `vpiArrayType?` `3` corresponds to `vpiAssocArray`.
I didn't noticed it when I was creating this issue.

RRozak, 2021-10-05:
The problem is also with other array types, so I created another issue for that: https://github.com/chipsalliance/Surelog/issues/2012

closed 2021-10-05 by RRozak