chipsalliance/Surelog #1979: Use std::filesystem instead of manhandling paths as strings

hs-apotell, 2021-09-30:
We are noticing quite a few issues emerging from use of strings to manage paths. Here are few instances,

* https://github.com/chipsalliance/Surelog/blob/094f1ad629c34d38c0c578f1a8c913eb8d7c1d82/src/SourceCompile/CompileSourceFile.cpp#L94
* https://github.com/chipsalliance/Surelog/blob/094f1ad629c34d38c0c578f1a8c913eb8d7c1d82/src/SourceCompile/CompileSourceFile.cpp#L126

Also, Windows limit of 260 characters applies when using strings are arguments to classes like std::ofstream but it doesn't when using std::fs::path. We are running into the path limit on Windows with certain tests (most recently with BlackParrot) and using std::fs seems to resolve it. Unfortunately, the documented solution to fix the path limit doesn't seem to work for Surelog.

hzeller, 2021-09-30:
+1 for using fs::path to assemble file names.

alaindargelas, 2021-10-01:
The BlackParrot test path is not a Surelog specific issue, it is a BlackParrot test structure issue.

alaindargelas, 2021-10-07:
Yup, please update the minimum gcc, send a PR.

hs-apotell, 2021-11-18:
For sake of documentation - Found another reason justifying this change. Surelog calling external processes using `system` command and using backslashes on Msys2 causes failure. Specific test case failing as of today is `DiffSimpleIncludeAndMacros` where the input to the `diff` command isn't being found because the input output directory on the command-line was using backslashes.

alaindargelas, 2021-12-01:
Fixed by #2253

closed 2021-12-01 by alaindargelas