chipsalliance/Surelog #1980: Path limit on Windows

hs-apotell, 2021-09-30:
The documented method to raise the path limit on Windows doesn't seem to work. Without unlocking this limit lots of tests start failing on Windows.

https://docs.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=cmd

If possible, make the solution part of the Surelog build rather than expecting every user to go make this change.

alaindargelas, 2021-10-01:
We fixed this long ago, the directory generated in the cache is a hash of the path name.
Why lots of tests start failing?
This was working for a long time.

Can you add here an example of a path failing?

hs-apotell, 2021-10-01:
It's not a CI failure. It is specific to location of the Surelog repository on local machines. If the directory is already deep, combining relative paths of test folders to location of the repository generates paths that are too long.

Plus, the back-slash vs. front-slash issue also showed up as a problem in recent tests. Certain places in code use relative paths to known files (like builtin.sv) using front-slash. These conditional-if conditions fails because on Windows those paths are back-slashes - and a string match (using strstr) would fail. Paths cannot be compared using simple string functions.

alaindargelas, 2021-10-01:
"It's not a CI failure. It is specific to location of the Surelog repository on local machines. If the directory is already deep, combining relative paths of test folders to location of the repository generates paths that are too long."

Please use the surelog .....  -o <some temp path> and maybe modify regression.tcl to automatically give you a tmp path or surelog   so all tests output in Windows go in a short temp path instead of the arbitrarily long regression directory structure which we don't have control on.

https://stackoverflow.com/questions/278439/creating-a-temporary-directory-in-windows
Submit a PR.

"Plus, the back-slash vs. front-slash issue also showed up as a problem in recent tests. Certain places in code use relative paths to known files (like builtin.sv) using front-slash. These conditional-if conditions fails because on Windows those paths are back-slashes - and a string match (using strstr) would fail. Paths cannot be compared using simple string functions."

Simply submit a PR.

closed 2021-10-01 by alaindargelas