chipsalliance/Surelog #2012: Problems with types of the arrays

RRozak, 2021-10-05:
`vpiArrayType` field of `vpiArrayVar` is unused.
Here is code with examples of dynamic array, associative array and queue:
```systemverilog
module top;
   int dynamic [];
   int assoc [int];
   int queue [$];
endmodule;
```
And here is uhdm:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module: work@top (work@top) array_types.sv:1:1: , endln:5:10, parent:work@top
  |vpiDefName:work@top
  |vpiFullName:work@top
  |vpiNet:
  \_logic_net: (work@top.dynamic), line:2:8, parent:work@top
    |vpiName:dynamic
    |vpiFullName:work@top.dynamic
  |vpiNet:
  \_logic_net: (work@top.assoc), line:3:8, parent:work@top
    |vpiName:assoc
    |vpiFullName:work@top.assoc
  |vpiNet:
  \_logic_net: (work@top.queue), line:4:8, parent:work@top
    |vpiName:queue
    |vpiFullName:work@top.queue
|uhdmtopModules:
\_module: work@top (work@top) array_types.sv:1:1: , endln:5:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiVariables:
  \_int_var: (work@top.dynamic), line:2:8, endln:2:15, parent:work@top
    |vpiName:dynamic
    |vpiFullName:work@top.dynamic
    |vpiAutomatic:1
    |vpiVisibility:1
  |vpiVariables:
  \_array_var: (work@top.assoc), line:3:8, endln:3:13, parent:work@top
    |vpiArrayType:1
    |vpiName:assoc
    |vpiFullName:work@top.assoc
    |vpiAutomatic:1
    |vpiRandType:1
    |vpiVisibility:1
    |vpiSize:1
    |vpiReg:
    \_int_var: (work@top.assoc), parent:work@top.assoc
      |vpiFullName:work@top.assoc
    |vpiRange:
    \_range: 
      |vpiLeftRange:
      \_constant: , line:3:14, endln:3:15
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiRightRange:
      \_constant: , line:3:14, endln:3:15
        |vpiConstType:6
        |vpiDecompile:associative
        |STRING:associative
        |vpiTypespec:
        \_int_typespec: , line:3:15, endln:3:18
  |vpiVariables:
  \_array_var: (work@top.queue), line:4:8, endln:4:13, parent:work@top
    |vpiArrayType:1
    |vpiName:queue
    |vpiFullName:work@top.queue
    |vpiAutomatic:1
    |vpiRandType:1
    |vpiVisibility:1
    |vpiSize:1
    |vpiReg:
    \_int_var: (work@top.queue), parent:work@top.queue
      |vpiFullName:work@top.queue
    |vpiRange:
    \_range: , line:4:15, endln:4:16
      |vpiLeftRange:
      \_constant: , line:4:15, endln:4:16
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiRightRange:
      \_constant: , line:4:15, endln:4:16
        |vpiConstType:10
        |vpiDecompile:$
        |vpiSize:1
        |STRING:$
```
Dynamic array is parsed as `int_var` and other arrays have `vpiArrayType: 1` and the information of their types is stored in `vpiRightRange`. It is against the standard and makes the handling of them more difficult. 37.17.21 says about possible types.

alaindargelas, 2021-10-05:
Good catch. I don't know how I missed that field. That would work better

alaindargelas, 2021-10-06:
Fixed by #2345

closed 2021-10-06 by alaindargelas