chipsalliance/Surelog #2013: Windows build on CI has been failing

hs-apotell, 2021-10-05:
Is anyone attending to this? The issue is in flatbuffers post build action. Here's the fix -

```
$ git diff CMakeLists.txt
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 4c2dc9ba..61531c62 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -496,6 +496,7 @@ if(PYTHONINTERP_FOUND AND
     TARGET flatc
     POST_BUILD
     COMMAND ${GENERATION_SCRIPT} "${FLATBUFFERS_FLATC_EXECUTABLE}"
+    WORKING_DIRECTORY "${CMAKE_CURRENT_SOURCE_DIR}"
     COMMENT "Generating code..."
     VERBATIM)
 endif()

```

hzeller, 2021-10-05:
Can you send a pull request ?

hs-apotell, 2021-10-06:
https://github.com/google/flatbuffers/pull/6878 fixes it.

hs-apotell, 2021-10-06:
@alaindargelas Could you update the flatbuffer submodule please. Thanks!

closed 2021-10-06 by hs-apotell

alaindargelas, 2021-10-06:
Done