chipsalliance/Surelog #2167: Problems with parameter substitution in nested modules

RRozak, 2021-11-03:
There is a problem when hierarchy is deeper than 2 and the parameter of top module isn't substituted, because it is not visible from the deepest submodules.
Here is the example code:
```systemverilog
package prim_pad_wrapper_pkg;
   typedef enum logic {
      BidirStd = 1'h1
   } pad_type_e;
endpackage : prim_pad_wrapper_pkg

package pinmux_pkg;
   import prim_pad_wrapper_pkg::*;
   typedef struct packed {
      pad_type_e dio_pad_type;
   } target_cfg_t;
endpackage : pinmux_pkg

module prim_generic_pad_attr(output logic o);
   import prim_pad_wrapper_pkg::*;
   parameter pad_type_e PadTypeInGeneric = 0;
   assign o = PadTypeInGeneric;
endmodule : prim_generic_pad_attr

module prim_submodule(output logic o);
   import prim_pad_wrapper_pkg::*;
   parameter pad_type_e PadType = 0;

   prim_generic_pad_attr #(
       .PadTypeInGeneric(PadType)
   ) u_impl_generic(
      .o(o)
   );
endmodule

module prim_pad_attr(output logic o);
   import prim_pad_wrapper_pkg::*;
   parameter pad_type_e PadType = 0;

   prim_submodule #(
      .PadType(PadType)
   ) u_submodule(
      .o(o)
   );
endmodule

module top(output logic o);
   import pinmux_pkg::*;
   parameter target_cfg_t TargetCfg = 1;

   prim_pad_attr #(
      .PadType(TargetCfg.dio_pad_type)
   ) u_prim_pad_attr(
      .o(o)
   );
endmodule
```
The uhdm tree is pretty long, lets consider the tree printed by uhdm-verilator:
```
     NETLIST 0x555556cb2000 <e1#> {a0aa}  $root [1ps/1ps]
    1: MODULE 0x555556cbd680 <e246#> {i0aa}  top  L2 [1ps]
    1:2: PORT 0x555556ce6820 <e103#> {d42az}  o
    1:2: VAR 0x555556c929a0 <e101#> {d42az} @dt=0@  o OUTPUT PORT
    1:2:1: BASICDTYPE 0x555556ce6750 <e99#> {d42az} @dt=this@(nw1)  logic kwd=logic
    1:2: PACKAGEIMPORT 0x555556cd87e0 <e104#> {d43al}  * -> PACKAGE 0x555556cbcc60 <e252#> {e0aa}  pinmux_pkg  L3 [LIB] [1ps]
    1:2: CELL 0x555556cba3c0 <e218#> {d50ae}  u_prim_pad_attr -> MODULE 0x555556cbd8c0 <e253#> {g0aa}  prim_pad_attr_u_prim_pad_attr0  L3 [1ps]
    1:2:1: PIN 0x555556cb32c0 <e208#> {d50ae}  o ->UNLINKED
    1:2:1:1: PARSEREF 0x555556ce7ba0 <e209#> {d49ak}  o [TEXT]
    1:2:2: PIN 0x555556cb33b0 <e215#> {d50ae}  PadType ->UNLINKED
    1:2:2:1: DOT 0x555556cf44d0 <e216#> {d47aq}
    1:2:2:1:1: PARSEREF 0x555556ce7c70 <e213#> {j0aa}  TargetCfg [TEXT]
    1:2:2:1:2: PARSEREF 0x555556ce7d40 <e214#> {j0aa}  dio_pad_type [TEXT]
[... 120 lines cut ...]
    1:2:1: PIN 0x555556cb2b40 <e91#> {c27ai}  o ->UNLINKED
    1:2:1:1: PARSEREF 0x555556d00b60 <e92#> {c27ak}  o [TEXT]
    1:2:2: PIN 0x555556cb2c30 <e86#> {c25aj}  PadTypeInGeneric ->UNLINKED
    1:2:2:1: PARSEREF 0x555556d00c30 <e87#> {c25ba}  PadType [TEXT]
    1: MODULE 0x555556cbc6c0 <e186#> {c14ai}  prim_generic_pad_attr  L5 [1ps]
    1:2: PORT 0x555556d00340 <e37#> {c14br}  o
    1:2: VAR 0x555556c92160 <e40#> {c14br} @dt=0@  o OUTPUT PORT
    1:2:1: BASICDTYPE 0x555556d00410 <e39#> {c14bl} @dt=this@(nw1)  logic kwd=logic
    1:2: PACKAGEIMPORT 0x555556cd8380 <e60#> {c15bf}  * -> PACKAGE 0x555556cbc120 <e187#> {c1aj}  prim_pad_wrapper_pkg  L6 [LIB] [1ps]
    1:2: VAR 0x555556c922c0 <e53#> {c16az} @dt=0@  PadTypeInGeneric GPARAM
    1:2:1: REFDTYPE 0x555556cb2780 <e51#> {c16ao} @dt=0@  pad_type_e -> UNLINKED
    1:2:3: CONST 0x555556cbc7e0 <e52#> {c16bs} @dt=0x555556d004e0@(G/swu32/1)  ?32?sh0
    1:2: ASSIGNW 0x555556d04000 <e59#> {c17an} @dt=0@
    1:2:1: PARSEREF 0x555556d00680 <e57#> {c17ap}  PadTypeInGeneric [TEXT]
    1:2:2: PARSEREF 0x555556d005b0 <e58#> {c17al}  o [TEXT]
    1: PACKAGE 0x555556cbc120 <e187#> {c1aj}  prim_pad_wrapper_pkg  L6 [LIB] [1ps]
    1:2: TYPEDEF 0x555556cdc000 <e19#> {c4ag} @dt=0@  pad_type_e
    1:2:1: DEFIMPLICITDTYPE 0x555556cb22d0 <e20#> {c2am} @dt=0@  __typeimpenum0
    1:2:1:1: ENUMDTYPE 0x555556cb21e0 <e18#> {c2am} @dt=0@
    1:2:1:1:1: BASICDTYPE 0x555556d00000 <e16#> {c2ar} @dt=this@(nw1)  logic kwd=logic
    1:2:1:1:2: ENUMITEM 0x555556d001a0 <e14#> {c3ah} @dt=0@  BidirStd
    1:2:1:1:2:2: CONST 0x555556cbc240 <e13#> {c3as} @dt=0x555556d000d0@(G/w1)  1'h1
```

Would it be possible to limit the substitution depth, such that all references that were used in substitutions would be visible?

alaindargelas, 2021-11-04:
Please don't use Verilator dumps to illustrate an issue, they are no use to me.

RRozak, 2021-11-04:
Ok, so lets use the uhdm:
```
|uhdmtopModules:
\_module: work@top (work@top) param_submodule.sv:42:1: , endln:51:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiPort:
  \_port: (o), line:42:25, endln:42:26, parent:work@top
    |vpiName:o
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: (work@top.o)
      |vpiName:o
      |vpiFullName:work@top.o
      |vpiActual:
      \_logic_net: (work@top.o), line:42:25, endln:42:26, parent:work@top
        |vpiName:o
        |vpiFullName:work@top.o
        |vpiNetType:36
  |vpiModule:
  \_module: work@prim_pad_attr (work@top.u_prim_pad_attr) param_submodule.sv:46:4: , endln:50:6, parent:work@top
    |vpiDefName:work@prim_pad_attr
    |vpiDefFile:param_submodule.sv
    |vpiDefLineNo:31
    |vpiName:u_prim_pad_attr
    |vpiFullName:work@top.u_prim_pad_attr
    |vpiPort:
    \_port: (o), line:31:35, endln:31:36, parent:work@top.u_prim_pad_attr
      |vpiName:o
      |vpiDirection:2
      |vpiHighConn:
      \_ref_obj: (work@top.o), line:49:10, endln:49:11
        |vpiName:o
        |vpiFullName:work@top.o
        |vpiActual:
        \_logic_net: (work@top.o), line:42:25, endln:42:26, parent:work@top
      |vpiLowConn:
      \_ref_obj: (work@top.u_prim_pad_attr.o)
        |vpiName:o
        |vpiFullName:work@top.u_prim_pad_attr.o
        |vpiActual:
        \_logic_net: (work@top.u_prim_pad_attr.o), line:31:35, endln:31:36, parent:work@top.u_prim_pad_attr
          |vpiName:o
          |vpiFullName:work@top.u_prim_pad_attr.o
          |vpiNetType:36
    |vpiModule:
    \_module: work@prim_submodule (work@top.u_prim_pad_attr.u_submodule) param_submodule.sv:35:4: , endln:39:6, parent:work@top.u_prim_pad_attr
      |vpiDefName:work@prim_submodule
      |vpiDefFile:param_submodule.sv
      |vpiDefLineNo:20
      |vpiName:u_submodule
      |vpiFullName:work@top.u_prim_pad_attr.u_submodule
      |vpiPort:
      \_port: (o), line:20:36, endln:20:37, parent:work@top.u_prim_pad_attr.u_submodule
        |vpiName:o
        |vpiDirection:2
        |vpiHighConn:
        \_ref_obj: (work@top.u_prim_pad_attr.o), line:38:10, endln:38:11
          |vpiName:o
          |vpiFullName:work@top.u_prim_pad_attr.o
          |vpiActual:
          \_logic_net: (work@top.u_prim_pad_attr.o), line:31:35, endln:31:36, parent:work@top.u_prim_pad_attr
        |vpiLowConn:
        \_ref_obj: (work@top.u_prim_pad_attr.u_submodule.o)
          |vpiName:o
          |vpiFullName:work@top.u_prim_pad_attr.u_submodule.o
          |vpiActual:
          \_logic_net: (work@top.u_prim_pad_attr.u_submodule.o), line:20:36, endln:20:37, parent:work@top.u_prim_pad_attr.u_submodule
            |vpiName:o
            |vpiFullName:work@top.u_prim_pad_attr.u_submodule.o
[... 268 lines cut ...]
            \_ref_obj: (TargetCfg), parent:TargetCfg.dio_pad_type
              |vpiName:TargetCfg
            |vpiActual:
            \_ref_obj: (dio_pad_type), parent:TargetCfg.dio_pad_type
              |vpiName:dio_pad_type
            |vpiTypespec:
            \_enum_typespec: (prim_pad_wrapper_pkg::pad_type_e), line:4:6, endln:4:16, parent:TargetCfg.dio_pad_type
              |vpiName:prim_pad_wrapper_pkg::pad_type_e
              |vpiBaseTypespec:
              \_logic_typespec: , line:2:17, endln:2:22, parent:prim_pad_wrapper_pkg::pad_type_e
                |vpiInstance:
                \_package: prim_pad_wrapper_pkg (prim_pad_wrapper_pkg::) param_submodule.sv:1:1: , endln:5:34, parent:work@top
              |vpiEnumConst:
              \_enum_const: (BidirStd), line:3:7, endln:3:15, parent:prim_pad_wrapper_pkg::pad_type_e
                |vpiName:BidirStd
                |vpiDecompile:1'h1
                |UINT:1
                |vpiSize:1
              |vpiInstance:
              \_package: prim_pad_wrapper_pkg (prim_pad_wrapper_pkg::) param_submodule.sv:1:1: , endln:5:34, parent:work@top
          |vpiLhs:
          \_parameter: (work@top.u_prim_pad_attr.u_submodule.u_impl_generic.PadTypeInGeneric), line:16:25, endln:16:45, parent:work@top.u_prim_pad_attr.u_submodule.u_impl_generic
```
The `TargetCfg.dio_pad_type` is not visible in that scope.
In "Would it be possible to limit the substitution depth, such that all references that were used in substitutions would be visible?" I meant to limit parameter substitution to module scope.

alaindargelas, 2021-11-04:
"The TargetCfg.dio_pad_type is not visible in that scope.", sure, but if you navigate up the instance tree following the vpiParent relation, you'll find the definition and the value, so you can pass the value to Verilator.

"I meant to limit parameter substitution to module scope." Maybe you can express that in pseudo code, it does not seem to match what the previous requests about not substituting

RRozak, 2021-11-05:
```
"The TargetCfg.dio_pad_type is not visible in that scope.", sure, but if you navigate up the instance tree following the vpiParent relation, you'll find the definition and the value, so you can pass the value to Verilator.
```
I am afraid that it won't work in more complicated cases. Parameter references are resolved by verilator in the later phase of verilation and I think it's better and safer to leave it to that phase.
What do you think about disabling the parameter substitution at all? I mean to start using the flag `--disable-feature=parametersubstitution` again?
I think it should be easier than checking if the parameter is visible.

alaindargelas, 2021-11-05:
I'd rather go with the replacement by the actual value

RRozak, 2021-11-05:
Ok, it will work too.

alaindargelas, 2021-11-05:
Just to be explicit, you have to fetch the value in the plugin as I described above:

"The TargetCfg.dio_pad_type is not visible in that scope.", sure, but if you navigate up the instance tree following the vpiParent relation, you'll find the definition and the **value**, so you can pass the value to Verilator.

RRozak, 2021-11-16:
I described in https://github.com/chipsalliance/Surelog/issues/2199 why solution proposed by you won't work. I provided the case on which the parameter name in that vpiParamAssign node is ambigous.

alaindargelas, 2021-11-18:
#2212 should fix this problem, please confirm

alaindargelas, 2021-11-20:
Let me know when we can close this issue

RRozak, 2021-11-22:
There is vpiHierPath node without the typespec node. It gives the similar problem to https://github.com/chipsalliance/Surelog/issues/2199#issuecomment-975422848.
We don''t have information needed to compute the value.

```
    |vpiParamAssign:
    \_param_assign: , line:33:25, endln:33:36, parent:work@top.u_prim_pad_attr
      |vpiOverriden:1
      |vpiRhs:
      \_hier_path: (TargetCfg.dio_pad_type), line:47:38, endln:47:60
        |vpiName:TargetCfg.dio_pad_type
        |vpiActual:
        \_ref_obj: (TargetCfg), parent:TargetCfg.dio_pad_type
          |vpiName:TargetCfg
        |vpiActual:
        \_ref_obj: (dio_pad_type)
          |vpiName:dio_pad_type
        |vpiExpr:
        \_constant: , line:44:39, endln:44:40
      |vpiLhs:
      \_parameter: (work@top.u_prim_pad_attr.PadType), line:33:25, endln:33:36, parent:work@top.u_prim_pad_attr
```

closed 2021-11-26 by alaindargelas