chipsalliance/Surelog #2183: Wrong value of parameter and type problems resulting from parameter substitution

RRozak, 2021-11-10:
Please consider the following example
```systemverilog
module prim_subreg;
   parameter logic [4:0] RESVAL = '0;
   int a = int'(RESVAL);
endmodule // prim_subreg

module prim_subreg_shadow;
   typedef struct packed {
      logic [2:0] a;
      logic [1:0] b;
   } struct_t;

   typedef enum logic [2:0] {
      ENUM_ITEM = 3'b000
   } enum_t;

   parameter struct_t RESVAL = '{
      a: ENUM_ITEM,
      b: '1
   };

   prim_subreg #(
      .RESVAL(RESVAL)
   ) staged_reg ();
endmodule // prim_subreg_shadow

module top;
   typedef struct packed {
      logic [1:0] a;
      logic [2:0] b;
   } struct_t;

   typedef enum logic [1:0] {
      ENUM_ITEM = 2'b11
   } enum_t;

   parameter struct_t CTRL_RESET = '{
      a: ENUM_ITEM,
      b: '0
   };

   prim_subreg_shadow #(
      .RESVAL(CTRL_RESET)
   ) u_ctrl_reg_shadowed ();
endmodule // top
```

The value of parameter in `int a = int'(RESVAL);` is wrong.
The problem is also with the type. `ENUM_ITEM` (in `vpiParamAssign` of `prim_subreg`) is the constant from enum declared in `top` module. There is no information about scope of definition, so the constant from `prim_subreg_shadow` would be taken instead. 
```
|uhdmtopModules:
\_module: work@top (work@top) pattern_enum.sv:26:1: , endln:44:10
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiModule:
  \_module: work@prim_subreg_shadow (work@top.u_ctrl_reg_shadowed) pattern_enum.sv:41:4: , endln:43:29, parent:work@top
    |vpiDefName:work@prim_subreg_shadow
    |vpiDefFile:pattern_enum.sv
    |vpiDefLineNo:6
    |vpiName:u_ctrl_reg_shadowed
    |vpiFullName:work@top.u_ctrl_reg_shadowed
    |vpiModule:
    \_module: work@prim_subreg (work@top.u_ctrl_reg_shadowed.staged_reg) pattern_enum.sv:21:4: , endln:23:20, parent:work@top.u_ctrl_reg_shadowed
      |vpiDefName:work@prim_subreg
      |vpiDefFile:pattern_enum.sv
      |vpiDefLineNo:1
      |vpiName:staged_reg
      |vpiFullName:work@top.u_ctrl_reg_shadowed.staged_reg
      |vpiInstance:
[... 543 lines cut ...]
        \_typespec_member: (b), line:29:19, endln:29:20, parent:struct_t
          |vpiName:b
          |vpiTypespec:
          \_logic_typespec: , line:29:7, endln:29:12, parent:b
            |vpiRange:
            \_range: , line:29:14, endln:29:17
              |vpiLeftRange:
              \_constant: , line:29:14, endln:29:15
                |vpiConstType:9
                |vpiDecompile:2
                |vpiSize:64
                |UINT:2
              |vpiRightRange:
              \_constant: , line:29:16, endln:29:17
                |vpiConstType:9
                |vpiDecompile:0
                |vpiSize:64
                |UINT:0
    |vpiLhs:
    \_parameter: (work@top.CTRL_RESET), line:36:23, endln:39:5, parent:work@top
  |vpiTypedef:
  \_enum_typespec: (enum_t), line:34:6, endln:34:12
  |vpiTypedef:
  \_struct_typespec: (struct_t), line:27:12, endln:27:18, parent:work@top.CTRL_RESET
```

alaindargelas, 2021-11-10:
- The parameter object always contain the value assigned in the scope (default value) as we discussed before, so it is not wrong, it behave as we defined it. 
The param_assign contains the overloaded value, in this case the struct coming from 2 levels up. Granted, the type is masked in between. I'll try to use the vpiInstance pointer to point to the scope where it is defined.

alaindargelas, 2021-11-11:
Fixed by #2601. By constant pushing the enum actual value.

closed 2021-11-11 by alaindargelas