chipsalliance/Surelog #2184: Wrong value substituted for nested module parameter

rkapuscik, 2021-11-10:
For the following file: tests/FourCells/top.sv
```verilog
package foo_pkg;
  parameter int VALUE_TEMP = '{default: 1};
endpackage

module bottom (output int bottom_o);
  parameter int VALUE = 8;
  assign bottom_o = VALUE;
endmodule

module lower (output int lower_o);
  import foo_pkg::*;
  parameter int VALUE = VALUE_TEMP + 2;
  bottom #(
    .VALUE(VALUE)
  ) bottom_u(
    .bottom_o(lower_o)
  );
endmodule

module upper (output int upper_o);
  parameter int VALUE = 6;
  lower lower_u (
    .lower_o(upper_o)
  );
endmodule

module top (output int top_o);
  parameter int VALUE = 5;
  upper upper_u(
    .upper_o(top_o)
  ) ;
endmodule
```
parameter `VALUE ` in `bottom_u` is incorrect:
```
        |vpiParamAssign:
        \_param_assign: , line:6:17, endln:6:26, parent:work@top.upper_u.lower_u.bottom_u
          |vpiOverriden:1
          |vpiRhs:
          \_constant: , line:6:25, endln:6:26
            |vpiConstType:9
            |vpiDecompile:6
            |vpiSize:32
            |UINT:6
            |vpiTypespec:
            \_int_typespec: , line:6:13, endln:6:16, parent:work@bottom.VALUE
          |vpiLhs:
          \_parameter: (work@top.upper_u.lower_u.bottom_u.VALUE), line:6:17, endln:6:26, parent:work@top.upper_u.lower_u.bottom_u
```
Interestingly, if we change the parameter name in `lower` to anything other (i.e. `val`), UHDM is correct:
```
        |vpiParamAssign:
        \_param_assign: , line:6:17, endln:6:26, parent:work@top.upper_u.lower_u.bottom_u
          |vpiOverriden:1
          |vpiRhs:
          \_operation: , line:14:12, endln:14:15
            |vpiOpType:24
            |vpiOperand:
            \_operation: , line:12:23, endln:12:33
              |vpiOpType:75
              |vpiOperand:
              \_tagged_pattern: 
                |vpiPattern:
                \_constant: , line:2:41, endln:2:42
                  |vpiConstType:9
                  |vpiDecompile:1
                  |vpiSize:64
                  |UINT:1
                |vpiTypespec:
                \_string_typespec: (default), line:2:32, endln:2:39
                  |vpiName:default
              |vpiTypespec:
              \_int_typespec: , line:2:13, endln:2:16
                |vpiInstance:
                \_package: foo_pkg (foo_pkg::) /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/tests/FourCells/top.sv:1:1: , endln:3:11, parent:work@top
                  |vpiDefName:foo_pkg
                  |vpiName:foo_pkg
                  |vpiFullName:foo_pkg::
                  |vpiParameter:
                  \_parameter: (foo_pkg::VALUE_TEMP), line:2:17, endln:2:43, parent:foo_pkg::
                    |vpiName:VALUE_TEMP
                    |vpiFullName:foo_pkg::VALUE_TEMP
                    |vpiTypespec:
                    \_int_typespec: , line:2:13, endln:2:16, parent:foo_pkg::VALUE_TEMP
                      |vpiInstance:
                      \_package: foo_pkg (foo_pkg::) /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/tests/FourCells/top.sv:1:1: , endln:3:11, parent:work@top
                  |vpiParamAssign:
                  \_param_assign: , line:2:17, endln:2:43, parent:foo_pkg::
                    |vpiRhs:
                    \_operation: , line:2:30, endln:2:43
                      |vpiOpType:75
                      |vpiOperand:
                      \_tagged_pattern: 
                        |vpiPattern:
                        \_constant: , line:2:41, endln:2:42
                          |vpiConstType:9
                          |vpiDecompile:1
                          |vpiSize:64
                          |UINT:1
                        |vpiTypespec:
                        \_string_typespec: (default), line:2:32, endln:2:39
                          |vpiName:default
                    |vpiLhs:
                    \_parameter: (foo_pkg::VALUE_TEMP), line:2:17, endln:2:43, parent:foo_pkg::
            |vpiOperand:
            \_constant: , line:12:36, endln:12:37
              |vpiConstType:9
              |vpiDecompile:2
              |vpiSize:64
              |UINT:2
            |vpiTypespec:
            \_int_typespec: , line:12:13, endln:12:16
          |vpiLhs:
          \_parameter: (work@top.upper_u.lower_u.bottom_u.VALUE), line:6:17, endln:6:26, parent:work@top.upper_u.lower_u.bottom_u
```

alaindargelas, 2021-11-12:
I simplified your testcase:
```
package foo_pkg;
  parameter int VALUE_TEMP = '{default: 1};
endpackage

module bottom ();
  parameter int VALUE = 8;
endmodule

module lower ();
  import foo_pkg::*;
  parameter int VALUE = VALUE_TEMP + 2;
  bottom #(
    .VALUE(VALUE)
  ) bottom_u();
endmodule
```
The result is correct, the param_assign contains the correct overloaded value. 
The parameter contains the default value (We decided that semantic long time ago).

```
|uhdmtopModules:
\_module: work@lower (work@lower), id:59 tests/ScratchPad.sv:9:1: , endln:15:10
  |vpiDefName:work@lower
  |vpiName:work@lower
  |vpiModule:
  \_module: work@bottom (work@lower.bottom_u), id:64 tests/ScratchPad.sv:12:3: , endln:14:16, parent:work@lower, parID:59
    |vpiDefName:work@bottom
    |vpiDefFile:tests/ScratchPad.sv
    |vpiDefLineNo:5
    |vpiName:bottom_u
    |vpiFullName:work@lower.bottom_u
    |vpiInstance:
    \_module: work@lower (work@lower), id:59 tests/ScratchPad.sv:9:1: , endln:15:10
    |vpiParameter:
    \_parameter: (work@lower.bottom_u.VALUE), id:65, line:6:17, endln:6:26, parent:work@lower.bottom_u, parID:64
      |vpiName:VALUE
      |vpiFullName:work@lower.bottom_u.VALUE
      |UINT:8               <<---------------------------- Default value
      |vpiTypespec:
      \_int_typespec: , id:66, line:6:13, endln:6:16, parent:work@lower.bottom_u.VALUE, parID:65
    |vpiParamAssign:
    \_param_assign: , id:55, line:6:17, endln:6:26, parent:work@lower.bottom_u, parID:64
      |vpiOverriden:1
      |vpiRhs:
      \_operation: , id:47, line:13:12, endln:13:17                 <<------------------------ Correct value
        |vpiOpType:24
        |vpiOperand:
        \_operation: , id:48, line:11:25, endln:11:35, parID:47
          |vpiOpType:75
          |vpiOperand:
          \_tagged_pattern: , id:49, parID:48
            |vpiPattern:
            \_constant: , id:51, line:2:41, endln:2:42, parID:49
              |vpiConstType:9
              |vpiDecompile:1
              |vpiSize:64
              |UINT:1
            |vpiTypespec:
            \_string_typespec: (default), id:50, line:2:32, endln:2:39, parID:49
              |vpiName:default
          |vpiTypespec:
          \_int_typespec: , id:52, line:2:13, endln:2:16, parID:48
            |vpiInstance:
            \_package: foo_pkg (foo_pkg::), id:0 tests/ScratchPad.sv:1:1: , endln:3:11, parent:work@lower, parID:56
              |vpiDefName:foo_pkg
              |vpiName:foo_pkg
              |vpiFullName:foo_pkg::
              |vpiParameter:
[... 79 lines cut ...]
          \_constant: , id:36, line:2:41, endln:2:42, parID:34
            |vpiConstType:9
            |vpiDecompile:1
            |vpiSize:64
            |UINT:1
          |vpiTypespec:
          \_string_typespec: (default), id:35, line:2:32, endln:2:39, parID:34
            |vpiName:default
        |vpiTypespec:
        \_int_typespec: , id:24, line:2:13, endln:2:16, parent:work@lower.VALUE_TEMP, parID:23
          |vpiInstance:
          \_package: foo_pkg (foo_pkg::), id:0 tests/ScratchPad.sv:1:1: , endln:3:11, parent:work@lower, parID:56
      |vpiOperand:
      \_constant: , id:38, line:11:38, endln:11:39, parID:37
        |vpiConstType:9
        |vpiDecompile:2
        |vpiSize:64
        |UINT:2
      |vpiTypespec:
      \_int_typespec: , id:27, line:11:13, endln:11:16, parent:work@lower.VALUE, parID:28
    |vpiLhs:
    \_parameter: (work@lower.VALUE), id:62, line:11:17, endln:11:39, parent:work@lower, parID:59
  |vpiTypedef:
  \_import: (foo_pkg), id:25, line:10:10, endln:10:20
```

rkapuscik, 2021-11-12:
Thanks for looking into this! Unfortunately, the example you provided is not equivalent and the error is still there for the original input (tested with current master https://github.com/chipsalliance/Surelog/commit/f6f36e6ba1c4efbbc075a4538699539e8674dc95). I suspect simplifying it had a similar effect as renaming the parameter.

Please notice that the incorrect node I provided is indeed a `param_assign`, where we would expect the final value, and not the `parameter` node. Indeed, the default value for this parameter is 8 as expected, it is just the final substituted one that is wrong.

alaindargelas, 2021-11-13:
Fixed by #2606

closed 2021-11-13 by alaindargelas